MarlinFirmware/Marlin #16137: [BUG] Compile Error Fysetc S6

TB1405, 2019-12-07:
### Bug Description
Compiling my config isnt possible with the Fysetc S6 (for SKR PRO it works)
Have also added my pins_FYSETC_S6 because you need for compiling a filrunout Pin 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3935287/Configuration.zip)
### Steps to Reproduce
load my config files into the marlin 2.0 build and try to compile  

Error 
https://pastebin.com/MZseXr1D

sjasonsmith, 2019-12-07:
That is an odd error, as it isn't actually a compilation problem, more like the linker was executed with invalid arguments.

I can build your configs just fine from PlatformIO within VSCode on Windows. I tested building them on top of both the 2.0 and bugfix-2.0.x branches.

TB1405, 2019-12-07:
But i only have a .elf firmware file. S6 needs also binary file or not?

sjasonsmith, 2019-12-07:
My build generates a firmware.bin file at `Marlin/.pio/build/FYSETC_S6/firmware.bin`

TB1405, 2019-12-07:
I sure i only have a firmware.elv file. 
I will check this tomorrow

sjasonsmith, 2019-12-07:
The error output you posted does look related to packaging the final binary, so I believe that you probably are missing the .bin file, I just don't know why.

Make sure you have all your components updated in Platform IO. If you are in VS Code, you can find it here:

[image]

TB1405, 2019-12-09:
Same problem after the update. it only generate a .elf file and the error 1

boelle, 2019-12-11:
tried to hit the trashcan symbol?

[image]

or delete these 2 folders?

[image]

of course need to restart vs code

TB1405, 2019-12-11:
@boelle 
tried both but same problem

TB1405, 2019-12-20:
have a new laptop and with this one it works.. dont know where was the problem.

closed 2019-12-20 by TB1405

ModestAndroid, 2019-12-30:
I'm having this same issue. I've tried some of the suggestions above, as well as reinstalling Atom. Same error for Fysetc S6...