MarlinFirmware/Marlin #16142: [BUG] (Compile Error LPC1768)

borland1, 2019-12-07:
### Bug Description

Compile 'Fails' if display MKS_12864OLED_SSD1306 is enabled with processor LPC1768,

But compile 'Succeeds' in PlatformIO (or Arduino IDE) with:  
default_envs = megaatmega2560
& 8-bit motherboard: MKS_GEN_L

### My Configurations
Marlin release version: 2.0.x
PlatformIO:   default_envs = LPC1768

### Steps to Reproduce

Compile 'Fails' with errors when enabling the following two lines
in file Configuration.h:

line 133, #define MOTHERBOARD BOARD_MKS_SGEN_L
line 2004,  #define MKS_12864OLED_SSD1306;

#### Additional Information

[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3935666/ConfigFiles.zip)

thisiskeithb, 2019-12-08:
It's because `MKS_12864OLED_SSD1306` isn't setup for the SGEN-L. Looking at the [pins file](https://github.com/MarlinFirmware/Marlin/blob/39fcb5b5f891cea26b1e8ab0d4b749565e830582/Marlin/src/pins/lpc1768/pins_MKS_SGEN_L.h), only the following LCDs/controllers are setup:

- `CR10_STOCKDISPLAY`
- `FYSETC_MINI_12864`
- `FYSETC_MINI_12864_1_2`
- `FYSETC_MINI_12864_2_0`
- `FYSETC_MINI_12864_2_1`
- `MKS_MINI_12864`
- `ULTIPANEL`

If you have the MKS_12864OLED_SSD1306 pinout, we can use that to map the pins.

borland1, 2019-12-08:
I can confirm that the MKS_12864OLED_SSD1306 display does work with the provided Firmware.bin file.
I also confirmed that SD-card data is read from this displays SD-card slot.  So, apparently you have the pin assignments correct for both ports EXP1 and EXP2.  

Thank you for the effort.

thisiskeithb, 2019-12-08:
Thanks for confirming the pinout is correct!

I’ll put in a PR later to get this screen/controller support added to the SGen-L.

thisiskeithb, 2019-12-11:
@borland1: The fix was merged into the latest `bugfix-2.0.x` code, so can you download a fresh copy & make sure everything works as intended?

boelle, 2019-12-11:
since a fix has been merged i will close this one

we can always reopen

closed 2019-12-11 by boelle