MarlinFirmware/Marlin #16146: G29 Probing failed after 1 Probe Point

ilakuli, 2019-12-08:
### Bug Description

After updating from Bugfix to Marlin 2.0 release, my bltouch on my CR10S with SKR Pro board is acting weird. It homes normally, also the M48 Test works fine. But when I start a print or issue a G29 manually through console, it probes the first point fine, moves to the second but then does not depoly the bltouch and then of course fails probing.

### Steps to Reproduce

1. Start Print with G29 in start script or issue G29 via Terminal

**Expected behavior:** [What you expect to happen]

Do 16 Point leveling (and then start printing if print started)

**Actual behavior:** [What actually happens]

BLtouch fails to deploy at point 2 and probing fails. Still starts printing normally if print started

Vertabreak, 2019-12-08:
gonna need to post configs for us to see whats happening.

ilakuli, 2019-12-08:
I'm sorry there must have been an error uploading. I updated my initial comment.

ilakuli, 2019-12-09:
No ideas?

boelle, 2019-12-11:
nope no idea, i never had an issue with G29

the only difference we have is the BLtouch, i have an inductive probe

but you cant reuse old configs, use a tool like winmerge to compare old config with the new default and then copy over settings

once firmware is flashed send M502 and then M500, settings can be hiding in eeprom and give the most funny problems

ilakuli, 2019-12-11:
I didn't use the same config files, I just copied the same settings to the new firmware by hand. But since then I changed to a different board flashed with the newest marlin release (just to test) and the Printer and bltouch work just fine. So I suspect the board has some kind of defect.

closed 2019-12-11 by ilakuli

boelle, 2019-12-11:
> So I suspect the board has some kind of defect.

that is not uncommon, a lot of these china boards are made fast and cheap, so errors will happen

Thijsvanzwam, 2020-04-25:
i have the same issue here on a BTT SKR 1,4 using an inductive probe. Tried a switch at the z-stop connector but same error. When activating ABL using G29 or on the screen it probes 1 point and doesn't respond tot the z-stop at the second probe point, giving an error.
z-stop works fine on autohome so the connection on the board works
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4533993/config.zip)

tried Marlin 2.0.5.3 and bugfix