MarlinFirmware/Marlin #16150: [FR] configuration loading at runtime

MarcelMo, 2019-12-08:
I know it is a major change, but it should be feasible to change the way the device configuration configuration is loaded, so that all device ( = printer ) features are loaded at runtime. for example out of eeprom or sd-card. 
So there would be one binary for each board ( or processor type), and no unique binary for every single printer setup that exists...
(I think smoothieware uses something like that)
This would really make support so much easier, since all the newbies don't need to compile marlin anymore and it would greatly improve the ease of an upgrade!
I guess one drawback might be increased binary size, since all possible features and modules must be present, but most users switch to 32bit anyway.

thisiskeithb, 2019-12-08:
This was [asked recently](https://www.facebook.com/groups/1049718498464482/permalink/1849835075119483/) on Marlin’s Facebook group and was answered with a “No.” from Scott (@thinkyhead) since it’d require a substantial rewrite:

> Marlin is specifically architected to allow for builds as small as 50K to fit on the most modest boards. To accomplish this Marlin uses meta-programming via the C++ preprocessor, constant values as much as possible, and so on. No consideration whatsoever has been made to allow for all features to be available for enablement and configuration during run-time (like RepRapFirmware). Because of this, a bottom-up rewrite would most definitely be required. And don’t even get me started on the complications introduced by supporting multiple architectures. It would be highly advisable to start with a mature base like RepRapFirmware or Smoothieware, port the Marlin HAL over to it, and then with great care port Marlin features and display support over to it. It would be advisable to divide up many parts of Marlin into libraries.

If you’re after boot-time config options, it’d be better to go with Klipper, Smoothieware, RepRapFirmware, etc.

InsanityAutomation, 2019-12-09:
And plenty of machines still ship with Melzi / Atmega1284p boards with only 128kb of program space....

M-Mommsen, 2019-12-10:
So I guess the question is what is the goal of this project! 
Sounds like "easy to deploy and maintain" is not as important as support for minimalistic hardware - I really thought the 16bit era was done... 

Thanks you, for the answers!

boelle, 2020-02-03:
as the maintaner has given a firm no this one will be closed

you can always submit a PR, or if you are not happy with the direction marlin is going you can always do a fork of marlin and do everything like you want it without having to compromise

closed 2020-02-03 by boelle

thinkyhead, 2020-02-04:
> So I guess the question is what is the goal of this project!

It will continue to evolve at its own pace. Unless you were offering to help out, it's pointless to demand more of us. I'm working 12 hours a day 7 days a week, and Your Wish is on my list, just not at the top.