MarlinFirmware/Marlin #16152: [BUG] SD card filenames are broken (stuck in t he dinosaur age).

gruvin, 2019-12-08:
### Bug Description

I have read the code of conduct and I'm trying really hard not to be an ass here ... but ...

Ancient 8.3 filenames died out with the dinosaurs. The ONLY filename format coming out of Marlin's M20 command should be ordinary, reasonable and perfectly normal filenames. The year is 2019, not 1991. 

This *IS* a bug, not a feature request. A "bug", is where code does not perform reasonably expected behaviour. This code is buggy. 👅 I'm sure there's some argument stating I'm wrong but I'm not. So forget about that and let's just move on eh? :p

### My Configurations

Marlin, running on anything.

### Steps to Reproduce

1. Execute M20
2. Try not to stress out in disbelief :p

**Expected behavior:** [What you expect to happen]
A list of filenames is output.

**Actual behavior:** [What actually happens]
A list of cryptic, time shackled abominations from a bygone error appear.

#### Additional Information
For the life of me, I can think of no reasonable argument in favour of even supporting 8.3 names, let alone having them the only option for M20. PROGMEM space on an ATmega1280, perhaps? The year is 2019, not 1991.

Yes, I know about M33 and I sorry but I frankly don't care. M33 is an inadequate bandaid to a problem that shouldn't exist in the first place.

boelle, 2019-12-08:
> This _IS_ a bug, not a feature request. A "bug", is where code does not perform reasonably expected behaviour.

What if the expected behaviour does not perform reasonble?

a bug is when the code does not perform as it was written, so if the code was written for 8.3 filenames and it handles those perfectly there is not bug

But a small secret, you are more than welcome to make a PR that changes marlin to behave just like you want it

no dev people here are paid so they are not forced to do anything

and since you dont pay for marlin you cant demand anything either, and that is the case for all of us

if we want to change marlin we have to make changes and send in a PR

gruvin, 2019-12-08:
> a bug is when the code does not perform as it was written
Begging your pardon, but ALL bugs perform EXACTLY as written.

> you are more than welcome to make a PR 
.. is just another way of saying, "if you're not a programmer then sod off"

> no dev people here are paid so they are not forced to do anything
Oh. I didn't realize that. Shiucks. My bad. 

> you cant demand anything either
Gosh. I guess that's a good point. Just as well I'm not demanding anything How would that even be done, anyway? Do tell.

> if we want to change marlin we have to make changes and send in a PR

Why? So some dude can say, "I am not paid so I'm not forced to accept this PR"?

I have zero confidence that long filenames will ever become the default. So what is left, other than to voice disappointment, through opinion?

boelle, 2019-12-08:
> "I am not paid so I'm not forced to accept this PR"?

PR's are accepted if they dont break marlin 

> ther than to voice disappointment, through opinion?

Constructive critism is good

boelle, 2019-12-08:
> ... is just another way of saying, "if you're not a programmer then sod off"

Depends on how you look at it

gruvin, 2019-12-08:
> PR's are accepted if they dont break marlin

Yes but right there is the issue at hand, is it not. Define, "break". If not here, then where should such a debate take place _and_ be noticed?

boelle, 2019-12-08:
> Define, "break". I

make something that worked not work

gruvin, 2019-12-08:
> make something that worked not work
How can a design bug be made to not work or work? Design predates code. Are we to write code to retain all past behaviours, even if they make no sense?

Does removing 8.3 filenames and replacing them with real ones cause Marlin to not work? I say no. The question is, will the lead developer agree? I say no. Evidence is everywhere.

So what to do? Convince him otherwise or try to fork the entire project? Neither seem plausible.

gruvin, 2019-12-08:
Resistance is futile.

closed 2019-12-08 by gruvin

ManuelMcLure, 2019-12-08:
>Does removing 8.3 filenames and replacing them with real ones cause Marlin to not work?

Yes, on a great majority of the boards Marlin supports where we're running into hard limits on the CPU power and RAM available. We have to fight for every byte of SRAM and flash.

tpruvot, 2019-12-08:
not the right day :p imo

sjasonsmith, 2019-12-08:
@gruvin, I doubt anyone disagrees that long filenames would be useful, there are just a great many complications which have already been discussed elsewhere. As much as it would be convenient for _you_ to change the behavior, it would cause problems in many other scenarios. Even if it can be done in a safe manner, there is a great deal of work involved in ensuring that it actually isn't breaking people. Even small changes often have unintended consequences. I can attest to this where my own changes have introduced new issues, and I suspect the same is true for anyone who has every actually contributed to the project.

Any productive conversation on the topic was completely defeated by the tone of the request. You aren't going to convince anyone to come to your aid by attacking them.

gruvin, 2019-12-09:
> @ ManuelMcLure: Yes, on a great majority of the boards Marlin supports where we're running into hard limits on the CPU power and RAM available.

In truth, I understand that very well. However, Marlin 1.x is still there and at _some_ point, support for older boards is going to have to be dropped. In the meantime, we can continue to add a zillion more `#if ENABLED()` pragmas, I suppose. 

Either way, it's a pretty poor excuse to ignore or mark as, "too hard" so many requests, over so many years already, as _absolutely_ _*has*_ been the case. I don't profess to know all the ins and outs. However, I don't much care either. This is never going to be fixed if we aren't allowed to at least get started.

To that end, please see my new pull request: https://github.com/MarlinFirmware/Marlin/pull/16156

(My programming skills are barely up to this task but, since I was told that only programmers capable of submitting pull request could even have a say (above) I figure I should at least try.

ManuelMcLure, 2019-12-09:
So, are you volunteering to maintain 1.1.x for everyone (including the tens of thousands of Ender-3s and the like with 8-bit boards) so that we can go hog-wild adding new features to 2.0.x? Maintaining two separate diverging code bases is a lot more work.

gruvin, 2019-12-09:
> @sjasonsmith: Any productive conversation on the topic was completely defeated by the tone of the request. You aren't going to convince anyone to come to your aid by attacking them.

Fair enough, though I don't really consider my OP as an attack on anyone. No names were mentioned and I thought I was clear about my general frustration and even used the obligatory, "imo" thingy. Well, I got schooled, again, I guess. *sigh* ANYWHO ...

(btw, your earlier comments noted and understood)

I am frustrated, if that was not clear.  As you alluded,  I and many others have battled or bitched about this topic on and off for YEARS. `M33` happened but ... well never mind.

It _*seems*_ that no one wants to even start on code to fix it, because everyone else just keeps coming up with a zillion reasons why it, "can't be done", because it's, "too hard" and "it will break XYZ and ABC and who knows what else." Some have tried in the past and just been rejected [citation needed]. Hence my tongue in cheek reference to dinosaurs.

In any case, *please see my new [pull request](https://github.com/MarlinFirmware/Marlin/pull/16156)* and comment there, if you like. I'm surprised I was able to even do it, since I'm hardly a pro programmer. I do worry that my work will be rejected but I just HAD to try.

### I have closed this issue ...
... as I too dislike its general tone. My bad. It has long since departed from any semblance of a bug report, after all. Forums don't work. I though maybe I could get some support and ideas here but I guess I miss fired.

Thanks to those who discerned my frustration and were kind enough to respectfully school me on my tone, whilst still getting my point. I'll try harder in future. (I'm 49 and _still_ struggle with this stuff. *sigh* Drama queen? :p)

sjasonsmith, 2019-12-09:
I think it’s great that you made an attempt and posted a PR as a starting point. I hope it results in productive discussion and figuring out good options to improve the situation. I think it is often just that a feature needs a champion who will help work through the issues that arise.

I don’t actually have any authority over anything, so maybe it will still be completely rejected. It seems like a reasonable optional capability to me, if it can be handled in a way that won’t break compatibility with existing applications and devices.

karabas2011, 2019-12-23:
The lib used in smoothieware supports long names and UNICODE names for other languages!!
I dont know is it too hard to port it to marlin.