MarlinFirmware/Marlin #16155: [BUG] Motor stuttering with linear advance on TMC2208 since Marlin 2.0

Schild0r, 2019-12-08:
Hey there, in the past I often tried to switch to Marlin 2.0 from 1.1.9 but there was always some hinderance.
Eventually I managed to sort out most of it (last was the motor knocking noise where you have to disable monitor_driver_status but I digress)

### Bug Description

So I ported all the settings from my working 1.1.9 configuration to Marlin 2 (running an MKS Gen L with 5x TMC2208 in UART mode, dual BMG bowden extruder). 
In 1.1.9 I made use of linear advance and dialed in my K factor at 1.45 Now with Marlin 2 I get this weird stuttering on my extruder motor when linear advance is used (no stutter when I set K = 0) 
Any ideas why this is?
(Here's also a video of that symptom: [https://www.reddit.com/r/3Dprinting/comments/e73s9h/marlin_2_tmc2208_stutter_with_linear_advance/](url) )

### Already tried

I already tried using square wave stepping or setting the minimum stepper pulse to 1 or 2 (with TMC2209s installed you would get an error which tells you to do so with linear advance enabled and Trinamic drivers however with TMC2208 drivers you don't get that error message but I tried it anyways). 
Also it makes no difference whether I use stealthchop or spreadcycle or make use of hybrid_threshold.
Running the driver in standalone mode or changing a TMC2209 driver and running that one in standalone mode makes no difference either

also see the attached configuration files (only changes I made are in these two files and always denoted with "//MS" in that line )
[Schild0r_Marlin2.0_config.zip](https://github.com/MarlinFirmware/Marlin/files/3937298/Schild0r_Marlin2.0_config.zip)

randellhodges, 2019-12-09:
Try disabling stealthchop on E. From everything that I read, TMC2208, linear advance and stealthchopped e don't mix.

Schild0r, 2019-12-09:
> Try disabling stealthchop on E. From everything that I read, TMC2208, linear advance and stealthchopped e don't mix.

As stated above this does not fix it. It only makes the stutter a little bit quieter (or it may only seem like it due to increased overall noise of the extruder motor/driver)

ManuelMcLure, 2019-12-09:
This seems like another instance of #10272 - there's a problem with Linear Advance in 2.0 that causes extruder rattling but I don't think anyone knows why yet.

Schild0r, 2019-12-10:
> This seems like another instance of #10272 - there's a problem with Linear Advance in 2.0 that causes extruder rattling but I don't think anyone knows why yet.

Agreed. Might be the same cause. But didn't you say in that post that using CLASSIC_JERK would mitigate the rattling? I will try that with my setup today and if it helps maybe print a set of benchy and calibration cube with JD and noLA vs JERK and LA

boelle, 2019-12-11:
btw is JD calculated correctly? 

http://blog.kyneticcnc.com/2018/10/computing-junction-deviation-for-marlin.html

Schild0r, 2019-12-11:
> btw is JD calculated correctly?
 if you are referring to the values I used above: No I didn't translate them since I did not know the formula. I just used the standard value of the basic config.h file.

So if I translated the Jerk values, my JD would be even lower. But again, I won't complain when I get better quality in less time.

boelle, 2019-12-11:
better quality in less time migt be impossible

the faster you print the quality will suffer, itsa just a matter of printing fast enough

YAost, 2019-12-15:
This issue is simillar to another request
https://github.com/MarlinFirmware/Marlin/issues/16184
You have S-curved acceleration enabled as well, worth checking without it.

ManuelMcLure, 2019-12-15:
In my testing S-curve seemed to make no difference, but your mileage may vary.

robustini, 2019-12-17:
Same issue here with TC2130, damn.

Vertabreak, 2019-12-17:
does that say linear advance k = 1.45 holy cow that's high never seen a working valid k over 0.5 even with over a foot of bowden tube.

robustini, 2019-12-17:
There's a confict with Linear Advance and Junct Deviation, i disable the second and the L.A. work fine here.

sjasonsmith, 2019-12-18:
Yesterday I noticed unexpected (to me) timing on the extruder with linear advance enabled. Perhaps it is related to this issue. I am only timing signals, I don't actually have a driver connected, so I can't be sure that this is related to the stuttering described.

[image]

In this image you can see a steady frequency, which corresponds to the Linear Advance interrupt rate. This is with square-wave stepping, so the narrow "glitches" correspond to two or three steps taken in a single Linear Advance ISR.

I suspect this could be causing the stuttering, and might even cause TMC2208 motors to shut down, based on past behavior I have observed.

ManuelMcLure, 2019-12-18:
When I turned on ``LA_DEBUG`` I was seeing a lot of instances of
```
More than 2 steps per eISR loop executed.
```
which matches up with what you're seeing in the stepper timing.

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

sjasonsmith, 2019-12-24:
I have reproduced 2208 drivers turning off related to the picture I posted. I don’t know whether that is the root cause of this or not, but it seems possible.

Schild0r, 2019-12-24:
> btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

Will test with the latest marlin build and stuff like s-curve turned off etc once I am home in the next few days. Happy holidays to y'all :)

Lord-Quake, 2019-12-26:
Using search I came to this open issue thread

Today I started to design a K-Calibration tower that I want to post on Thingiverse.
I'm working on a real life calibration object as the values coming from the K-factor Calibration Pattern didn't corollate with the quality I was expecting.
This how the prototype tower looks like:
[image]
Here the G-Code file for internal use only: [K-Tower_Calibration_Leo_N.zip](https://github.com/MarlinFirmware/Marlin/files/4002682/K-Tower_Calibration_Leo_N.zip)
(Edit the header to your setup accordingly)

I'm posting here because when K was at 0.04 I received the following debug output:
`More than 2 steps per eISR loop executed.`
At 0.4 and 0.0 all was fine again.

Marlin 2.0.1 was used.
Config files: [K-Configurations_Leo_N.zip](https://github.com/MarlinFirmware/Marlin/files/4002693/K-Configurations_Leo_N.zip)

I can test if needed......

ManuelMcLure, 2019-12-26:
Can you check out @sjasonsmith 's fork linked from https://github.com/MarlinFirmware/Marlin/issues/15473#issuecomment-568802775 and see if that affects the behavior? This is not ready for prime time, although I myself have tested it and it seems to work well.

Lord-Quake, 2019-12-27:
Test completed. No change, same outcome.

I compiled the FW from @sjasonsmith using my configs from above.

On another note:
For this test I mistakenly used the gcode from the K-Pattern script instead of my cal. tower.
When the printer drew the very first K line the console gave out "More than 2 steps per eISR loop executed." from the get-go and when the print head came to the end of the first line it paused for a brief period before it moved on to print the second K line. I don't recall this happening with the 2.0.1 FW.

qwewer0, 2020-01-03:
I have this problem too, thought it was normal even if it's annoying.
Ender 3, SKR Mini E3 v1.2, Marlin 2.0.0/2.0.1, LA K 0.4 - 1.4, S-Curve, Junction deviation on/off

boelle, 2020-01-10:
@Schild0r  is this still an issue when using bugfix 2.0.x ?

qwewer0, 2020-01-10:
Compiled a fresh bugfix 2.0.x, the problem is still there.

boelle, 2020-01-23:
how often do you update? bugfix is updated almost daily

qwewer0, 2020-01-24:
Recompiled the latest bugfix build [2020.01.24.], the problem is still here.
Ender 3, SKR Mini E3 v1.2, TMC2209.

boelle, 2020-01-28:
i just got TMC2209 and dont have the problem, so the issue is likely to be the  SKR Mini E3 v1.2 (or MKS Gen L which is what @Schild0r uses) as that is the only difference

Schild0r, 2020-01-28:
Compiled the latest bugfix (or at least the one from 5 hours ago) and did some testing. Found out that decreasing the K Value makes the stutter less pronounced/shorter which makes sense I guess since the extruder does not have to decelearate as long as it would have to at higher values.

I then tested at a K value of 0.4 and found that - while at higher values you hear the stutter at every junction - at that low value the stutter seems to only occur on small or sharp movements. 
So Lets say we print a 20x20x20 calibration cube with a brim of 5 lines. The vibration is notable on the corners of the cube since they are sharp. But since the corners of the brim get rounded by the slicer, you can only hear it at that point where the transition from one brim line to the next inner brim line is located (same with solid infill etc. only where the small transition to the next line is, stutter/vibration occurs).

While it was printing I decreased the feedrate to 80% then 50% but even at 30% I could still hear and feel the vibration regardless without a notable difference which does NOT make sense to me since linear advance is a function of speed and decreasing speed should then decrease deceleration needed doesn't it?

ManuelMcLure, 2020-01-28:
> i just got TMC2209 and dont have the problem, so the issue is likely to be the SKR Mini E3 v1.2 (or MKS Gen L which is what @Schild0r uses) as that is the only difference

NO!! This happens with TMC2209 (or A4988) on Re-ARM as well. I _can't_ use Linear Advance on my system (Re-ARM, BMG extruder,  in direct drive, either TMC2209 or A4988 on the extruder) without rattling and underextrusion. If it works for you, great! But don't say it's due to the SKR Mini, or due to TMC2208. There are known issues with LA generating pulses out of sync with the normal extruder stepper pulses that cause rattling.

qwewer0, 2020-01-28:
So, the higher the K value the more the motor has to compensate, and thus makes more noise. Makes sense.
I didn't even realise it, but I did hear it more with K = 1.2 than K=0.4.

boelle, 2020-01-28:
when i did the calibration patterni ended up with an insane low value of 0.05
i started out with 0-2 in the pattern and i had to go below 1 so did 0-1 in 0.01 steps

might explain why i dont hear or see anythin

ManuelMcLure, 2020-01-28:
I see it with values as low as 0.06.

kingofmonkeys, 2020-02-04:
So I just updated my firmware to 2.0.3 and I think i am having this issue now. Skr pro, tmc2209s, LA and s curve enabled. The printer looks like its working great BUT on some of the long flat surfaces i saw a pattern in the print. When i investigated i realized the extruder head (im running direct drive hemera) was vibrating at a pretty high frequency in those sections.  After the current print is done im going to see if i can recreate it in a smaller print for testing.

thinkyhead, 2020-02-04:
@kingofmonkeys — Try disabling S-curve and/or Junction Deviation and compare to see what kind of results you get. There does seem to be bad interaction between S-curve and JD according to reports, and this could extend to the extruder.

One recent (much needed) update we got in a recent version was for step pulse timing, so we should also look at that and see if it could be involved in the behavior.

kingofmonkeys, 2020-02-04:
@thinkyhead I forgot to mention I already have JD disabled, I have been running without it for a while because it seemed to cause issues.  

I updated from 2.0.0 to 2.0.3 so i jumped a few versions (there wasn't a ton of time between those versions) but there were a bunch of updates in there.  I'm currently wondering if the DEFAULT_EJERK could be causing my issue or at least making it noticeable.  I had it set to 1.0 in the 2.0.0 build but saw the default of 5 when updating and decided to split the difference and make it 3.0.

ManuelMcLure, 2020-02-05:
> @kingofmonkeys — Try disabling S-curve and/or Junction Deviation and compare to see what kind of results you get. There does seem to be bad interaction between S-curve and JD according to reports, and this could extend to the extruder.

I did some testing under #10272 including disabling S-curve and found it to have no effect on my problem: https://github.com/MarlinFirmware/Marlin/issues/10272#issuecomment-547226919

kingofmonkeys, 2020-02-05:
So i created a simple test print and ran it under current settings last night.  Tonight i plan to run one with LA turned off to see the results.

Lord-Quake, 2020-02-06:
@kingofmonkeys Could you post your g-code file that is showing the issue.

kingofmonkeys, 2020-02-06:
So this is the first test file, its just a rectangle.  I tested last night with K set to 0 and i'm still getting the odd pattern.  I'm considering going back a few versions to see if it goes away. Since it did the same thing without LA I'm thinking this problem might not be related to the issue here. 

[simple flat.zip](https://github.com/MarlinFirmware/Marlin/files/4164946/simple.flat.zip)

Lord-Quake, 2020-02-06:
I just took a look at it.... over 2:15 hours print and over 11 meters of filament :-O
That's a huge test print. I think 2 cm height should be enough to see the result. I'll be using todays 2.0.x bugfix.

kingofmonkeys, 2020-02-06:
yeah it could stand to be shorter.  the print that i first saw this on was taller and i didn't see it lower in the print.  I know now that i didn't see it lower because of the details lower in the print, the effect is only see on straight sections.  I can shorten this one up but its just a basic rectangle nothing fancy.

kingofmonkeys, 2020-02-09:
Ok after a bunch of testing I found that my issue is NOT the same as this issue and not related to the firmware, sorry for the false alarm.  I had a belt mount that was giving out and causing minor vibrations that I could see in the print.  I just happened to notice the defects after I updated the firmware.

Lord-Quake, 2020-02-09:
Using your g-code I wasn't able to reproduce the issue as well.

kingofmonkeys, 2020-02-10:
yeah it was definitely the faulty bracket, after i found and replaced it no more vibration and no more pattern on the print.  again sorry about that.

Scope666, 2020-02-10:
Myself and a few others have a similar bug, but it's not Marlin's fault.  I believe it's an actual hardware bug with the TMC2209s on the SKR E3 Mini 1.2.  The symptoms are louder than normal motors, INSANELY hotter than normal motor temps, and missed steps.  It would happen maybe once in 10-20 times powering on the machine.  Power cycling would fix it.  Crank the volume to hear it:

https://www.youtube.com/watch?v=rcYc3-wiAYw

thinkyhead, 2020-02-10:
Duplicate of #10272

closed 2020-02-10 by thinkyhead

qwewer0, 2020-02-10:
> I believe it's an actual hardware bug with the TMC2209s on the SKR E3 Mini 1.2.

Could you upload your config files? I have the same board and I had never experienced that you did, 
 I want to make sure that it is not something in the firmware.

Schild0r, 2020-02-10:
Why is this issue closed now?
I see that this might be a duplicate of above mentioned issue but that one is also closed without any solution to the problem and since that conversation moved to discord the conversation gets split even more without anybody knowing if there is someone working on that, what a possible solution could be etc

Scope666, 2020-02-10:
> > I believe it's an actual hardware bug with the TMC2209s on the SKR E3 Mini 1.2.
> 
> Could you upload your config files? I have the same board and I had never experienced that you did,
> I want to make sure that it is not something in the firmware.

Sure, I'm not using the E3 anymore   (on SKR 1.4 now) but my last config was this:

qwewer0, 2020-02-10:
Didn't find any unusual in your configs, but can `SQUARE_WAVE_STEPPING` have this kind of unexpected effect?

Scope666, 2020-02-10:
> Didn't find any unusual in your configs, but can `SQUARE_WAVE_STEPPING` have this kind of unexpected effect?

When I first ran into the issue I tried disabling that, no difference.  I even went so far as to run Klipper for a few weeks, the bug remained... that's how I figured out it was a hardware issue.

qwewer0, 2020-02-10:
Dang

thinkyhead, 2020-02-11:
Methinks in your case @Schild0r we may be seeing a limitation of the MKS Gen L board in keeping up with the demands of so many enabled features and high stepping rates. If the eISR is hitting its limit, it means that Linear Advance is best not used. And then you kind of have to tune from there until performance becomes more acceptable.

We are chasing various things related to stepper timing and the relatively new STM32 boards, and so on, and something will shake out of that. But we are not seeing anything that we can take action on from this meandering thread.

axl12ap, 2020-02-12:
Hello,
what's the situation in the current bugfix 2.0.x of Marlin?? Is still present the LA problem or solved?

I'm using an SKR1.3 with TMC2208 in uart mode, and firmware with s-curve and JD enabled.

I'd like to use also LA but last time (2 month ago) I was having lot of ruttles from extruder motors and leave the function disabled.

Thanks guys

hazedav, 2020-02-23:
I also have LIN_ADVANCE extruder jerky/jittering with TMC 2209 (board SKR mini E3) on my ender 3. Marlin 2.0-bugfix (2.0.3). I notice this issue is closed, but it still seems like a thing.

Please advise.

onek24, 2020-03-26:
SKR Mini E3 with TMC2209 on an Ender 3. 
Lin Advance configured at 1.2 and using classic jerk. 
Version 2.0.5.1 still has this issue, didn't test the newest bugfix. 

The noise sounds almost like someone trying to scratch glass with silicone gloves or with a soapy hand. Or perhaps metal scratching at metal.

dan-and, 2020-03-26:
Same for me with TMC2209 on SKR v1.3 and Marlin bugfix-2.0.x commit 0aeee64cca2364be6025c58bbf1970ff4c77e710 (2020-03-25). 
My k-value is 0.06

jorgenagel, 2020-04-05:
Having this issue since I started using Linear Advance.  
Marlin 2.0_DW3, otherwise stock Ender 5plus
Can't seem to get rid of it even with setting k=0   (was using 0.08 when tuned)

_Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z800.00 E97.18
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E50.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.08
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Unified Bed Leveling:
Recv: echo:  M420 S1 Z2.00
Recv: 
Recv: Unified Bed Leveling System v1.01 active
Recv: 
Recv: Active Mesh Slot: 0
Recv: EEPROM can hold 22 meshes.
Recv: 
Recv: echo:PID settings:
Recv: echo:  M301 P14.72 I0.89 D61.22
Recv: echo:  M304 P690.34 I111.47 D1068.83
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 X-45.00
Recv:   M851 Y-4.00
Recv:   M851 Z-3.32
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L600.00 U700.00
Recv: echo:Filament runout sensor:
Recv: echo:  M412 S1 D5.00
Recv: ok P15 B3_

smerdy, 2020-04-13:
Hello, any progress on this? I have the same issues, TMC2209 on SKR v1.2 and Marlin 2.0.5.3, with k-value of 0.3 and classic jerk. I have experimented with extruder / retractor acceleration and hybrid threshold value for the extruder, no dice. Really want to make use of linear advance but it's too noisy at the moment.

Cirromulus, 2020-04-25:
Same here.
It seems that linear advance tries to load/unload pressure in a corner for every (small) segment, thus creating vibration and overloading the extruder

zvoniimiir, 2020-05-01:
I am still having this issue on ender 3 pro with skr mini e3.

[6 more replies]