MarlinFirmware/Marlin #16158: [BUG] BIGTREE_SKR_PRO_1v1  Pins mixed: calling (pin PF7) turns on (PF8)

carl1961, 2019-12-09:
### Bug Description

Found while using RGB LED after Updating Marlin 2.0.x 

Last working correctly (RGB LED)  Marlin-bugfix-2.0.x "2019-11-10"
Configuration.h

#define RGB_LED
//#define RGBW_LED

#if EITHER(RGB_LED, RGBW_LED)
  #define RGB_LED_R_PIN PF9
  #define RGB_LED_G_PIN PF8
  #define RGB_LED_B_PIN PF7
  //#define RGB_LED_W_PIN -1
#endif

#define STRING_DISTRIBUTION_DATE "2019-11-10"

C:\BIGTREETECH\Working Mega CoreXY\newer\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\variants
BIGTREE_GENERIC_STM32F407_5X

First BROKEN  Version traced back to Marlin-bugfix-2.0.x "2019-11-14"

 #define STRING_DISTRIBUTION_DATE "2019-11-14"

Pins mixed:

turning  Blue LED On (pin PF7) turns on Green LED (PF8) Blue LED never comes on so pin PF7 is not active.

I think issue started when

variants   BIGTREE_GENERIC_STM32F407_5X was updated (changed), with (to) BIGTREE_SKR_PRO_1v1

C:\BIGTREETECH\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\variants
BIGTREE_SKR_PRO_1v1

I have not found any other issues, printer runs fine except for these pins used for RGB LED.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration_Files.zip](https://github.com/MarlinFirmware/Marlin/files/3938180/Configuration_Files.zip)

### Steps to Reproduce

1.  Compile Firmware Marlin-2.0.x for BIGTREE_SKR_PRO  
2.  Issue  M150 B255  from Pronterface or from Display LED Menu

**Expected behavior:** 
Blue LED Light coming on (Pin PF7)
**Actual behavior:** [What actually happens]
Green LED comes on  (Pin PF8)

arminth, 2019-12-21:
Thanks a lot! I will try this now because I was struggeling with the exact same behaviour!

Will report!

arminth, 2019-12-21:
And I can confirm: YOU ARE RIGHT!! Iwas already thinking I am stoo stupid and have to give back my enigineering degree!
Finally I can use my cheapo LIDL LED strip on my printer!
Thx a lot - I hope somone fixes this ASAP in the bugfix version!

Chears
Armin

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

carl1961, 2019-12-25:
soon as I get home, I will check for you.

carl1961, 2019-12-25:
@boelle  New Marlin 2.0.1 Update did not fix this issue, because the (PeripheralPins.c) file or the same. (Identical)
 **but** un-commenting Lines: 188-191 in  Marlin-2.0.x\buildroot\share\PlatformIO\variants\BIGTREE_SKR_PRO_1v1\PeripheralPins.c  fixed this issue.
[image]

boelle, 2019-12-25:
@thinkyhead can you cook up a fix from the above post?

viper93458, 2019-12-26:
Confirmed to fix the generic RGB string LEDs for me as well on the SKR pro. If I used the PC9 PWM pin, then the LED channel/color connected to that PIN would work, but none of the others would. I tried various pins and pin definitions in the lines below:
#define RGB_LED_R_PIN xxx
#define RGB_LED_G_PIN xxx
#define RGB_LED_B_PIN xxx

and ultimately settled on PF9, PF8 and PF7 like the OP did since it was mentioned that those pins did indeed work a while back. I tried other pins just for experimentation.

I hope that uncommenting those lines doesn't break anything else that @LinoBarreca has been working on for the SKR Pro board!

-William

carl1961, 2019-12-26:
> Confirmed to fix the generic RGB string LEDs for me as well on the SKR pro. If I used the PC9 PWM pin, then the LED channel/color connected to that PIN would work, but none of the others would. I [...]

looks to be just timers for those 4 pins that is effected.

LinoBarreca, 2020-01-02:
@boelle this is duplicate with the last one we were talking on.

LinoBarreca, 2020-01-02:
@carl1961 those lines MUST be commented.

boelle, 2020-01-02:
dublicate of #16300, so closing, we only need one open with the same issue

closed 2020-01-02 by boelle

LinoBarreca, 2020-01-02:
If you are wondering why it worked before my modifications the answer is simple:
you were using the board outside its specifications and BTT (stupidly) allowed you to do it (this was causing other issues with timers)
I didn't "break it", I just fixed a bug from BTT making the board work as intended.
### SORRY ABOUT THAT
but any software developer can't (and shouldn't) care about out-of-specs usage by the end-user.
And the end user **must not** believe that an out-of-specs usage of something will be for-ever-working.

Anyway....I realize that it could be useful to attach a RGBW strip to digital pins, though, that's why we opened a feature request for it (using SOFT-PWM). Not many boards have so many analog pwm..so it's a logical feature request....to workaround, at the moment, swap RGB and FANS and `#define FAN_SOFT_PWM`