MarlinFirmware/Marlin #16159: ILI9328 TFT No longer functions [MKS Robin]

sakza, 2019-12-09:
Uncertain which update caused the display to stop functioning, however an older MKS Robin only build also no longer works, which makes me suspicious of some sort of platformio update that has broken support for the ILI9328 based TFT1.1 display/touchscreen present on the MKS Robin. Backlight etc seems to function, as well as the touch buttons (when pressing the screen near where the buttons should display produces the click sound). Uncommenting the reset pin line present in the pins_MKS_robin.h file causes the screen to be 100% white, while leaving it commented nothing is displayed past the booting... message, which is generated by the bootloader and not Marlin.

This issue is generated with a fresh clone of the current Marlin 2.0 repository as of 2019-12-08, compiled with platformio and vscode, windows 10. 

Unfortunately my knowledge of these components runs out about there, and I have no idea how to find the potential update/culprit responsible for the display not generating an image.

Thanks!

sakza, 2019-12-09:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3938537/Configuration.txt)

Configuration.h is attached, however I don't believe it is likely a configuration issue as a previously working build no longer functions without config.h changes

ConceitedCode, 2019-12-10:
Can confirm that I am running into the same issue using a MKS Robin Nano.

I made a minor configuration change (extruder settings) on a build that I downloaded on Sept 15th and using the build on my printer resulted in the above issue. This checkout of the code worked as expected back in September.

I downloaded the latest copy of the 2.0-bugfix branch / compiled / put new build on printer and am running into the same issue.

I downloaded a new copy of the 2.0-bugfix branch using a checkout from August / compiled / put new build on printer and am running into the same issue.

Looks like there is an issue open in the MKS repo for this.
https://github.com/makerbase-mks/MKS-Robin/issues/124

The MKS repo also has an old Marlin 2.x build where they hacked on some adjustment for the display. I'd merge the changes and give it a shot, but the code bases have diverged enough that I don't want to spend the time trying to understand the impact. Hopefully it points someone in the right direction.
https://github.com/makerbase-mks/MKS-Robin/commit/e0aa9ec3c55ea887cfe6ad414ed3fbd97b4e9023#diff-c16abde63f70433fe57793dd686e251f

boelle, 2019-12-11:
tried to do an update all in vscode? what if using arduino ide 1.9 beta (if that can be used)

[image]

sakza, 2019-12-11:
PIO Update all didn't make any difference - I've compiled it on Debian too with no luck hence I don't believe it's my environment causing the issue

tpruvot, 2019-12-16:
I don't think there is a difference between the 1.1 and the 2.0 except for the touchscreen orientation.

the TFT 2.0 which is a ILI9328 is currently ok with the Robin v2.4 (16 dec. build)

tpruvot, 2019-12-16:
rmm, now ive a doubt if its a ILI9328 or a ST7789V ... erk i need to double check

tpruvot, 2019-12-16:
so title is incorrect, ILI9328 was never supported in Marlin, only in MKS repo

sakza, 2019-12-20:
Ah, good point - any idea how straightforward it would be to implement? 
I'll give it a crack regardless and hopefully I'll make some progress

jmz52, 2019-12-22:
ili9328 is quite different from from st7789 and ili9341
Main problem is init sequence that requires 16-bit values while U8g only works with 8-bit U8G_ESC sequences.
There is an 11 months old code that support for ili9328 via custom 16-bit escape sequences - 
[MKS-Robin-Touch-POC](https://github.com/jmz52/Marlin/commits/MKS-Robin-Touch-POC)
Implementation details can be found in this commit - https://github.com/jmz52/Marlin/commit/0cc2feec66ca6886251260299e07c6c6ca2fd0c9

Also [touch screen calibration tool](https://github.com/jmz52/MKS-Robin-toolkit/tree/master/lcd_and_touch) was updated at about the same time to support ili9328.
Compiled firmware to be used with vendor's bootloader - [Touch_Screen_Calibration](https://github.com/jmz52/MKS-Robin-toolkit/blob/master/lcd_and_touch/Robin.bin)

jmz52, 2019-12-22:
Calibration data for MKS Robin TFT 1.1 (ili9328) are in commented section of [config/examples/Mks/Robin/Configuration.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Mks/Robin/Configuration.h)
`/* MKS Robin TFT v1.1 */`

Data source: https://imgur.com/a/pRdGEVa

jmz52, 2019-12-24:
@sakza I've pieced together some code for ili9328 but I have no hardware to test it.
Could you please check if this code works for your MKS Robin TFT v1.1 ?
https://github.com/jmz52/Marlin/tree/MKS-Robin-ILI9328

Two tests required, one with enabled
`#define LCD_USE_DMA_FSMC`
line in pins_MKS_ROBIN.h file and another one with disabled FSMC

Also please enable following line in pins_MKS_ROBIN.h for bother tests to validate initialization sequence for ILI9328.
`#define LCD_RESET_PIN      PF6`

sakza, 2019-12-24:
Awesome, I'm away at the moment, but I'll get back to you when I can - I've got it to the point where I can make it display a white screen instead of the bootloader text so anything last that will be a worthwhile!

mattdog01, 2019-12-31:
@jmz52 
First of all thank you so much for all your work getting the MKS-Robin working with Marlin.  Totally awesome. I don't mean to jump in and hijack your work with @sakza but I did the tests on my TFT v1.1 and it is the closest that I have been to getting Marlin working.  I get the Marlin home screen and then the status screen with #define LCD_USE_DMA_FSMC enabled and with it commented out.  I had #define LCD_RESET_PIN PF6 enabled for both tests. But the wonderful news is that the v1.1 tft  does display in Marlin.  Thank you.

I have a different issue that is preventing further progress and that my board goes into flashing blue LED mode (I think it is a protective mode).  This is likely the reason I can't get the touch buttons to work. Any thoughts there?  I have had this problem since the beginning of my Marlin adventure (but without the awesome Marlin screens working).

sakza, 2019-12-31:
Good timing @mattdog01, I finally was able to test it and I have 100% functionality @jmz52, including the touch screen buttons. I've absolutely butchered my marlin fork trying to get it working, so I really don't know what aspect of mine has helped the buttons, but I'd recommend having a look at my configuration at the top of the issue and see if anything is different regarding the touch screen

mattdog01, 2019-12-31:
@sakza 
Thanks I'll have a look at your config file. 
Any idea why my board is going into the error sensing / self protection mode (flashing blue LED)? It doesn't happen with stock firmware which makes me think it something in my settings. Thanks again to all of you for working on this.

sakza, 2019-12-31:
Not sure about that one, I don't think I've run into it at all - If you can't get it working, I'll put my whole fork back up on github so you can make sure it's not your hardware

jmz52, 2019-12-31:
@sakza
Flashing blue LED is a heartbeat, is indicates that Marlin is fine.

Default touch screen calibration data are for MKS Robin TFT 2.0 and they are not suitable for v1.1
There is a commented section in `config/examples/Mks/Robin/Configuration.h` with touch screen calibration data for v1.1

You need to disable/remove v2.0 values and use v1.1 values instead.
Alternatively you can use [calibration tool](https://github.com/jmz52/MKS-Robin-toolkit/blob/master/lcd_and_touch/Robin.bin) to get these values for your touch screen.
```
//
// ADS7843/XPT2046 ADC Touchscreen such as ILI9341 2.8
//
#define TOUCH_BUTTONS
#if ENABLED(TOUCH_BUTTONS)
  #define BUTTON_DELAY_EDIT  75 // (ms) Button repeat delay for edit screens
  #define BUTTON_DELAY_MENU 100 // (ms) Button repeat delay for menus

  /* MKS Robin TFT v2.0 */
  #define XPT2046_X_CALIBRATION    12013
  #define XPT2046_Y_CALIBRATION    -8711
  #define XPT2046_X_OFFSET           -32
  #define XPT2046_Y_OFFSET           256

  /* MKS Robin TFT v1.1 */
  //#define XPT2046_X_CALIBRATION -11792
  //#define XPT2046_Y_CALIBRATION   8947
  //#define XPT2046_X_OFFSET         342
  //#define XPT2046_Y_OFFSET         -19
```

jmz52, 2019-12-31:
And there is no need to enable `#define LCD_RESET_PIN PF6` anymore.
It was needed to test initialization sequences and now when we know that initialization works you can leave LCD_RESET_PIN disabled.

mattdog01, 2019-12-31:
@jmz52 @sakza 
Thanks for the help.  Assuming the flashing blue LED is the "heart beat" then the only issue I have is the touch feature doesn't work on the four buttons at the bottom of the display.  I have confirmed that I have the config files the same as provided in the examples, except where there is an obvious difference (eg bed size, personal preferences, etc.) Is there a thread that you can point me too to help sort this out?

3Dangel, 2019-12-31:
@jmz52 Man... you've away for so long!!!!
missed your "brain"......
again, THANK YOU for your talent and support for this old board! (mine was never used and wanted to use it)
will give a try and hope the main MARLIN could include your work in it!!!

sakza, 2019-12-31:
@mattdog01 I do remember a couple of other things I changed, although it's almost a month ago at this point - I'll do some digging when I've got some time and I'll see if I can find anything specific

3Dangel, 2019-12-31:
confirmed.
MKS-Robin V2.1 TFT V1.1 ili9328 processor is fully functional.
@sakza 
your config shows you are using the defaul one and not the one from the sample config folder for MKS
as jmz52 pointed out, you must use the right x,y coordinate for the bottoms to work:
  /* MKS Robin TFT v1.1 */
  #define XPT2046_X_CALIBRATION -11792
  #define XPT2046_Y_CALIBRATION   8947
  #define XPT2046_X_OFFSET         342
  #define XPT2046_Y_OFFSET         -19
#endif

mattdog01, 2019-12-31:
@sakza 
thank you so much for any help you can offer.  You have already given me hope that Marlin will one day be fully functional on my printer.  

@3Dangel 
I am not using the default config.  I'm using the one provided by @jmz52.  I even ran the screen config and my numbers are almost exactly the same.  My config is attached
[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/4013258/Config.files.zip)

mattdog01, 2019-12-31:
@3Dangel 
[image]
I think my numbers are close enough to just go with provided numbers.

3Dangel, 2019-12-31:
you missed
#define NUM_SERIAL 2

3Dangel, 2019-12-31:
[image]
[image]
[image]

3Dangel, 2019-12-31:
last picture shows my SD card's info.
SD cards used 2Gb. FAT

mattdog01, 2020-01-01:
@3Dangel 
Way to go eagle eye.  Thanks.  
> you missed #define NUM_SERIAL 2

It didn't help my situation though.  Still no touch function.  So sad.

Your screen looks awesome.  You must be happy.  
I'm glad that it will be able to see folders on the sd card.  That is how I like to arrange my stuff.

I'm wondering if you would be so kind as to compile my config files and send me the bin file?
I'm suspecting that is is my compilers environment.  Everyone seems to be able to get it figured out but I'm still stuck with dysfunctional touch buttons.

tpruvot, 2020-01-01:
try to click on the top of the screen, the v1.0 touchscreen orientation may be different than v1.1 one

3Dangel, 2020-01-01:
It will stop to work as soon as you start to enabling functions.
Mine stopped working when tried advanced pause for filament changes.
Or other staffs I have enabled.

Try the bare bones from the jmz52’s repository and without changing things
and try.

mattdog01, 2020-01-01:
@3Dangel 
@jmz52 
@tpruvot 

Hey Guys,
Happy New Year!!!
Thank you to all of you.
I allowed my VS code to update (I have been holding it at 1.39.2). So now I'm at Version: 1.41.1 (user setup) and PlatformIO home 3.0.1 and core 4.1.0.  This is just FYI as there are some out there that say the newer versions of vscode have caused some of the issues with compiling but I have not idea about those wonderful things. (I fix aircraft not code)

I tried the fresh config files as provided from jmz52’s repository without any changes.
The buttons worked but at the top of the screen and in a different order. I was able to navigate the menus and perform different functions.  Victory at last!!!!!!

I went into the the config files and commented the v2.0 defaults and uncommented the v1.1 defaults.
the buttons and image where together at last and worked correctly.

Then I went in and configured for my machine. corexy, x, y and e directions, accelerations and speeds.
Worked great, homed.

"No media" was display in the menus.
Changed the advanced features to add a feature to the menus to check the SD card and now I can read the card, navigate the folders and select jobs to print.

Added fun features like baby stepping and tried to add filament change but I'm still working on that.

I still need to learn where everything is normally located in the menus.

I have some replacement parts printing now and everything seems to be working perfectly.

I think I will be able to increase movement speeds now that I have better more accurate control over them (as an example in the stock firmware I needed to double the steps per mm to get the travel correct)

[image]
[image]

3Dangel, 2020-01-02:
Very happy to see another successful revival of the Robin with TFT on
Marlin!!!

@jmZ52 again thank you for your
The great coding and sharing it with us.

Also to every contributors of Marlin, big thanks and best wishes and
greater success in 2020.

I believe in open source and people around the world will continue to
contribute and push the boundary of Marlin FW and make it to be even
greater !

tpruvot, 2020-01-03:
@boelle issue can be closed i think

mattdog01, 2020-01-03:
> @boelle issue can be closed i think

@tpruvot  I think you are right sir.  Marlin 2.0 is working perfectly on my MKS-Robin v2.3 with the TFT v1.1.  The ILI9328 TFT works great on the MKS Robin.

I'm so thankful that you guys are support the Marlin project.  I can't do it so it is great to have you doing it.  Thank you.  If you ever need any help with MKS-Robin v2.3 with the TFT v1.1 firmware testing I'll be happy to help.  I would like to give back.

kindest regards,
Matt and his Hypercube

sakza, 2020-01-03:
Thanks everyone for the input and hard work! Looks like everything is working 100%, and merged!

closed 2020-01-03 by sakza