MarlinFirmware/Marlin #16160: [BUG] ignore inserted SDCARD on boot

karabas2011, 2019-12-09:
### Bug Description

When I boot printer with sdcard inserted to LCD 
Marlin shows no sdcard. And I need to eject and insert to make it visible.
On octobers snapshots all works correctly. 

### My Configurations

Marlin 2.0 29-nov-2019
SKR1.3 LCD MKS  MINI12864 v2

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1 Power off printer

2. Insert SDCARD
3. Power on printer

**Expected behavior:** [What you expect to happen]

Sdcard  exists in menu and I am able to choose files.

**Actual behavior:** [What actually happens]

No sdcard 

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3938422/Marlin.zip)

J-charles-C, 2019-12-10:
I,

You can choose between Motherboard or LCD Sdcard here : (configurationadv.h)

#if HAS_SDCARD_CONNECTION
    /**
     * Set this option to one of the following (or the board's defaults apply):
     *
     *           LCD - Use the SD drive in the external LCD controller.
     *       ONBOARD - Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
     *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
     *
     * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
     */
    **//#define SDCARD_CONNECTION LCD**

Best

karabas2011, 2019-12-10:
I uncomment
#define SDCARD_CONNECTION LCD
no changes. No sdcard after booting.

borland1, 2019-12-15:
This is similar to reported bug (issue #16233), except it was for different controller board.  Pin assignment of SD_DETECT_PIN was not assigned in pin mapping file for the specific board.

For your SKR 1.3 board with MKS Mini 12864 display, file named pins_BTT_SKR_V1_3.h, lines 271-277,
could be modified as follows:

    #if ENABLED(MKS_MINI_12864)
       #define SD_DETECT_PIN P1.31  // add this line to map display's SD-DET pin to processor
       #define DOGLCD_CS    P1_21
       #define DOGLCD_A0    P1_22
       #define DOGLCD_SCK   P0_15
       #define DOGLCD_MOSI  P0_18
       #define FORCE_SOFT_SPI
    #endif
 
Then card should be recognized after enabling #define SDCARD_CONNECTION LCD in Configuration_adv.h file.

karabas2011, 2019-12-16:
I have #define SD_DETECT_PIN P1.31 and it works when eject-insert.
And it works on hot reset button press.
CDdetect does Not work only on power on booting.

borland1, 2019-12-16:
Marlin firmware does not configure the bootloader.  The bootloader only looks at the ONBOARD SD-Slot for a firmware update. If it does not find a firmware.bin file there, it will try to boot any firmware previously loaded.

For your SKR board, the wiring diagram shows no SD_DETECT_PIN being wired to the ONBOARD SD-slot card.

Since the SKR bootloader is not a Marlin firmware feature, you should consider closing this report.

karabas2011, 2019-12-16:
Why october snapshot works?

shitcreek, 2019-12-20:
For the BTT TFT 24 v1.1 which is a MKS MINI12864, use **#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER** instead of **#define MKS_MINI_12864**. Plus **#define SDCARD_CONNECTION LCD**.

thisiskeithb, 2019-12-20:
> BTT TFT 24 v1.1 which is a MKS MINI12864

Can you clarify this? BigTreeTech has always said to use `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` for their combo 12864/TFT displays.

shitcreek, 2019-12-20:
Perhaps I should have said it is essentially the mks mini12864.

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle