MarlinFirmware/Marlin #16161: [BUG] Runout sensor requires manual turn on after triggering

karabas2011, 2019-12-09:
### Bug Description

I use Runout sensor  ( actually 2 in 1, runout and spool loop)
During printing when first triggering I see menu 
where I choose extrude more or turn sensor ON. 
I press On and it continues printing. But sensor now is actually OFF. And no second triggering is possible until I go to main menu settings and turn on the sensor!

### My Configurations
Marlin 2.0 29-nov-2019
skr1.3 lcd MKS mini 12864 V2
TMC2208

### Steps to Reproduce

1. Start print
2. force runout to trigger
3. continui
4  force runout to trigger

**Expected behavior:** [What you expect to happen]
Pause as many times as it triggers

**Actual behavior:** [What actually happens]

Paused only first triggering
So menu item - turn on sensor - does not do anything. Or does not apply.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3938478/Marlin.zip)

 
#### Additional Information
Yes I can hack it to insert function to menu that will  autoturn on sensor 

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2019-12-09:
That menu item toggles the state if the sensor is currently off. It probably needs a condition added to either never or always show it next to continue for motion monitoring sensors, since input state cannot be used directly there.

karabas2011, 2019-12-10:
My hack. Is it correct?
```
void menu_pause_option() {
  START_MENU();
  #if LCD_HEIGHT > 2
    STATIC_ITEM(MSG_FILAMENT_CHANGE_OPTION_HEADER);
  #endif
  ACTION_ITEM(MSG_FILAMENT_CHANGE_OPTION_PURGE, [](){ pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE; });
  #if HAS_FILAMENT_SENSOR
    // if (runout.filament_ran_out)
      runout.enabled = true;
      runout.reset();
      EDIT_ITEM(bool, MSG_RUNOUT_SENSOR, &runout.enabled, runout.reset);
   // else
  #endif
      ACTION_ITEM(MSG_FILAMENT_CHANGE_OPTION_RESUME, [](){ pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT; });
  END_MENU();
} 
```

karabas2011, 2019-12-10:
I think it is old bug. M119 is not in sync with  FIL_RUNOUT_INVERTING, and the whole runout code behaviouring is inverted.

InsanityAutomation, 2019-12-10:
Its going to need a bit more than that as the extui handler needs updating as well to take 3 potential responses for runout in case of the encoder style and it needs to be set for only the encoder style.

karabas2011, 2019-12-11:
No encoder at all. Just one endstop

karabas2011, 2019-12-16:
```
  #if HAS_FILAMENT_SENSOR
    if (runout.filament_ran_out) { 
      runout.enabled = false;
      EDIT_ITEM(bool, MSG_RUNOUT_SENSOR0, &runout.enabled, runout.reset); 
      }
    else {
      runout.enabled = true;
      runout.reset();
   //   EDIT_ITEM(bool, MSG_RUNOUT_SENSOR, &runout.enabled, runout.reset);
    }
  #endif
```

tpruvot, 2019-12-16:
see PR #16230 it should fix that

karabas2011, 2019-12-17:
Thanx! It fixes problem with start printing pause ( when I forget to close sensor ) 
But for my type of sensor I need menu  to prevent resume until I close the sensor.
this is what my code currently does.

karabas2011, 2019-12-23:
New problem
Menu Resume does not reflect off-state of  sensor until you rotate encoder!

boelle, 2020-01-07:
seems this one was solved, if a new issues had surfaced create a new PR for that

if this issue is not solved we can reopen

closed 2020-01-07 by boelle