MarlinFirmware/Marlin #16164: [BUG] Long list of errors during compile Marlin 2.0 for Anet A2 in Arduino IDE

dzamosenchuk, 2019-12-09:
Hello!
I read your manual for compile Marlin in Arduino IDE and tried to made it.
My steps:
1. Download actual Marlin 2.0.
2. Unzip
3. Replace Configuration files to files from "example/Anet/Anet A2 plus"
4. Download last version of Arduino IDE. 1.8.10
5. Install it
6. Download Anet board definition from https://github.com/SkyNet3D/anet-board
7. Unzip it to "Documents/Arduino/hardware"
8. Open Marlin.ino in Arduino IDE.
9. Chose Anet V1.0 in Tools tab.
10. Try to compile (Ctrl+R)
Have errors:
`Arduino: 1.8.10 (Windows 10), Board: "Anet V1.0"
In file included from \AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\cores\arduino/Arduino.h:258:0,
                 from sketch\Marlin.ino.cpp:1:
\Documents\Arduino\hardware\anet\avr\variants\sanguino/pins_arduino.h:43:0: warning: "TIMER2" redefined
 #define TIMER2  5

In file included from C:\Users\Damon\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\cores\arduino/Arduino.h:258:0,

                 from c:\users\damon\appdata\local\temp\arduino_build_308824\sketch\src\hal\shared\marduino.h:36,

                 from c:\users\damon\appdata\local\temp\arduino_build_308824\sketch\src\hal\hal_avr\hal.h:21,

                 from c:\users\damon\appdata\local\temp\arduino_build_308824\sketch\src\hal\hal.h:26,

                 from sketch\src\inc/MarlinConfig.h:30,

                 from sketch\src\Marlin.h:24,

                 from sketch\src\Marlin.cpp:31:

C:\Users\Damon\Documents\Arduino\hardware\anet\avr\variants\sanguino/pins_arduino.h:43:0: warning: "TIMER2" redefined

 #define TIMER2  5

 

In file included from c:\users\damon\appdata\local\temp\arduino_build_308824\sketch\src\hal\shared\marduino.h:36:0,

                 from c:\users\damon\appdata\local\temp\arduino_build_308824\sketch\src\hal\hal_avr\hal.h:21,

                 from c:\users\damon\appdata\local\temp\arduino_build_308824\sketch\src\hal\hal.h:26,

                 from sketch\src\inc/MarlinConfig.h:30,

                 from sketch\src\Marlin.h:24,

                 from sketch\src\Marlin.cpp:31:

C:\Users\Damon\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\cores\arduino/Arduino.h:211:0: note: this is the location of the previous definition

 #define TIMER2  6

 

In file included from c:\users\damon\appdata\local\temp\arduino_build_308824\sketch\src\pins\pins.h:71:0,

                 from c:\users\damon\appdata\local\temp\arduino_build_308824\sketch\src\inc\marlinconfig.h:32,

                 from sketch\src\HAL\HAL_AVR\HAL_SPI.cpp:34:

c:\users\damon\appdata\local\temp\arduino_build_308824\sketch\src\pins\ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"

     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"

      ^~~~~

[... 234 lines cut ...]
           disable_Z();

           ^~~~~~~~~

exit status 1
Error compiling for board Anet V1.0.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
I have the same errors (about TIMER) with Marlin 1.1.x, 1.1.9, bugfix etc. Tried use Arduino 1.6 and 1.8.0 and 1.8.5
Similar issues: https://github.com/MarlinFirmware/Marlin/issues/10995
https://github.com/MarlinFirmware/Marlin/issues/11783
If I compile Marlin 2.0 bugfix I have next error:
`Arduino: 1.8.10 (Windows 10), Board: "Anet V1.0"
sketch\src\HAL\HAL_AVR\u8g_com_HAL_AVR_sw_spi.cpp:65:10: fatal error: U8glib.h: No such file or directory
 #include <U8glib.h>
exit status 1
Error compiling for board Anet V1.0.
`
I added U8glib by library manager. Version 1.19.1 by oliver.
And try compile again. And got errors about "TIMER" and etc. described high.
[errors log.txt](https://github.com/MarlinFirmware/Marlin/files/3939559/errors.log.txt)
Full log in attached file

boelle, 2019-12-09:
is the full log txt the same as in the issue?

dzamosenchuk, 2019-12-09:
short part of log In the body. Log is very long

boelle, 2019-12-09:
have you tried arduino ide 1.9 beta? or vscode+platfirm.io extension ?

boelle, 2019-12-09:
i took the freedom and moved the errors in the issue to pastebin so the post is shorter

dzamosenchuk, 2019-12-09:
I've tried 1.9 beta with Marlin 1.1.x and 1.1.x bugfix. And got errors about timer.

boelle, 2019-12-09:
try bugfix 2.0.x first :-D

and if you need a video on how to install vs code here it is: https://youtu.be/bPdZvv8jJJ0?t=196

boelle, 2019-12-09:
btw. title said 2.0?

boelle, 2019-12-09:
good thing with vs code is that it downloads most of thing for you

dzamosenchuk, 2019-12-09:
I've installed VS Code with PlatformIO. But I don't know how to add definition of my board.

boelle, 2019-12-09:
you just open the marlin folder

boelle, 2019-12-09:
you dont have to add anything

boelle, 2019-12-09:
[image]

boelle, 2019-12-09:
[image]

boelle, 2019-12-09:
do not pick the marlin folder here, you need to see it like this or else you are a level to deep

dzamosenchuk, 2019-12-09:
Arduino IDE 1.8.10 Marlin 2.0 bugfix errors in log file
[New Text Document (2).txt](https://github.com/MarlinFirmware/Marlin/files/3939665/New.Text.Document.2.txt)

boelle, 2019-12-09:
should look like this when folder is open

[image]

boelle, 2019-12-09:
if you edit platformio.ini at the top you will see

[image]

default_envs has to one of these 3 but i dont know which

[image]

dzamosenchuk, 2019-12-09:
Thanks a lot of your advice. I'll try it!

boelle, 2019-12-09:
i tried to google and found a pic of the board and the chip is 1284p

i dont use arduino ide at all and my board is the re-arm so i update firmware with the sd card slot directly

but vs code should be able to upload firmware with usb (i think)

thisiskeithb, 2019-12-09:
> Download Anet board definition from https://github.com/SkyNet3D/anet-board

What’s wrong with the Marlin’s stock board definitions?

boelle, 2019-12-09:
> What’s wrong with the Marlin’s stock board definitions?

for arduino ide i think, not for marlin

but i guided him to vscode as it take care of 99% of things for him so he dont have to bother so much

thisiskeithb, 2019-12-09:
> for arduino ide i think, not for marlin

Oh, that makes more sense. I thought they were talking about the board pins. 🙂

boelle, 2019-12-09:
nope :-D

just so it is in the boards list in arduino IDE

dzamosenchuk, 2019-12-09:
> i tried to google and found a pic of the board and the chip is 1284p
> 
> i dont use arduino ide at all and my board is the re-arm so i update firmware with the sd card slot directly
> 
> but vs code should be able to upload firmware with usb (i think)

Changed in platforio.ini 
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = sanguino_atmega1284p

Try build:
rocessing sanguino_atmega1284p (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sanguino_atmega1284p.html
PLATFORM: Atmel AVR 2.0.0 > Sanguino ATmega1284p (16MHz)
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduino-avr 5.0.0
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 16 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.1
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.2.5        
|-- <Adafruit MAX31865 library> 1.0.2
|   |-- <SPI> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <TMC26XStepper>
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio\build\sanguino_atmega1284p\src\src\HAL\HAL_AVR\HAL.cpp.o
Compiling .pio\build\sanguino_atmega1284p\src\src\HAL\HAL_AVR\HAL_SPI.cpp.o
Compiling .pio\build\sanguino_atmega1284p\src\src\HAL\HAL_AVR\MarlinSerial.cpp.o
Compiling .pio\build\sanguino_atmega1284p\src\src\HAL\HAL_AVR\Servo.cpp.o
In file included from Marlin\src\HAL\HAL_AVR\../../inc/../pins/pins.h:71:0,
                 from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\HAL_AVR\HAL.cpp:25:
Marlin\src\HAL\HAL_AVR\../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
In file included from Marlin\src\HAL\HAL_AVR\../../inc/../pins/pins.h:71:0,
                 from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\HAL_AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\HAL_AVR\../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
In file included from Marlin\src\HAL\HAL_AVR\../../inc/../pins/pins.h:71:0,
                 from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\HAL_AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\HAL_AVR\../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
[... 40 lines cut ...]
STM32F4                       IGNORED
STM32F7                       IGNORED
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_lite                IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
include_tree                  IGNORED
======================================================================================== 1 failed, 0 succeeded in 00:00:05.246 ======================================================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

boelle, 2019-12-09:
> if you edit platformio.ini at the top you will see
> 
> [image]
> 
> default_envs has to one of these 3 but i dont know which
> 
> [image]

dzamosenchuk, 2019-12-09:
> > if you edit platformio.ini at the top you will see
> > [image]
> > default_envs has to one of these 3 but i dont know which
> > [image]

Yep. I saw your comment. Google said for Anet need to use sanguino_atmega1284p. But It didn't build

boelle, 2019-12-09:
oki, someone more smart than us will prop chime in with a hint

boelle, 2019-12-09:
what board have you set in marlin configuration.h ?

dzamosenchuk, 2019-12-09:
anyway thanks a lot of your help!

dzamosenchuk, 2019-12-09:
> what board have you set in marlin configuration.h ?

Anet A2plus

dzamosenchuk, 2019-12-09:
took it from configuration example

boelle, 2019-12-09:
example says this

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_ANET_10
#endif

https://github.com/MarlinFirmware/Marlin/blob/2.0.x/config/examples/Anet/A2/Configuration.h#L131

thisiskeithb, 2019-12-09:
Try with `default_envs = melzi` instead.

dzamosenchuk, 2019-12-10:
> Try with `default_envs = melzi` instead:
> 
> https://github.com/MarlinFirmware/Marlin/blob/39fcb5b5f891cea26b1e8ab0d4b749565e830582/Marlin/src/pins/sanguino/pins_ANET_10.h#L93

> Executing task in folder Marlin-2.0.x: C:\Users\Damon\.platformio\penv\Scripts\platformio.exe run <

Processing melzi (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sanguino_atmega1284p.html
PLATFORM: Atmel AVR 2.0.0 > Sanguino ATmega1284p (16MHz)
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduino-avr 5.0.0
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing U8glib-HAL
Using cache: C:\Users\Damon\.platformio\.cache\a7\b7a9b473d1c239d9843ca943887cd0a7
U8glib-HAL @ 0.4.1 has been successfully installed!
Looking for LiquidCrystal library in registry
Conflict: More than one library has been found by request {"name": "LiquidCrystal", "requirements": "1.3.4"}:
LiquidCrystal
Automatically chose the first available library (use `--interactive` option to make a choice)=============

#ID: 136
LiquidCrystal Library is faster and extensable, compatible with the original LiquidCrystal library

Keywords: lcd, hd44780
Compatible frameworks: Arduino
Compatible platforms: Atmel AVR, Espressif 32, Espressif 8266
Authors: F Malpartida

LiquidCrystal
=============
#ID: 887
Allows communication with alphanumerical liquid crystal displays (LCDs).

Keywords: display
Compatible frameworks: Arduino
Compatible platforms: Atmel AVR, Atmel SAM, Espressif 32, Espressif 8266, Intel ARC32, Microchip PIC32, Nordic nRF51, Nordic nRF52, ST STM32, Teensy, TI MSP430
Authors: Arduino, Adafruit

Found: https://platformio.org/lib/show/136/LiquidCrystal
LibraryManager: Installing id=136 @ 1.3.4
Using cache: C:\Users\Damon\.platformio\.cache\56\e51065281cd05bea572117c1aab6cb56
LiquidCrystal @ 1.3.4 has been successfully installed!
Looking for TMCStepper library in registry
Found: https://platformio.org/lib/show/5513/TMCStepper
LibraryManager: Installing id=5513 @ >=0.5.2,<1.0.0
Using cache: C:\Users\Damon\.platformio\.cache\4a\00c7fd7875e9cd0a4cf8d7b2771b3c4a
TMCStepper @ 0.6.1 has been successfully installed!
Looking for Adafruit NeoPixel library in registry
Found: https://platformio.org/lib/show/28/Adafruit NeoPixel
LibraryManager: Installing id=28 @ 1.2.5
Adafruit NeoPixel @ 1.2.5 is already installed
LibraryManager: Installing Adafruit_MAX31865
Adafruit_MAX31865 @ 1.0.2 is already installed
LibraryManager: Installing LiquidTWI2
LiquidTWI2 @ 0.0.0 is already installed
LibraryManager: Installing Arduino-L6470
Arduino-L6470 @ 0.7.0 is already installed
LibraryManager: Installing SailfishLCD
SailfishLCD @ 0.0.0 is already installed
LibraryManager: Installing SailfishRGB_LED
SailfishRGB_LED @ 0.0.0 is already installed
LibraryManager: Installing SlowSoftI2CMaster
SlowSoftI2CMaster @ 0.0.0 is already installed
LibraryManager: Installing TMC26XStepper
TMC26XStepper @ 0.0.0 is already installed
Found 15 compatible libraries
[... 82 lines cut ...]
STM32F4                       IGNORED
STM32F7                       IGNORED
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_lite                IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
include_tree                  IGNORED
======================================================================================== 1 failed, 0 succeeded in 00:00:20.286 ======================================================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

Vertabreak, 2019-12-10:
U8glib wasn't installed its clearly listed in the error log and 90% of it is just warnings. the second error log is all warnings and at the end it shows compiled. suggest going to preferences and selecting warning = none all warnings should be ignored if it compiled its fine.

boelle, 2019-12-10:
will close this one as it likely a user error and not a bug with marlin

closed 2019-12-10 by boelle

thisiskeithb, 2019-12-11:
I just tried the default [A2 plus configs](https://github.com/MarlinFirmware/Marlin/tree/07a61057ba592769e8e1d7275ca2f19a2d971ab7/config/examples/Anet/A2plus) & using `default_envs = melzi` works fine.

@dzamosenchuk Clean your project files by clicking on the trash icon in the bottom-left of VSCode or manually delete `.pio` folder and build again.

[image]

dzamosenchuk, 2019-12-11:
Thank everybody, who helped in my case!
@Vertabreak was right! I've added U8glib, checked preferences (Warnings was in "none" state) and try to compile. It done.
I think need to add this information to Documentation. Because it is not clear for newbie. 
@thisiskeithb I made cleaning, but it didn't help. I have the same errors.