MarlinFirmware/Marlin #16165: [QUESTION] 1.1.x and bugfix-1.1.x slowly drifting away from each other?

prof7bit, 2019-12-09:
After the release of 1.1.9 there have been dozens of 1.1 fixes committed, half of them to the 1.1.x branch, half of them to the bugfix-1.1.x branch. Is there the intention to ever merge these two branches into one branch again so there will remain only one 1.1.x fixes branch from now on or will there continue to be two separate branches slowly drifting away from each other?

Which branch is recommended to use when one wants to have the latest and greatest post 1.1.9 build, including all fixes since 1.1.9? Do I have to merge these branches myself to receive a complete set of all available 1.1.x fixes?

boelle, 2019-12-09:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-12-09 by boelle

prof7bit, 2019-12-09:
> This Issue Queue is for Marlin bug reports and development-related issues

This is a development related issue! Where are fixes for 1.1.x supposed to be committed? There seem two branches for it and nobody seems to know which one is the correct one, half of them commit to the 1.1.x and the other half to the bugfix branch!

boelle, 2019-12-09:
there is only one not 2

bugfix-1.1.x

boelle, 2019-12-09:
btw  i see no PR send to 1.1.x but only 1 or 2 sent to bugfix-1.1.x the rest is sent to bugfix-2.0.x

prof7bit, 2019-12-09:
Then what about the 64 unmerged commits in 1.1.x since 1.1.9? Will they be
merged into bugfix-1.1.x eventually or have they been forgotten?

boelle, 2019-12-09:
double checking now....

boelle, 2019-12-09:
there are 33 open PR's and i have checked 25 so far

boelle, 2019-12-09:
all except 2 PR's are sent to bugfix-2.0.x

boelle, 2019-12-09:
i checked all once more

only https://github.com/MarlinFirmware/Marlin/pull/16141

goes to bugfix-1.1.x

boelle, 2019-12-09:
list is here: https://github.com/MarlinFirmware/Marlin/pulls?page=1&q=is%3Apr+is%3Aopen

prof7bit, 2019-12-09:
I am not talking about open pull request but about the commits in both branches **that have already been made** since 1.1.9.

Here is the list for 1.1.x:
````
git log  --pretty=format:"%h%x09%x09%ad%x09%s" 624fd8c9ff71f114329dc6da69a6909ab568afc2..origin/1.1.x
f4bf94f7b		Sat Nov 30 09:10:33 2019 -0600	Links for the "New Issue" page
014ec3e81		Sun Nov 3 21:04:33 2019 +0000	Update bug_report.md
0110faa71		Sun Aug 25 04:49:49 2019 -0500	Fix Creality bed thermistor
909c01554		Fri Aug 23 22:59:02 2019 -0300	Fix PT-BR strings (#15023)
f3faaed24		Wed Aug 21 05:45:29 2019 -0300	Fix #else extra tokens (#15013)
e5a4f19e5		Tue Aug 6 23:34:55 2019 -0500	[1.1.x] Autobuild formatting (#14858)
37bfc5ab9		Wed Aug 7 00:06:24 2019 -0400	[1.1.x] BLTouch 3.0 - 3.1 (#14839)
1ab59d6cd		Mon Aug 5 22:55:49 2019 -0500	Revert "move PR #14857 Autobuild update into bugfix 1.1.x"
fb276f677		Mon Aug 5 22:53:05 2019 -0500	move PR #14857 Autobuild update into bugfix 1.1.x
65ff80411		Mon Apr 15 17:33:11 2019 -0400	[1.1.x] BLTOUCH_V3 parity with 2.0.x (#13682)
a5fe0ed15		Sat Apr 6 19:53:33 2019 -0500	Reduce XY homing feedrate in Creality configs
f73829318		Sun Apr 7 02:48:44 2019 +0200	Add support for the V3 of the BlTouch (#13596)
5f783b4fe		Wed Mar 20 22:38:43 2019 +0100	[1.1.x] Fix German language (#13438)
8d9380be7		Mon Mar 18 22:32:03 2019 +0100	Fix missing quotation marks (#13424)
cc6d7812c		Sat Mar 16 03:45:03 2019 +0330	Fix pin assignment to KILL_PIN in MKS_MINI_12864 (#13397)
9180235ad		Fri Mar 8 19:53:22 2019 -0600	German patch
6a45d52bb		Sun Mar 3 05:37:15 2019 -0600	Remove dead code in get_pid_output, consolidate
2ecd75c14		Wed Feb 27 17:26:21 2019 +1300	UBL G29 T current position fix  (#12845)
9ede2cfe0		Fri Feb 22 22:49:01 2019 -0300	Fix spanish translations (#13235)
8899b5804		Fri Feb 22 22:29:38 2019 +0100	[1.1.x] Fix autotemp_factor menu setting (#13227)
951ed3fd6		Wed Feb 20 05:33:32 2019 -0600	Fix PSU pin initialization
7aedf18fd		Thu Feb 14 03:50:11 2019 +0100	Optional ADC_KEYPAD pin with ANET_10 (#13158)
dea37780c		Thu Feb 14 03:43:39 2019 +0100	Fix LCD button / newbutton issue (#13157)
1f0fe23c5		Thu Feb 14 03:31:34 2019 +0100	[1.1.x] Update German language (#13149)
93c3d4b33		Tue Feb 12 19:54:11 2019 +0100	[1.1.x] Update German language (#13136)
9c5eebf65		Tue Feb 12 00:10:56 2019 +0100	[1.1.x] Fix TEMP_STAT_LED startup init (#13121)
ffc424f55		Wed Feb 6 02:56:53 2019 -0200	Fix language_pt-br warnings due to extra tokens at end of #else directive (#13084)
46daed8b0		Tue Feb 5 20:27:58 2019 -0700	Fix probe location bug....
6960cd8e2		Wed Feb 28 16:54:09 2018 -0600	Apply lowercase to some common methods
ca36be46d		Thu Jan 24 09:43:26 2019 +0800	Fix the delta homing issue (#13001)
a659f55cf		Fri Jan 11 02:25:02 2019 +0100	[1.1.x] Level bed corners Z-hop height option (#12859)
658f62e8b		Sun Dec 30 17:42:06 2018 +0100	[1.1.x] Correct Ender 3 bed size (#12768)
544990604		Thu Dec 27 18:43:07 2018 +0100	[1.1.x] Limit PID autotune target to maxtemp-15 (#12703)
3e459a071		Thu Dec 27 18:40:34 2018 +0100	[1.1.x] Limit hotend / bed temperature to maxtemp-15. (#12702)
2a2c6c2d6		Thu Dec 20 22:32:24 2018 +0100	[1.1.x] Update Czech translation (#12684)
ef2f09fd2		Tue Dec 18 20:35:35 2018 +0100	[1.1.x] Set DEFAULT_NOMINAL_FILAMENT_DIA to 1.75 in Anet A6 config (#12663)
e99a1fb46		Tue Dec 18 20:34:25 2018 +0100	[1.1.x] Enable PIDTEMPBED in Anet A8 example config (#12654)
eddb785be		Tue Dec 18 19:44:17 2018 +0100	[1.1.x] G33 clean up (#12649)
d7e82b9b8		Mon Nov 19 02:17:36 2018 -0600	Fix MBL error messages
97da93fbb		Sun Nov 18 18:06:14 2018 -0600	Fix kill with MSG_POWER_LOSS_RECOVERY
eb0b685c5		Mon Nov 5 03:42:24 2018 -0600	Fix MBL XY parameter error messages
d1cf029de		Tue Sep 4 00:47:55 2018 -0500	Rename to match case
c63d466fd		Sun Oct 28 00:19:51 2018 +0200	Update French language (#12246)
c7ed706f4		Fri Oct 26 03:52:37 2018 -0500	Fix endstops.monitor_flag
9ea9fde49		Tue Oct 23 01:52:39 2018 +0200	Init Hangprinter D-axis, count D-axis advance_dividend, delta_offsets (#12181)
83fe07d80		Mon Oct 22 00:25:46 2018 -0500	One more IS_SD_PRINTING
6110a5463		Fri Oct 19 12:23:46 2018 -0700	Add M524 to abort SD printing (#11386)
cee63a168		Fri Oct 19 14:19:51 2018 -0500	Add parentheses to card macros
68464005c		Fri Oct 19 20:47:27 2018 +0200	Fix hole to hole rod length for Plus model (#12157)
e443ebb98		Tue Oct 16 17:15:19 2018 -0500	Don't use ENABLED with *INVERTING
3dbe53283		Tue Oct 16 08:56:36 2018 +0200	Anycubic Kossel Z probe invert fix (#12101)
1a5f2fa2c		Tue Oct 16 08:51:14 2018 +0200	[1.1.x] Reduce HOMING_FEEDRATE_XY for Ender 3 (#12104)
9959a91ec		Fri Oct 12 17:04:33 2018 -0500	Fix lite status screen SD %
0c74e47a3		Mon Oct 8 22:29:48 2018 -0500	EINSY_RETRO uses same delays as EINSY_RAMBO
1321941d0		Sat Oct 6 21:53:45 2018 -0500	Safer delays for EINSY_RAMBO + RRDFGSC
aa9e4687d		Thu Oct 4 20:26:07 2018 -0500	Fix Anycubic Kossel config
c32d2e3ae		Thu Oct 4 14:36:19 2018 -0500	Fix lightweight UI hotends display
f94c3210b		Thu Aug 8 05:45:33 2019 -0500	Create FUNDING.yml
340fb15d2		Fri Jul 19 00:12:05 2019 -0500	Fix flattr link
5d96a6d98		Thu Apr 4 00:11:23 2019 -0500	No Support Requests
5befb3070		Fri Feb 22 15:12:58 2019 -0600	Update issue templates
24b7070bb		Sun Feb 10 14:32:52 2019 -0600	CR10mini E steps of 95
3a70028d5		Fri Nov 9 15:23:38 2018 -0600	Update README.md
5cf6c44f4		Sat Sep 22 03:43:32 2018 -0400	Version 1.1.9
````
[... 48 lines cut ...]
[... cut ...]
5e0dffd9a		Fri Oct 12 17:04:33 2018 -0500	Fix lite status screen SD %
1e57b1351		Mon Oct 8 22:29:48 2018 -0500	EINSY_RETRO uses same delays as EINSY_RAMBO
b33463aa6		Sat Oct 6 21:53:45 2018 -0500	Safer delays for EINSY_RAMBO + RRDFGSC
385c990d6		Thu Oct 4 20:26:07 2018 -0500	Fix Anycubic Kossel config
ae4808046		Thu Oct 4 14:36:19 2018 -0500	Fix lightweight UI hotends display

````

These branches are diverging and nobody is ever going to merge them back into one again because there will be no more 1.1.x release!

boelle, 2019-12-09:
@thinkyhead is the maintainer

he will answer when he have time, and he is also the only one doing commits

ManuelMcLure, 2019-12-09:
If you actually look at the lists that you created, you'll see that most of the listed commits are in both - they have different commit IDs but the same comment. If you actually compare the commit messages you can see that other than documentation and website fixes there are no commits in 1.1.x that are not also in bugfix-1.1.x.
Here's the list of commits from your lists that are in 1.1.x but not in bugfix-1.1.x:
* Create FUNDING.yml
* Fix flattr link
* Links for the "New Issue" page
* No Support Requests
* Update bug_report.md
* Update issue templates
* Update README.md
* Version 1.1.9

ManuelMcLure, 2019-12-09:
In any case, PRs should _always_ be done against the bugfix branch.