MarlinFirmware/Marlin #16166: [BUG] Cant compile with backlash feature

italocjs, 2019-12-09:
Hello, ive been trying to compile my firmware (version 2.0.0 stable which is working well with the backlash disable) but when i enable it i get some errors that i cant really understand.  any help is apreciated.

Features i'm using that i turned on/off to test, but did no effect:
-s curve
-jerk

Features i cant really turn off
-dual stepper / dual endstop for Y

I have no probe (like bl touch or servo), just a endstop on Xmin, Ymax and Zmax

Error:

```
C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\italo\\AppData\\Local\\Temp\\arduino_build_753782\\sketch\\src\\feature\\cancel_object.cpp" -o "C:\\Users\\italo\\AppData\\Local\\Temp\\arduino_build_753782\\sketch\\src\\feature\\cancel_object.cpp.o"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\italo\\AppData\\Local\\Temp\\arduino_build_753782\\sketch\\src\\feature\\caselight.cpp" -o "C:\\Users\\italo\\AppData\\Local\\Temp\\arduino_build_753782\\sketch\\src\\feature\\caselight.cpp.o"
In file included from c:\users\italo\appdata\local\temp\arduino_build_753782\sketch\src\hal\hal_avr\hal.h:23:0,

                 from c:\users\italo\appdata\local\temp\arduino_build_753782\sketch\src\hal\hal.h:26,

                 from c:\users\italo\appdata\local\temp\arduino_build_753782\sketch\src\inc/MarlinConfig.h:30,

                 from c:\users\italo\appdata\local\temp\arduino_build_753782\sketch\src\marlin.h:24,

                 from c:\users\italo\appdata\local\temp\arduino_build_753782\sketch\src\module\planner.h:33,

                 from C:\Users\italo\AppData\Local\Temp\arduino_build_753782\sketch\src\feature\backlash.h:25,

                 from C:\Users\italo\AppData\Local\Temp\arduino_build_753782\sketch\src\feature\backlash.cpp:27:

c:\users\italo\appdata\local\temp\arduino_build_753782\sketch\src\hal\hal_avr\fastio.h:63:36: error: pasting "DIO" and "-" does not give a valid preprocessing token

 #define _READ(IO)             TEST(DIO ## IO ## _RPORT, DIO ## IO ## _PIN)

                                    ^

c:\users\italo\appdata\local\temp\arduino_build_753782\sketch\src\hal\hal_avr\fastio.h:95:31: note: in expansion of macro '_READ'

 #define READ(IO)              _READ(IO)

                               ^~~~~

C:\Users\italo\AppData\Local\Temp\arduino_build_753782\sketch\src\feature\backlash.cpp:129:29: note: in expansion of macro 'READ'

     #define TEST_PROBE_PIN (READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING)

                             ^~~~

C:\Users\italo\AppData\Local\Temp\arduino_build_753782\sketch\src\feature\backlash.cpp:137:80: note: in expansion of macro 'TEST_PROBE_PIN'

     while (current_position.z < (start_height + BACKLASH_MEASUREMENT_LIMIT) && TEST_PROBE_PIN)

                                                                                ^~~~~~~~~~~~~~

c:\users\italo\appdata\local\temp\arduino_build_753782\sketch\src\hal\hal_avr\fastio.h:63:57: error: pasting "DIO" and "-" does not give a valid preprocessing token

 #define _READ(IO)             TEST(DIO ## IO ## _RPORT, DIO ## IO ## _PIN)

                                                         ^

c:\users\italo\appdata\local\temp\arduino_build_753782\sketch\src\hal\hal_avr\fastio.h:95:31: note: in expansion of macro '_READ'

 #define READ(IO)              _READ(IO)

                               ^~~~~

C:\Users\italo\AppData\Local\Temp\arduino_build_753782\sketch\src\feature\backlash.cpp:129:29: note: in expansion of macro 'READ'

     #define TEST_PROBE_PIN (READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING)

[... 149 lines cut ...]

                               ^~~~~

C:\Users\italo\AppData\Local\Temp\arduino_build_753782\sketch\src\feature\backlash.cpp:129:29: note: in expansion of macro 'READ'

     #define TEST_PROBE_PIN (READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING)

                             ^~~~

C:\Users\italo\AppData\Local\Temp\arduino_build_753782\sketch\src\feature\backlash.cpp:129:34: note: in expansion of macro 'Z_MIN_PIN'

     #define TEST_PROBE_PIN (READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING)

                                  ^~~~~~~~~

C:\Users\italo\AppData\Local\Temp\arduino_build_753782\sketch\src\feature\backlash.cpp:137:80: note: in expansion of macro 'TEST_PROBE_PIN'

     while (current_position.z < (start_height + BACKLASH_MEASUREMENT_LIMIT) && TEST_PROBE_PIN)

                                                                                ^~~~~~~~~~~~~~

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\italo\\AppData\\Local\\Temp\\arduino_build_753782\\sketch\\src\\feature\\fanmux.cpp" -o "C:\\Users\\italo\\AppData\\Local\\Temp\\arduino_build_753782\\sketch\\src\\feature\\fanmux.cpp.o"
exit status 1
Erro compilando para a placa Arduino/Genuino Mega or Mega 2560
```

italocjs, 2019-12-09:
Update:  it has something to do with the backlash gcode feature, when i disable it i can compile the backlash code using the preset values.

boelle, 2019-12-09:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Vertabreak, 2019-12-10:
need a probe to measure backlash when probing so this makes sense to me.

boelle, 2019-12-10:
will close this one until we have the config, the issue template says to attach them

also if you are trying to probe without a probe you have a config error and not a bug

closed 2019-12-10 by boelle

italocjs, 2019-12-11:
@Vertabreak , makes sense, i wasnt really thinking about using the auto-probing for now, so no problem for me. i have added a 0,05mm backlash on axis X and Y and the difference was good enough.   Is there a way to display a error saying clearly:  "you didnt enabled a probe, which is needed for this feature". A lot of features shows errors like this, which really helps.  im truly noob at programming these things, but i can try if you guys point me in the right direction

Sorry for not attaching it! @boelle  Here it is (this one is working now, to replicate the issue you just need to uncomment //#define MEASURE_BACKLASH_WHEN_PROBING)

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3950218/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3950219/Configuration.txt)

nicksears, 2020-05-13:
had the same issue when enabling BACKLASH_COMPENSATION
disabling #define MEASURE_BACKLASH_WHEN_PROBING
did NOT fix it because it was nested within something that was not enabled to begin with:

#define BACKLASH_COMPENSATION
#if ENABLED(BACKLASH_COMPENSATION)
  // Define values for backlash distance and correction.
  // If BACKLASH_GCODE is enabled these values are the defaults.
  #define BACKLASH_DISTANCE_MM { 0, 0, 0 } // (mm)
  #define BACKLASH_CORRECTION    1.0       // 0.0 = no correction; 1.0 = full correction

  // Set BACKLASH_SMOOTHING_MM to spread backlash correction over multiple segments
  // to reduce print artifacts. (Enabling this is costly in memory and computation!)
  //#define BACKLASH_SMOOTHING_MM 3 // (mm)

  // Add runtime configuration and tuning of backlash values (M425)
  //#define BACKLASH_GCODE

  #if ENABLED(BACKLASH_GCODE)
    // Measure the Z backlash when probing (G29) and set with "M425 Z"
    #define MEASURE_BACKLASH_WHEN_PROBING

nicksears, 2020-05-14:
it looks like enabling BLTOUCH caused the issue. 

I tried with a fresh copy (2.0.5.3) and only enabled backlash, bltouch and Z_SAFE_HOMING, and it worked fine. Transplanted the rest of my config changes and almost back in business!