MarlinFirmware/Marlin #16167: Homing failed after enabling skew correction(YZ)

jannorthoff, 2019-12-09:
After activating and calibrating skew correction for Z the printer (prusa style) would fail to home if Y was triggering Y-min Endstop before X did or Y was triggered at the start of the homing sequence it would instantly fail, while X did a ghost bump anywhere it was to that time. 

#define QUICK_HOME  
turned on or off didn't make a difference if Y was triggering at starting G28 but
#define HOME_Y_BEFORE_X  
did fix my problem but i guess only because YZ is the only Skew correction i'm applying and it makes it impossible for Y Position to be less than Y_MIN_POS + 1
due to bumping from the end stop

everything was also tested with a clean Vanilla of Bugfix and normal 1.1.x only activating the features i use and narrowed it down to skew correction 

I think that has something to do with planner.h but i honestly don't really have a clue, but i confirmed that it was only giving that error with skew correction enabled and was replicated every single time so its no noise or end stop issue.
Its working now, but i don't think that is a proper solution but just a get around 

`
#if ENABLED(SKEW_CORRECTION)

      FORCE_INLINE static void skew(float &cx, float &cy, const float &cz) {
        if (WITHIN(cx, X_MIN_POS + 1, X_MAX_POS) && WITHIN(cy, Y_MIN_POS + 1, Y_MAX_POS)) {
          const float sx = cx - cy * xy_skew_factor - cz * (xz_skew_factor - (xy_skew_factor * yz_skew_factor)),
                      sy = cy - cz * yz_skew_factor;
          if (WITHIN(sx, X_MIN_POS, X_MAX_POS) && WITHIN(sy, Y_MIN_POS, Y_MAX_POS)) {
            cx = sx; cy = sy;
          }
        }
      }

      FORCE_INLINE static void unskew(float &cx, float &cy, const float &cz) {
        if (WITHIN(cx, X_MIN_POS, X_MAX_POS) && WITHIN(cy, Y_MIN_POS, Y_MAX_POS)) {
          const float sx = cx + cy * xy_skew_factor + cz * xz_skew_factor,
                      sy = cy + cz * yz_skew_factor;
          if (WITHIN(sx, X_MIN_POS, X_MAX_POS) && WITHIN(sy, Y_MIN_POS, Y_MAX_POS)) {
            cx = sx; cy = sy;
          }
        }
      }`

boelle, 2019-12-09:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

jannorthoff, 2019-12-10:
Yes, sorry here they are
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3944901/config.zip)

boelle, 2019-12-11:
i used skew correction once and it did not fail

but i see you mention 1.1.x, is that what you use?

jannorthoff, 2019-12-11:
Yes, we did not upgrade to 2.0, i thought 1.1.x would be the last version for 8-bit only and since we use a Rumba board(Mega 2560) and Ram is already at its limits...do you think its worth a try going for Marlin 2? Since a lot of testing already went into 1.1.x and its running stable with custom menu options and stuff i'd rather not.
It does home normally if X triggers before Y, and with #define HOME_Y_BEFORE_X its working just fine, I couldn't find anything else besides not having the issue without skew correction as only difference, which is also being applied and working.

boelle, 2019-12-11:
yep, but keep your 1.1.x copy also

bugfix 2.0.x is where the development happens for now, the dev people does not work so active on 1.1.x unless someone makes PR's

but use a tool like winmerge to compare config files and copy over settings, the format of the config files are different so you cant just reuse them

jannorthoff, 2019-12-11:
Ok thank you so far, will try if i can get some time to throw at it but probably not this year though

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle