MarlinFirmware/Marlin #16168: [BUG] Marlin 2 compile for MKS Robin mini v2 fails

BoHoehne, 2019-12-09:
### Bug Description

Compilation of Marlin 2 for a MKS Robin Mini V2 (STM32F1) fails.

.piolibdeps\mks_robin_mini\TMCStepper_ID5513\src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such file or directory

### My Configurations

The board has 5x A4988 soldered driver chips on all axis and extruders.
TMC Drivers where NOT configured.

### Steps to Reproduce
Environment is VSC with PlatformIO and installed SoftwareSerial and TMCStepper libraries.
PlatformIO -> set board to "mks_robin_mini"

1. -> Configuration.h -> Configure BOARD_MKS_ROBIN_MINI as board
2. -> Configuration.h -> Configure A4988 on all axis
3. Hit compile

**Expected behavior:** [What you expect to happen]

Successfull compilation.

**Actual behavior:** [What actually happens]
The following error breaks compilation:
...
.piolibdeps\mks_robin_mini\TMCStepper_ID5513\src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such file or directory

boelle, 2019-12-09:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

BoHoehne, 2019-12-11:
Latest pull from Marlin 2 solved the problem (10.12.2019).
Marlin 2 compile for MKS_Robin_mini now works.

closed 2019-12-11 by BoHoehne