MarlinFirmware/Marlin #16169: [BUG] SKR PRO Bltouch doesn't work after flash or M502

BastR, 2019-12-09:
### Bug Description

Under certain conditions the bltouch does not deploy during a G28

### My Configurations

[Myconfig.zip](https://github.com/MarlinFirmware/Marlin/files/3941137/Myconfig.zip)

### Steps to Reproduce

1. Prepare firmware, compile and flash
2. Use BLTOUCH LCD menu to try BLTOUCH
3. BLTOUCH works as expected
4. Run M502, M500 then M501
5. Use BLTOUCH LCD menu to try BLTOUCH
6. BLTOUCH doesn't work
7. Use LCD reset button to reset the card
8. Use BLTOUCH LCD menu to try BLTOUCH
9. BLTOUCH works as expected

It seems that if you use a command related to the eeprom, the BLTouch is no longer functional until the next card reset.

carl1961, 2019-12-09:
I don't have this issue, but I'm using IC2 eeprom module, TMC 5160's  and I don't have senseless homing enabled. I'm also Using Marlin-bugfix-2.0.x  Version "2019-11-10" BTT SKR Pro 1.1 

 Do you have  senseless homing wired on Z?

dch1921, 2019-12-10:
@BastR are you using flash storage for eeprom emulation by any chance. I had a similar issue with BLTouch stop working after saving eeprom with flash emulation. Once I switched to sdcard emulation problem went away. I never did any additional testing to verify that was the cause

BastR, 2019-12-10:
@carl1961 what issues do you have?
I don't remember if I have sensorless homing wired on Z but what I'm sure of is that I don't use it in my firmware.

@dch1921 Yes you're right, I'm using flash storage...I'm waiting for @LinoBarreca to finish his awesome work here https://github.com/LinoBarreca/Marlin/pull/7 to access the onboard sd card

ilakuli, 2019-12-10:
https://github.com/MarlinFirmware/Marlin/issues/16146

Maybe related. I am also using Flash Eeprom Emulation

LinoBarreca, 2019-12-10:
@BastR unfortunately development is slow because I don't yet have the board and i need others to "pull/build/upload logs" but it's going on and making progress day after day (thanks to the others who provide feedback, of course)...my board is somewhere in italy...should be here soon.

carl1961, 2019-12-10:
@BastR  , I don't have any issues, but according to your Files above, you have sensorless home enabled. that may be a issue. 
[image]

BastR, 2019-12-10:
@carl1961 I don't think, as @ilakuli said I think it's related to Flash Eeprom Emulation because it only happen right after running M502 M500. Bltouch and more generally my printer works has expected after flash if I don't run M502. @dch1921 said this problem has been solved switching to sdcard emulation.

Dev5994, 2019-12-10:
I have this same problem. I'm using sensorless homing with tmc2130's on XY, Flash EEPROM, and using a BLTOUCH for my zmin. Z driver diag pin is not connected, and Z_STALL_SENSITIVITY is commented out.
1. Reset Board
2. BLTOUCH deploys/stows as expected via lcd menu, G28 works.
3. Adjust z offset and store settings.
4. issue G28
5. XY homes as expected, but probe does not deploy before z begins to move down.

What I've noticed, is that after the probe does not deploy for G28, if I reset the probe then deploy it manually the probe will not retract when I push it up with my finger. It instead displays the error leds

boelle, 2019-12-11:
could this maybe related to using a clone bltouch ? those are known to cause issues from time to time

dch1921, 2019-12-11:
I'm certain this is related to using flash emulation since the issue comes up after a M500 command. I had the same issue with a real BLTOUCH and it went away when I switched back to sdcard eeprom emulation.

> could this maybe related to using a clone bltouch ? those are known to cause issues from time to time

BastR, 2019-12-11:
@boelle I have an original too

sjasonsmith, 2019-12-11:
It is hard for me to imagine how storing to flash would cause the BLTouch to stop functioning... I'm not saying it doesn't, just that it doesn't immediately seem obvious how they would be related.

@randellhodges has been working on some changes to flash eeprom emulation for the SKR Pro. I don't know whether has has a BLTouch he could use to try to reproduce this, or if he would be interested in doing so.

Dev5994, 2019-12-11:
@boelle I have a genuine BLTouch as well.

I switched to sdcard emulation last night and every thing is working great now.

boelle, 2019-12-11:
could it be thinkable that there is a problem with the eeprom/flash on the board?

randellhodges, 2019-12-11:
I'll be able to test this at some point in the future, but, I'm probably weeks out before I swap boards. I might pull my BLTOUCH off my printer and connect it to my board that is just sitting on my desk, but no promises.

When you say `Use LCD reset button to reset the card` do you mean reset/reboot the board?

I have been up to my elbows in the PRO and flash emulation. I'm not entirely certain how it could be related, at least not directly. 

How do you have the BLTOUCH wired? Are you using the dedicated BLTOUCH set of pins? I remember reading a while back about people having issues with those sets of pins. Could it be possible there is some sort of interaction here? Seems like a stretch, but I'm still learning.

BastR, 2019-12-11:
@randellhodges I mean the reset button on the LCD
[image]

I don't remember if I have try to reset the board by the onboard reset button.

My wiring is like this
[image]

Dev5994, 2019-12-12:
@randellhodges 
My BLTouch is connected using the BLTouch header. PA1 for the servo pin, PA2 for the Probe pin. I changed the z endstop define in the board file to be PA2.

Can you link me the pins issue you mentioned?

randellhodges, 2019-12-12:
I don't have specific link, just things I've read in various places and see on Discord. @BastR  above has a nice diagram of the generally accepted way people are wiring it up.

I have the board, just not ready to start using it. I jumped in this thread because I'm currently working on some of flash eeprom emulation enhancements and I was just curious of the steps so I can try later to duplicate it myself.

danym21, 2019-12-12:
I have the same problem with Marlin-bugfix-2.0.x from 27th of November.
the Version Marlin-bugfix-2.0.x from 06th of October it was ok.
So the a change between this version has created the Issue.
SKR Pro v1.1 with Sensorless Homing activated same wiring like @BastR and Flash-based EEPROM emulation activated.

BastR, 2019-12-16:
I just compiled with today's version and I no longer have the bug. I've made a test on a second printer and it no longer has the bug either.

I close the bug report

closed 2019-12-16 by BastR

DarkShortie, 2019-12-17:
Is it working for everyone in this issue now? I'm asking because I'm just migrating to a SKR PRO and have similar issues, but I don't know if the issue is on my side or this bug still persists.

The BL Touch worked on my previous board (SKR 1.3) without any issues. On the pro it works on the different test items in Configuration -> BL Touch and also M43 probe test works without any issues.
`
Send: M43 S
Recv: Servo probe test
Recv: . using index:  0, deploy angle: 10, stow angle:   90
Recv: . Probe Z_MIN_PROBE_PIN: 37
Recv: . Z_MIN_PROBE_ENDSTOP_INVERTING: false
Recv: . Check for BLTOUCH
Recv: = BLTouch Classic 1.2, 1.3, Smart 1.0, 2.0, 2.2, 3.0, 3.1 detected.
Recv: ** Please trigger probe within 30 sec **
Recv: . Pulse width (+/- 4ms): 12
Recv: = BLTouch pre V3.1 (or compatible) detected.
`

On G28 it still does not deploy.

carl1961, 2019-12-17:
enabling Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN , this is proper way to tell marlin that the BLTouch is the Z min endstop .

DroneMang, 2020-04-22:
I have the same issue, I use Octoprint, and if I turn the PSU off turning power off the the SKR PRO and reconnect, then I can use the BLTouch plugin and see if a deploy and retract command works, if it does, homing will also work.

These two bugs are probably related:
https://github.com/MarlinFirmware/Marlin/issues/17627

https://github.com/MarlinFirmware/Marlin/issues/17628

GhostlyCrowd, 2020-04-22:
Just wanted to Post that this is still an issue on Buxfix 2.0.x branch on the SKR Pro 1.1 Exact same issue any eeprom saving causes tool head crash because of failure to deploy probe.

SD Eeprom Emulation is a work around that fixes this

randellhodges, 2020-04-22:
I was using an I2C eeprom and had problems with the bltouch not working
after a print finished. Had to power cycle.

I noticed the blue light wasn't in, so the pwm signal stopped working.

I have switched back to my 1.4 turbo.