MarlinFirmware/Marlin #16170: [FR] Add sanity check for Z_STEPPER_ALIGN_XY

MoshiBin, 2019-12-09:
### Description

Currently, setting coordinates in `Z_STEPPER_ALIGN_XY` that cannot be reached by the probe (due to a probe offset, for example) compiles successfully but fails during `G32`.

I'm suggesting adding a sanity check to make sure Z_STEPPER_ALIGN_XY is valid for the currently set bed size and probe offset.

InsanityAutomation, 2020-02-04:
@boelle this can get closed. This can now be calculated at runtime as well eliminating the need to set the variable in most cases. It can also be overridden with M422.

closed 2020-02-04 by boelle

ahmetcemturan, 2020-05-09:
When No coordinates are entered it will compile and upload, but when You try to execute a Z-leveling (G34) You get an Error.  (Probing failed) Note that Auto Bed leveling works fine...   
If I override with M422 and enter  25, 100 and 190, 100 it works.   But these coordinates are the limit. (In correspondence with 25mm probe from edge and Bed-size.

What I want to say is I have basically entered what leaving the coordinates commented out should have assumed. 

My Probe limits are: 
#define NOZZLE_TO_PROBE_OFFSET { 18, 12 , -2} 

#define MIN_PROBE_EDGE 25

#define X_BED_SIZE 215
#define Y_BED_SIZE 195

#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 215
#define Y_MAX_POS 195
#define Z_MAX_POS 170

randellhodges, 2020-05-09:
I am using `Z_STEPPER_AUTO_ALIGN` and I do not have `Z_STEPPER_ALIGN_XY` defined and it probes just fine.

I am on about a month old version of bugfix (5a5be7e287183e633ee3235ee1bcd79a72a1a1f5) however, so maybe this is only in newer versions.

sjasonsmith, 2020-05-09:
I am the same as @randellhodges. Three point auto-selection is fine for me, although also slightly older.

Try enabling DEBUG_LEVELING_FEATURE (and enable prints with the gcode in the comment next to it), and see if it reports the points it is trying to use.

It would be better to continue discussing this over on discord or another support forum to help figure out if it is a configuration issue or a bug. If it is a bug it should be reported as a me issue and not tracked onto a closed one.

PR #16231 (2019-12-19): Add sanity check for Z_STEPPER_ALIGN_XY
### Description

When `Z_STEPPER_AUTO_ALIGN` is enabled, it doesn't check during compile time that the points defined are probeable (taking into account the probe offset). This means that when a user misconfigures it (like myself), the G34 command doesn't work - it homes XYZ, then prints a vague error to serial (I think it was "error: 0" or something like that).

### Benefits

This change adds a sanity check that verifies that each point in `Z_STEPPER_ALIGN_XY` is probeable (within XY min/max ± probe offset). The misconfiguration will be visible during c [... cut]