MarlinFirmware/Marlin #16172: UBL does not respect soft end stops.

dlingo, 2019-12-09:
### Bug Description

During G29 P1 the extruder carriage reaches end of Y/X travel and then the steppers skip.

### My Configurations

[Marlin_Config.zip](https://github.com/MarlinFirmware/Marlin/files/3941933/Marlin_Config.zip)

### Steps to Reproduce

Power on after flashing
M502
M500
M851 Z -2.3
M500
G28
G29 P1 -> about half way through fails to stop before endstop.
I have tried changing the MIN_PROBE_EDGE from 0 -20.
I have also checked my Z probe offsets to be X: -43, Y: 2, Z-2.3.
I have checked that my full bed is reachable using G1 X255 Y255 without an issue.

**Expected behavior:** [What you expect to happen]

Expected to probe mesh points and respect max travel distances.

**Actual behavior:** [What actually happens]
Goes past soft limits set.

#### Additional Information
Printer is a Geeetech A20M that is dual extruder single nozzle.

ellensp, 2019-12-10:
I wish someone would change that label. they are not endstops they are homing switches.   Unless you enable ENDSTOPS_ALWAYS_ON_DEFAULT in Configuration_adv,h ...    I don't know if this affects your issue or not but give it a try.

tatusah, 2019-12-10:
I've been under impression that the travel limits should also understand the limitations caused by the probe offset, still for extra precaution I've always set the probe edge limits for each side separately. If the collision with the travel movement was happening in your case with x axle then you could try setting the probe edge limit for the right side of the bed to 43 mm in configuration_adv.h file. At least then it should try to move the carriage further than where the nozzle can reach in that direction.

Vertabreak, 2019-12-10:
i personally spent days with him on this and im at a loss for why it does what it does on his machine but not on any of mine. i even compiled a copy on my machine and sent that to him it till did it. we verified as many variables as i could think of and even tried some crazy stuff to no effect.

dlingo, 2019-12-12:
I have done some more testing. With my bed size set to x 250 y 250, x 0 y 0 being at the bottom left. My probe is 43mm to the left of nozzle and 1mm forward on the y axis (offset of x -43  y 1 z -2.3) This is my debug log from my board. Still ignore minimum probe edge along the x axis and crashes into the right side.
```
Send: M105
Recv: ok T:22.70 /0.00 B:21.91 /0.00 @:0 B@:0
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok P15 B3
Send: M105
Recv: ok T:22.62 /0.00 B:21.80 /0.00 @:0 B@:0
Send: M111 S32
Recv: echo:DEBUG:LEVELING
Recv: ok P15 B3
Send: M105
Recv: ok T:22.81 /0.00 B:21.72 /0.00 @:0 B@:0
Send: M502
Recv: reset_bed_level
Recv: Axis X min:-7.00 max:250.00
Recv: Axis Y min:0.00 max:250.00
Recv: Axis Z min:0.00 max:250.00
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok P15 B3
Send: M105
Recv: ok T:22.93 /0.00 B:21.87 /0.00 @:0 B@:0
Send: M500
Recv: echo:Settings Stored (616 bytes; crc 13063)
Recv: ok P15 B3
Send: M105
Recv: ok T:23.01 /0.00 B:21.68 /0.00 @:0 B@:0
Send: T0
Recv: >>> T(0)
Recv:   current_position= X-7.00 Y0.00 Z0.00 : BEFORE
Recv:   current_position= X-7.00 Y0.00 Z0.00 : AFTER
Recv: <<< T()
Recv: ok P15 B3
Send: M105
Recv: ok T:22.77 /0.00 B:21.60 /0.00 @:0 B@:0
Send: M105
Recv: ok T:22.66 /0.00 B:21.80 /0.00 @:0 B@:0
Send: M105
Recv: ok T:22.85 /0.00 B:21.76 /0.00 @:0 B@:0
Send: G28
Recv: >>> G28
Recv: Machine Type: Cartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X-43.00 Y2.00 Z-2.30 (Left-Back & Below Nozzle)
Recv: Auto Bed Leveling: UBL
Recv:  (disabled)
Recv: 
Recv: Raise Z (before homing) to 5.00
Recv: >>> do_blocking_move_to X-7.00 Y0.00 Z5.00
Recv: <<< do_blocking_move_to
Recv:   current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
Recv: >>> do_blocking_move_to X-385.50 Y-375.00 Z5.00
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: echo:busy: processing
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, -375.00, [40.00])
Recv: <<< do_homing_move(Y)
Recv: Move Away:
Recv: >>> do_homing_move(Y, 5.00, [40.00])
Recv: <<< do_homing_move(Y)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Y, -10.00, 20.00)
Recv: <<< do_homing_move(Y)
Recv: >>> set_axis_is_at_home(Y)
Recv:   current_position= X0.00 Y0.00 Z5.00 :
Recv: <<< set_axis_is_at_home(Y)
[... 617 lines cut ...]
Recv: 
Recv: >>> probe_at_point(162.50, 50.00, raise, 0, probe_relative)
Recv:   current_position= X93.00 Y48.00 Z7.30 :
Recv: >>> do_blocking_move_to X205.50 Y48.00 Z7.30
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:   current_position= X205.50 Y48.00 Z7.30 : set_probe_deployed
Recv: deploy: 1
Recv:   current_position= X205.50 Y48.00 Z7.30 : >>> run_z_probe
Recv:   current_position= X205.50 Y48.00 Z7.30 : >>> do_probe_move
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to X205.50 Y48.00 Z0.30
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2793
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2793)"
Connection closed, closing down monitor

```

tatusah, 2019-12-12:
What did you had set for the min probe edge this time? Have you tried to set something like `#define MIN_PROBE_EDGE_RIGHT MIN_PROBE_EDGE 50` in configuration_adv.h?

Vertabreak, 2019-12-12:
The issue seems to appear when min probe edge is large enough to create a extra row for the mesh. I tested up to min probe edge 30 without getting the issue.

tatusah, 2019-12-12:
OK, interesting. And definitively think also that high min edge limit is silly if used for each side of the bed even on those sides that don't need that large margin. I've set NOZZLE_TO_PROBE_OFFSET { -48, -10, 0 }, MIN_PROBE_EDGE_LEFT 1, MIN_PROBE_EDGE_RIGHT 48, MIN_PROBE_EDGE_FRONT 5, MIN_PROBE_EDGE_BACK 10 and mesh inset of 1 on all sides except for the right side which has 10. UBL 9x9. I set these to optimize the mesh probing to have maximum points when considering the probe offset and travel limits on the right side. Now the 8th column gets probed as close to the edge where the probe can reach and the last column is filled as per your earlier guidance with G29 P3. This setup works for me and no crashes with travel limits.

I agree with the quirk thinking. I think I could review through again the probe point calculation code again as I did previously when I decided to optimize the probing area.

Vertabreak, 2019-12-12:
once i was finally able to reproduce it i quickly figured out what was happening and started playing with it more its kinda weird.

boelle, 2019-12-13:
but you where not actually able to spot precise what is wrong?

tatusah, 2019-12-13:
Based on probe.h:98 probe_max_x is `_MIN((X_MAX_BED) - (MIN_PROBE_EDGE_RIGHT), (X_MAX_POS) + probe_offset.x)` so smaller of the two. At least with the settings I described previously my printer seems to play by that rule. So in @dlingo situation it would be interesting to see how far actually the printer would actually try to go if the bed size was configured something like 20mm smaller and maybe having a measuring tape or a ruler at hand to check how far the probe/nozzle would actually travel outside this smaller bed area. And double checking that the measured probe offset is actually correct value and perhaps set the right side probe to match the offset value.

Or do you @Vertabreak have some specific steps to reproduce this or should this be regarded as a configuration issue?

Vertabreak, 2019-12-19:
its not exactly a bug and we did workaround the quirk with configuration settings. i would consider this matter resolved but i defer to @boelle  & @dlingo. last i spoke with dlingo it seemed to be resolved and he had it working.

dlingo, 2019-12-23:
I tested the bed leveling again today. It worked the first time I did it.
G28
G29 P1
G29 P3 
G29 S1
G29 A
M500

It exceeded the min probe edge the second time I did it and crashed into the side.
I'm not sure what is making this issue intermittent. I did not flash firmware between tests or since it was last working when I spoke with Vertabreak....

Vertabreak, 2019-12-23:
ill dig in and do more testing with you msg me on discord.

dlingo, 2019-12-23:
I updated my firmware and used a minimum probe edge of 20 and 10.
With a MPE of 20 it worked the first time then I updated the probe offset with M851 Z-3.35.
The second run it crashed into the side with the following log. The log shows that it doesn't exceed the mesh boundaries, yet it is travelling further then expected. Seems to only be after completing the first mesh and saving or updating the probe offset value?
```
Send: M111 S38
Recv: echo:DEBUG:INFO,ERRORS
Recv: ok P15 B7
Send: M105
Recv: ok T:31.38 /0.00 B:35.33 /0.00 @:0 B@:0
Send: M105
Recv: ok T:31.28 /0.00 B:35.35 /0.00 @:0 B@:0
Send: G29 P1 V4
Recv: //action:notification Doing G29
Recv: Mesh invalidated. Probing mesh.
Recv: Probing around (165.00,125.00).
Recv: 
Recv: 
Recv: Probing mesh point 1/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 72.500 Y: 125.000 Z: 0.000
Recv: 
Recv: Probing mesh point 2/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 125.000 Y: 125.000 Z: 0.027
Recv: 
Recv: Probing mesh point 3/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 72.500 Y: 72.500 Z: -0.092
Recv: 
Recv: Probing mesh point 4/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 72.500 Y: 177.500 Z: -0.065
Recv: 
Recv: Probing mesh point 5/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 125.000 Y: 177.500 Z: 0.067
Recv: 
Recv: Probing mesh point 6/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 125.000 Y: 72.500 Z: 0.017
Recv: 
Recv: Probing mesh point 7/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 20.000 Y: 125.000 Z: -0.135
Recv: 
Recv: Probing mesh point 8/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 20.000 Y: 72.500 Z: -0.158
Recv: 
Recv: Probing mesh point 9/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 20.000 Y: 177.500 Z: -0.100
Recv: 
Recv: Probing mesh point 10/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 177.500 Y: 125.000 Z: 0.108
Recv: 
Recv: Probing mesh point 11/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 177.500 Y: 72.500 Z: 0.057
Recv: 
Recv: Probing mesh point 12/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 177.500 Y: 177.500 Z: 0.125
Recv: 
Recv: Probing mesh point 13/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 72.500 Y: 230.000 Z: 0.160
Recv: 
Recv: Probing mesh point 14/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 125.000 Y: 230.000 Z: 0.922
Recv: 
Recv: Probing mesh point 15/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 72.500 Y: 20.000 Z: 0.372
Recv: 
Recv: Probing mesh point 16/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 125.000 Y: 20.000 Z: 0.710
Recv: 
Recv: Probing mesh point 17/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 20.000 Y: 20.000 Z: 0.372
Recv: 
Recv: Probing mesh point 18/25.
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
```

dlingo, 2019-12-26:
After doing some more testing the first time after power up the UBL completes successfully and does not ignore the min probe edge.
```
G28
G29 P1
G29 P3
G29 F10
G29 S1
G29 A
M500
```
Result: Successful UBL while respecting min probe edge.
Run second UBL without power-cycling.

```

G28
G29 P1
G29 P3
G29 F10
G29 S1
G29 A
M500
```
Result:
Ignores min probe edge.

dlingo, 2019-12-26:
This issue seems to be intermittent somehow. It completes successfully sometimes and other times it has issues with the coordinates.

Vertabreak, 2019-12-26:
I have an A20T on the way now for testing i intend to get to the bottom of what the hell is happening this is my white whale now.

Roxy-3D, 2019-12-27:
> I have an A20T on the way now for testing i intend to get to the bottom of what the hell is happening this is my white whale now.

Any help on this issue will be much appreciated! 

Thank You!

Vertabreak, 2019-12-27:
so i woke from another dream about code and shortly after that i tested a few things and i think i have pin down the problem. there seems to be a conflict between mesh inset and min probe edge that results in some odd side effects. setting mesh inset to 0 and min probe edge to 10 the desired effect seem to be achieved UBL skipped 1 row on all sides. ill post a video showing this shortly and ill have some of my community also run tests to verify.

Roxy-3D, 2019-12-27:
> mesh inset seems to be the more favorable of the 2 settings with the way UBL works.

Mesh Inset was the original method to control (or limit) the actual space on the bed.   Its main purpose was to keep the nozzle away from where a person would clamp a piece of glass to the bed.

Vertabreak, 2019-12-27:
the problem seems to arise in some instances of combination of min probe edge and mesh inset. simply setting one to 0 and only using just the one seems to remedy the problem.

Roxy-3D, 2019-12-27:
Yes...  That information is valuable and very helpful.    But we should fix the logic so it doesn't cause problems in the future.

tatusah, 2019-12-27:
I checked your video @Vertabreak  Did the findings explain why the movement limits weren't respected? Would it be possible that the custom include of _start_here.h_ file was not working correctly with conditionals and sanity check? Also there seems to be setting x_min_pos -7 in the configuration.h file which would lead to x_bed_size calculation of x_max_pos - (-7) if checking the custom definitions was checked in wrong order (if the definition of x_bed_size wasn't caught from the custom configuration structure). 

I've been using UBL with different bugfix-2.0.x versions and also with 2.0.1 release without problems even with both mesh inset and min probe edge set to values larger than 0. The settings for min_edge and mesh inset are in relation to `NOZZLE_TO_PROBE_OFFSET { -48, -10, 0 }`. I haven't seen a situation where the probing would happen outside of the movement limits.

```
#if PROBE_SELECTED && !IS_KINEMATIC
  // Custom probe edge limits
  #define MIN_PROBE_EDGE_LEFT 1
  #define MIN_PROBE_EDGE_RIGHT 48
  #define MIN_PROBE_EDGE_FRONT 5
  #define MIN_PROBE_EDGE_BACK 10
#endif

#if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  #define MESH_MIN_X 1
  #define MESH_MIN_Y 5
  #define MESH_MAX_X (X_BED_SIZE - 1)
  #define MESH_MAX_Y (Y_BED_SIZE - 10)
#endif
```

Some time ago the comments in configuration.h told to set the min probe edge equal to the largest offset of the probe which would make sense if there wasn't the soft endstops present. Most people probably don't want to waste that large amount of space which could actually be probed on other sides of the bed when only one side would need the large edge limit. So that's why I also set the individual edge and mesh limits making sure that the bed clips are not in the way.

With the custom include file probe offset `#define NPO { -43, 1, -2.3 }` @dlingo has it would make sense to test setting the #define MIN_PROBE_EDGE_RIGHT 43 and see if that prevents collision. Also if by testing with a bed size something like 50mm smaller (x-axis) should reveal how far the system actually wants to go over the soft endstop limits. And if setting the right side min probe edge to the value matching the probe offset solves the problem then it would mean that the while doing ubl mesh the soft endstop limits are not respected. Like the case with global min probe edge of 20 would mean that with -43 probe offset the would move the nozzle 23 mm over the right edge if the soft endstop was not respected.

I'm still curious why the issue seems to happen now only on the second UBL cycle with dlingo's setup and if the issue was somehow related to the `x_min_pos` or using custom repository with the custom include settings file.

Vertabreak, 2019-12-27:
In my testing it functions correct now awaiting further tests.

jgwinner, 2020-03-30:
Well, I just blundered into it,

I was using bilinear; I have a BLTouch. My probe is pretty far 'back' from the nozzle (63mm). Used to work fine, but now despite doing a MIN_PROBE_EDGE of 30mm, it'll probe right at the Y=0, and trigger the endstop. As it moves from the home position right over to x=270 or so, maybe I have some y skew.

I always figured the MIN_PROBE_EDGE_FRONT was an offset; and it is.

However, between G29.CPP and PROBE.CPP, the probe offset is added, then subtracted.

However, the min is calculated WITH the probe offset. So, basically the min is always 0 UNLESS the MIN_PROBE_EDGE_FRONT is more than the probe offset.

I think what should happen is the MIN_PROBE_EDGE_FRONT should have the probe offset added to it after configure_adv is calculated, but it's probably easier and less breaking to simply make a note in configure_adv.h that you must have a MIN_PROBE_EDGE_FRONT that is greater than NOZZLE_TO_PROBE_OFFSET. 

So the 'real' offset when the probe starts probing is any MIN_PROBE_EDGE_FRONT greater than NOZZLE_TO_PROBE_OFFSET.

Once I set that (like 73 or so) probing didn't hit my clips :)

So, I think the fix is either messy or a document update.

jgwinner, 2020-04-07:
I know this is closed, but it's still a bug, or I wouldn't have had to increase the min front to 73 to get it to move 20mm ... thoughts?

thinkyhead, 2020-04-08:
@jgwinner — A log from you would be very helpful to demonstrate the bad logic now being applied to determine where to probe.

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. Stop the procedure if it might hurt something!
- Copy the log output into a **.TXT file** and attach it to your next reply.

I have a strong notion of what the bad logic is, but it would be good to see the exact bad numbers in the log.

Jbchow, 2020-04-26:
@thinkyhead I am having a similar issue with UBL. It goes off the bed on every time.
The Left hand side the probe goes off the bed when it is less than 30 mm

Jbchow, 2020-04-26:
Probe count continues to climb up but it is not moving (X/Y)
Photos of where head is: https://photos.app.goo.gl/2J1fhCArayE8Qkcb6

`READ: >>> do_blocking_move_to X254.39 Y193.47 Z2.73
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X254.39 Y193.47 Z2.73
READ: <<< do_blocking_move_to
READ: 2nd Probe Z:2.73 Discrepancy:-0.04
READ:   current_position= X254.39 Y193.47 Z2.73 : <<< run_z_probe
READ: >>> do_blocking_move_to X254.39 Y193.47 Z7.73
READ: <<< do_blocking_move_to
READ: echo:busy: processing
busy: processing
READ: <<< Probe::probe_at_point
READ: 
READ: Probing mesh point 26/100.
READ: 
READ: >>> Probe::probe_at_point(189.30, 243.94, raise, 0, probe_relative)
READ:   current_position= X254.39 Y193.47 Z7.73 : 
READ: >>> do_blocking_move_to X232.99 Y224.04 Z7.73
READ: <<< do_blocking_move_to
READ:   current_position= X232.99 Y224.04 Z7.73 : Probe::set_deployed
READ: deploy: 1
READ:   current_position= X232.99 Y224.04 Z7.73 : >>> Probe::run_z_probe
READ:   current_position= X232.99 Y224.04 Z7.73 : >>> Probe::probe_down_to_z
READ: >>> do_blocking_move_to X232.99 Y224.04 Z0.00
READ: <<< do_blocking_move_to
READ: echo:busy: processing
busy: processing
READ:   current_position= X232.99 Y224.04 Z2.66 : sync_plan_position
READ:   current_position= X232.99 Y224.04 Z2.66 : <<< Probe::probe_down_to_z
READ: 1st Probe Z:2.66
READ: >>> do_blocking_move_to X232.99 Y224.04 Z7.66
READ: <<< do_blocking_move_to
READ: echo:busy: processing
busy: processing
READ:   current_position= X232.99 Y224.04 Z7.66 : >>> Probe::probe_down_to_z
READ: >>> do_blocking_move_to X232.99 Y224.04 Z0.00
READ: <<< do_blocking_move_to
READ: echo:busy: processing
busy: processing
READ:   current_position= X232.99 Y224.04 Z2.62 : sync_plan_position
READ:   current_position= X232.99 Y224.04 Z2.62 : <<< Probe::probe_down_to_z
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.63
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.63
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.64
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.64
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.65
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.65
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.66
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.66
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.67
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to X232.99 Y224.04 Z2.67
READ: <<< do_blocking_move_to
READ: 2nd Probe Z:2.67 Discrepancy:-0.01
READ:   current_position= X232.99 Y224.04 Z2.67 : <<< run_z_probe
READ: >>> do_blocking_move_to X232.99 Y224.04 Z7.67
READ: <<< do_blocking_move_to
READ: echo:busy: processing
busy: processing
READ: <<< Probe::probe_at_point
[... 908 lines cut ...]
READ: bltouch.deploy_proc() end
READ: >>> do_blocking_move_to X190.18 Y71.16 Z13.00
READ: <<< do_blocking_move_to
READ:   current_position= X190.18 Y71.16 Z13.00 : >>> Probe::run_z_probe
READ:   current_position= X190.18 Y71.16 Z13.00 : >>> Probe::probe_down_to_z
READ: >>> do_blocking_move_to X190.18 Y71.16 Z0.00
READ: <<< do_blocking_move_to
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ:   current_position= X190.18 Y71.16 Z0.00 : sync_plan_position
READ:   current_position= X190.18 Y71.16 Z0.00 : <<< Probe::probe_down_to_z
READ: FAST Probe fail! - No trigger.
READ:   current_position= X190.18 Y71.16 Z0.00 : <<< run_z_probe
READ:   current_position= X190.18 Y71.16 Z0.00 : Probe::set_deployed
READ: deploy: 0
READ: Probe::move_z(10.00)
READ: >>> do_blocking_move_to X190.18 Y71.16 Z13.00
READ: <<< do_blocking_move_to
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
`

Jbchow, 2020-04-26:
@thinkyhead Running more tests if I clear out manual settings (mesh min/max and min probe edge) and set MESH_INSET to 100 it doesn't leave the bed but it also doesn't seem to be doing a square probe area 100mm off edges of bed. It is defiantly biased to the front left. I also noticed the homing probe is no longer in the middle of the bed it is also biased to the front left. 
Related notes:
2209 steppers running in uart mode.
Changed XY_PROBE_SPEED from 6000 to 4500 to see if i was maybe missing steps. Did exactly the same behavior. Returned to 6000.
I keep on getting "homing failed' after running ubl and then trying to home again. it moves to about the half way point on the bed and then tries to do a bump but doesn't do the bump. I am running physical homing switches wired to different pins I updated the pins file to match.

Jbchow, 2020-04-26:
I also tried doing a "manual probe" from one of the UBL sub menus. It went to -25x -32y and asked me to place shim and measure - it crashes into front left side....

Jbchow, 2020-04-27:
If I set the probe points to 3 it works as expected. If I set it to more than 3 it does that same strange behavior (tested 5,9,10 and 11)

Jbchow, 2020-04-27:
Sometimes G29 J works flawlessly other times it does the first probe point then the second probe point without moving then goes off diagonally left (X-), back (Y+) It goes all the way back to the Y+ point it would normally go to on point 3. The issue is it bottoms out on the left hand side the whole way back.

I am also seeing random issues where it it does not deploy the BLTOUCH Prob on G28.

Jbchow, 2020-04-28:
Log from Failed G28
[Failed G28.txt](https://github.com/MarlinFirmware/Marlin/files/4542781/Failed.G28.txt)

Jbchow, 2020-04-28:
Update: I believe I have an intermittent X limit switch. I did a G29 J and hit the limit switch and it immediately stopped and probed where it was. I am not sure this is a desirable feature since it still tries to go to the back middle position as if it made it to the far right position. Might be more useful to return and error at that point. I am starting to think this was also my issue with UBL. I am going to replace the wire and see if the issues goes away. Hope this helps someone else.

Jbchow, 2020-04-28:
Replaced questionable parts of wire system was working more reliably then the issue started again. I enabled debugging_pins and turned on end stop. The issue persists No end stop switch triggered but X axis stopped mid path and tried to home and failed because no switch triggered. See log attached.
[G28 Fail - new wire.txt](https://github.com/MarlinFirmware/Marlin/files/4543282/G28.Fail.-.new.wire.txt)

Jbchow, 2020-04-28:
As another test I ran G29 P1 and when it started probing I unplugged the limit switches it continued to move forward and left never backwards and right. It would also probe the same place over and over then move off the bed. For some reason the end stops mess up UBL.

Jbchow, 2020-04-30:
As I was reading through the end stop code I found out about the `ENDSTOP_NOISE_THRESHOLD` function. Code describing it says increase samples by 7 brings likely hood of false probe down to 1% at a cost of 7ms.  so I set it to 7. It is unclear why it seems to take 7 seconds after the second probe bump to continue on but it is working now. I was able to complete a G29 with 100 points and it worked great. Only issue was the 7s timeout caused the BLtouch to timeout so the next probe it would no longer be deployed and would fail to hit and would deploy and try again. This happened nearly every time. I suspect that all of my issues here had to do with the fact that i remapped the end stop pins, so i was able to use the diagnostic pins on my 2209 drivers, The normal input pins have built in 4.7k pull ups and a 100nF capacitor. The capacitor is recommended in the configuration.h document. I am going to go install some 100nF capacitors and 4.7k pull ups. This should resolve the issue. Hope helps someone else.

AnHardt, 2020-04-30:
> Hope helps someone else.

From my experience - No!
Since over 5 years about every week someone discovers the not so surprising fact a functionality for remembering changed settings over resets , and if possible over firmware updates, does its job.
One report more or less does not make a difference anymore.
I guess everyone has to make this experience by him/her self. Education useless.

Jbchow, 2020-04-30:
Update to close this out: Caps resolved issue. I turned `ENDSTOP_NOISE_THRESHOLD` off after I installed them and everything works! Glad the notes were in the code helped to clue me in. Thank you for your support!

jgwinner, 2020-04-30:
@Jbchow Your issue is completely aside from the MIN_PROBE_EDGE issue.  Hopefully it'll help someone, but obviously unrelated to this bug. Honestly, github issues aren't the right place to go on about this.

@thinkyhead Sorry for they delay, been busy printing COVID-19 masks for USC-Keck/Cedars.

Here are the files. I've had a lot of issues with z probe offset, but you should be able to see the x/y probe offset. I eventually changed mine from such a large MIN_PROBE_EDGE, as I moved to a magnetic bed and quit popping off my binder clips, but I still have the weird issue that it won't probe close to the edges. I think it's due to the large Y probe offset I have.

[ZProbe_LogGood.txt](https://github.com/MarlinFirmware/Marlin/files/4561682/ZProbe_LogGood.txt)

I stripped out 'busy' and temp chatter. Let me know if you also need my config.

Roxy-3D, 2020-04-30:
@jgwinner   Should this be re-opened?

jgwinner, 2020-05-01:
@Roxy-3D That would be great - it is still a problem as near as I can tell. My situation is probably unique (probe is 63mm further back in "Y" ... Crearibo carriage, not enough room to move the probe +/-X). I found the spot in the code where it's figuring it wrong, but forgot to make a note :(

        == John ==

Roxy-3D, 2020-05-01:
Well...   If you can find the spot in the code where it is figuring it wrong again...  Cut and paste it here.   We can all look at it and see if we can agree that is the right place to fix it.   And if so...   It will happen quickly!!!   :)

closed 2020-05-09 by Roxy-3D