MarlinFirmware/Marlin #16175: [BUG] Extruder motors not working

LordBBQ, 2019-12-10:
After flashing marlin 2.0 release on a skr 1.3 with TMC 2209 drivers, neither extruder works. One does not turn at all, the geared one thumps, and vibrates.
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3943456/Archive.zip)

m00anj00, 2019-12-10:
sounds like you have set the current to an unreasonable value?  I have tmc2208 which do the same thing if set too low.  could this be "endstop detection" by stallguard?

LordBBQ, 2019-12-11:
[image]
This is the calculation for the VREF. Does it look correct? The 1st extruder is a different motor so is different. I think calculations for TMC drivers are all the same. Stallguard endstops are not enabled.

LordBBQ, 2019-12-11:
What would you recomend?

m00anj00, 2019-12-11:
hold on, I just realized a clue.  vibrating motors can be a pin not connected.  are you sure that your motor plugs are fine?  did you pull the board to reflash it?  
I use 1.3A nema 17s and usually target 900ma.  you are running at 700ma which is a lot lower than me.  
I thought TMC2209 drivers were software set and not set by the pot any more?
you have set up all pins correctly for comms with the driver and set the TMC section correctly havent you?  as in you have wired the 1k resistor and single wire RX/TX with the board also?

not trying to teach you to suck eggs, just going through obvious stuff first

m00anj00, 2019-12-11:
to add:  are you using softserial or real serial (RX2/TX2) in parallel and indexed the drivers?  if so the board MS jumpers will make a difference.  Theres a debug for TMC drivers (M122) which should show all stepper drivers with unique IDs and state of each by setting

m00anj00, 2019-12-11:
Not for TMC2209 but very similar to do:
https://www.instructables.com/id/TMC2208-UART-on-BigTreeTechBIQU-SKR-V11-and-V13-Co/

rado79, 2019-12-11:
maybe this could be helpful https://github.com/MarlinFirmware/Marlin/issues/15926#issuecomment-563228807

LordBBQ, 2019-12-13:
Motor plugs _look_ fine. Tried M122, and all showed up fine. Tried square wave stepping, still no joy. However, one is working. That still leaves the geared one. These drivers are set in configuration_adv. These drivers were ordered configured for UART. The geared one makes 2 clicks, then refuses to move unless I reset the board. This video describes the issue well: https://www.youtube.com/watch?v=sG2uG-XWRxg 

> to add: are you using softserial or real serial (RX2/TX2) in parallel and indexed the drivers? if so the board MS jumpers will make a difference.

What do you mean?

m00anj00, 2019-12-13:
You can drive tmc2208 and tmc2209 using real serial instead.  Grab the data
sheets for details.  You use pins 16 and 17 (serial2) instead of all of the
uart pins you are using and using only the serial 2 pins in parallel.  You
use the ms jumpers as ID.  With 2208 you have to multiplex tx but 2209 can
do this directly.  Uses 2 pins for all stepper drivers.

boelle, 2019-12-13:
@LordBBQ tried what @m00anj00 said?

LordBBQ, 2019-12-14:
Changed VREF to 900. It looks like the problem is with the E0 port, not the driver. I tried swapping cables motors, and the problem stayed with E0. Tried moving drivers, and the issue was still at E0. I forced the driver into 'dumb' mode, and it still did not move. I also pointed a fan at it. Still nothing. I tried printing with just E1, and it works for testing, but as soon as I try to print, it does not extrude.

m00anj00, 2019-12-14:
This is sounding familiar to my issue.  Mine is that pin 9 wont become
active so no second extruder.  Do me a favour please and try to remap a
couple of pins and see if they actually do remap.  See my post : second
extruder not working.

LordBBQ, 2019-12-15:
I remapped the extruder to run the z axis, and it did not move. I also noticed that the motor was not enabled, and was freely movable. I also noticed that in the `pins_BTT_skr_1_3.h` file there were no definitions for the Step, Direction, and enable pins as shown here:

 `#define E1_STEP_PIN        P0_01
#define E1_DIR_PIN         P0_00
#define E1_ENABLE_PIN      P0_10`

New config files below:
[Archive 2.zip](https://github.com/MarlinFirmware/Marlin/files/3964594/Archive.2.zip)

sjasonsmith, 2019-12-18:
@LordBBQ, can you reproduce this problem and provide the output from the M122 command here?

I want to verify whether the number of microsteps is set correctly on your E drivers. I have a theory that sometimes it may not set correctly for the _last_ extruder, but I have not found a way to reproduce it on my own SKR. I provided a workaround to a user who indicated it solved their issue, but they disappeared before I could fully understand why it worked. I would have only expected my workaround to change anything with single-pin serial, but the SKR by default uses two-pin serial.

Also, can you provide the pins file you are using? I would like to verify the serial pin configuration.

LordBBQ, 2019-12-19:
[pins_BTT_SKR_V1_3.h.zip](https://github.com/MarlinFirmware/Marlin/files/3981147/pins_BTT_SKR_V1_3.h.zip)

`Send: M122
Recv: 		X	Y	Z	E	E1
Recv: Address		0	0	0	0	0
Recv: Enabled		false	false	false	false	false
Recv: Set current	760	760	760	760	800
Recv: RMS current	734	734	734	734	795
Recv: MAX current	1035	1035	1035	1035	1121
Recv: Run current	23/31	23/31	23/31	23/31	25/31
Recv: Hold current	11/31	11/31	11/31	11/31	12/31
Recv: CS actual	11/31	11/31	11/31	11/31	12/31
Recv: PWM scale	13	13	13	13	14
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: tstep		max	max	max	max	max
Recv: pwm
Recv: threshold	98	98	658	283	31
Recv: [mm/s]		100	100	3	30	30
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time	4	4	4	4	4
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2	2
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E	E1
Recv: stst		*	*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0B:00:00
Recv: 		Y	0xC0:0B:00:00
Recv: 		Z	0xC0:0B:00:00
Recv: 		E	0xC0:0B:00:00
Recv: 		E1	0xC0:0C:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: Testing E1 connection... OK`

LordBBQ, 2019-12-22:
> I provided a workaround to a user who indicated it solved their issue
What is this workaround?

sjasonsmith, 2019-12-22:
There might be an issue when you are using single-pin serial (you are not) which can cause the last driver to not be programmed correctly. When I have seen this M122 reports the default 256 microsteps on the last E driver. Yours looks fine, so I don't think that is your problem.

LordBBQ, 2019-12-24:
I tried using an 8-bit MKS Gen L, but there was a issue with the mosfets on both on the heated bed, and all the terminals were shorted. Back to the drawing board!

LordBBQ, 2019-12-25:
I tried the same thing on Marlin 2.0.1, on my SKR, and no improvement.

LordBBQ, 2019-12-26:
On Marlin 2.0.1, If I change the extruder count back down to 1, it works fine. This is not a solution, but merely a workaround.

boelle, 2020-01-10:
what happens on bugfix 2.0.x ?

m00anj00, 2020-01-10:
That was on bugfix 2, ver2, 2.1 when I tested

boelle, 2020-01-10:
bugfix 2?
ver2?
2.1?

we have 2.1
but not 2 and bugfix 2

but we have these :

[image]

bugfix 2.0.x is the most updated one

m00anj00, 2020-01-10:
Sorry you are right.  Last I tried was a little while ago but I tried every
version and v1 versions worked but both v2 versions did not.  As I said I
have now moved to a diamond hotend which though 3 motors only uses 1
extruder and seems ok so far.

LordBBQ, 2020-01-10:
I could potentially try the bugfix tomorrow.

boelle, 2020-01-18:
did you get time to try bugfix ?

LordBBQ, 2020-01-19:
No, having other hot end issues as well. May get time this week.

LordBBQ, 2020-01-22:
Thanks so much, it works now. I need to do more testing to make sure it is fully fixed, but it prints!

boelle, 2020-01-22:
:-)

boelle, 2020-01-26:
does it still print without problems?

LordBBQ, 2020-01-26:
At the moment. Still need to test if the other motor works, but for now it is fine.

boelle, 2020-01-28:
will close this one and called it fixed for now

we can always reopen if there are still gremlins hiding in the closet

closed 2020-01-28 by boelle