MarlinFirmware/Marlin #16179: [BUG] Printer Halts and Requires Restart During Attempted Auto-Home

T-SquaredProductions, 2019-12-10:
### Bug Description
(Continuation of issue #16132)
The Auto-Home does not work properly, and halts the printer, requiring a restart.

### My Configurations

[MARLINCONFIG2.zip](https://github.com/MarlinFirmware/Marlin/files/3946700/MARLINCONFIG2.zip)

### Steps to Reproduce

1. Marlin 2.0.x is required
2. Invert the logic of the endstops, if you have the red endstops with the included noise-reducing capacitor.
3. Go to Motion > Auto Home

**Expected behavior:** The printer raises the Z-axis, and automatically goes through the auto-home sequence

**Actual behavior:** The printer raises the Z-axis, stops, suspends all movement for about 5 seconds, and then throws up a "ERROR: PRINTER HALTED - Please Reset" error message. Pronterface reports a "()kill called!" error.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Patag, 2019-12-10:
Maybe you could ask to @boelle ;-)

T-SquaredProductions, 2019-12-12:
Ok, after looking at someone else's similar problem (Issue #12852), I decided to try activating one of the endstops myself.

I started with the X-axis only, as that's the one the firmware starts with. To my surprise, manually activating the X-axis endstop allowed it to continue normally, and it moved on with the Y-Axis, which actually moved into position properly. However, the next problem came with the Z-axis, as it moved down, then microstepped down (like when the system detects the endstop in autohoming), and threw up the "PRINTER HALTED" message.

I tried again, and the second time created some odd jittery movement in the X-axis, and started to barf random characters on the LCD screen. Furthermore, the system threw up the same message, but then acted as if it had been reset. (I don't remember hitting the reset button on the RAMPS board.)

Is there power or data noise that I'm dealing with here?

karabas2011, 2019-12-13:
The same on X homing from far position.  Arduino with ramps 1.4
Marlin 1.1.x, several printers.
it starts Xhoming then suddenly it looks like endstop triggering. the head moves back (as in config_adv) and moves forward slowly  but no endstop in expected place - immediate kill().

boelle, 2019-12-13:
> Marlin 1.1.x,

1,1,x are no longer actively worked on, use 2.0.x

T-SquaredProductions, 2019-12-15:
Never mind. I found the software filter at "#define ENDSTOP_NOISE_THRESHOLD" :D

closed 2019-12-15 by T-SquaredProductions