MarlinFirmware/Marlin #16181: [BUG] Impossible Flash SKR mini E3 V1.2

Rhteam67, 2019-12-10:
Hello,

I have a problem with my SKR mini E3 V1.2 card.

I compiled a marlin to activate the touchmi successfully, the card has been flashed from an SD card with the file firmware.bin. The printer starts normally no problem.

I now want to switch to a BLtouch, so I compiled a firmware since marlin 2.0 bugfix with VScode.
But when I insert the SD card with the firmware.bin file, the printer starts
but without being flashed with my new firmware, she keeps the old Marlin with touchmi.
- I tried with a new SD card, with only the firmware.bin file
- I reset the EEPROM from the printer in advanced configuration

If you have an idea of ​​why my SKR card does not want to be flashed anymore.

Thank you in advance

Bonjour,

Je rencontre un souci avec ma carte SKR mini E3 V1.2.

J'ai compilé un marlin pour activer le touchmi avec succès,  la carte a bien été flashée depuis une carte SD avec le fichier firmware.bin. L'imprimante démarre normalement pas de souci.

Je veux maintenant passer à un BLtouch, donc j'ai compilé un firmware depuis marlin 2.0 bugfix avec VScode.
Mais quand j'insère la carte SD avec le fichier firmware.bin, l'imprimante démarre 
mais sans être flashé avec mon nouveau firmware, elle garde l'ancien Marlin avec touchmi.
- J'ai essayé avec une nouvelle carte SD,  avec seulement le fichier firmware.bin
- J'ai réinitialisé l'EEPROM depuis l'imprimante dans configuration avancée

Si vous avez une idée du pourquoi ma carte SKR ne veut plus être flashé. 

Merci d'avance

boelle, 2019-12-10:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Rhteam67, 2019-12-10:
Merci,
J'utilise le fichier du site SKR avec bl touch.

Rhteam67, 2019-12-10:
Merci, voici les deux fichiers
[Downloads.zip](https://github.com/MarlinFirmware/Marlin/files/3946900/Downloads.zip)

brew99, 2019-12-10:
What is the default env. you have set in platformio?

What is the firmware.bin size you are trying to flash?

Rhteam67, 2019-12-11:
J'utilise VSCode studio avec platformio 1.10

Le fichier Firmware.bin pèse 216ko

brew99, 2019-12-11:
Unplug your BLtouch and try to flash.  If it works, then check the wiring pin positions of the bltouch servo connector with respect to the mini e3 bird pins.  The many of them have been swapped

Rhteam67, 2019-12-11:
Malgré le bltouch débranché, impossible à flasher la carte. Elle reste sur mon premier firmware.

Rhteam67, 2019-12-11:
I have a reset button on the map, is it useful for something?
And why when I do a reset of the EEPROM, it beep but nothing happens?

Can I flash it from a PC?

thank you

J'ai un bouton reset sur la carte, sert il à quelque chose ?
Et pourquoi quand je fais une réinitialisation de L'EEPROM, elle bip mais rien ne se passe ?

Je peux la flasher depuis un PC ?

merci

brew99, 2019-12-11:
The reset button can be used to reboot the board without turning power to the printer off/on

It’s normal for initializing EEPROM to beep but not back out of the screen

If you connect a STlink device, then you can read/write to the MCU directly.  That is how I changed my bootloader

RedOctobyr, 2019-12-12:
VS Code, but PlatformIO 1.10? I am new at this, but I have PlatformIO 4.1.0, is that the version you mean? 

Does the printer take longer-than-usual to boot up, showing that it's looking at the new firmware file? Longer than if you boot without the SD card inserted? 

To understand if it has to do with your software or configuration, what happens if you try to flash a pre-compiled firmware file? Such as the one from BigTreeTech, here: 
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2
I have used the "firmware-bltouch-for-z-homing.bin" file.

damvcoool, 2019-12-12:
What kind of SD card are you using, SD, sdhc or newer kind? 
Is firmware.bin the only file on the card?

Rhteam67, 2019-12-12:
Yes platformio 4.10 I use forgiveness.

When I put a firmware.bin, it starts quickly without taking the time to read the map.

I use the SD 128 mo card recovers with the SKR in FAT and only the firmware.bin file

I tried with a Sandisk SDHC 128go card and the same, the skr tries quickly without reading the SD card.

When I use the firmware.bin found on https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2 nothing happens, it starts quickly

damvcoool, 2019-12-12:
Does the printer read regular gcode files from the SD card? 
Have you tried updating the printer via USB?

dseven, 2019-12-12:
Note that if the new firmware is not getting loaded off the SD card, it may be a problem with the boot loader, not necessarily with what's on the SD card. May have to contact BTT support for advice. Maybe a good first step would be to try to load the [firmware provided by BTT](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2) - if that loads, your boot loader is OK.

@damvcoool - AFAIK there is no easy way to load firmware on this board without using the SD card - I think it'd require a special programmer (hardware). The method to install over USB involves accessing the SD card via composite USB, so you're still placing the firmware.bin file on the card and then instructing the firmware to reload. See https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/14

Rhteam67, 2019-12-12:
Yes, the printer reads a gcode file on the SD card.

I have just formatted several cards and I turn on the printer with the file firmware.bin but still no flash.

She does not even try to read the map on ignition.

shitcreek, 2019-12-13:
Some boards may need to be put into the bootloading mode. Try holding down the reset button until the second LED flashes red and then primary one stays lit.

boelle, 2019-12-13:
@Rhteam67 tried what @shitcreek said?

Rhteam67, 2019-12-15:
A big thank you to you. I finally managed to flash my card, the solution was to check the SD card from right click property.
But that I compile and flash the card to activate my bltouch, this one does a leveling in the middle and after a bottom left of the board but outside of it.
I have configured my #define NOZZLE_TO_PROBE_OFFSET

Rhteam67, 2019-12-15:
[image]

Rhteam67, 2019-12-15:
[image]

shitcreek, 2019-12-15:
Closing this issue because it looks like it was operator error.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).)

closed 2019-12-15 by shitcreek