MarlinFirmware/Marlin #16182: Cant use 2 heaters on 2.0 or 2.1 branch

m00anj00, 2019-12-10:
Hi
I have been using V1 for a long time on Ramps 1.4 with 2 extruders.  I have decided to move to Arduion Due (3.3v) and now use V2.0 / V2.1.  
My setup for mosfet driving is:

pin -100ohm  -- 1 opto  4   -- 12v
                gnd -- 2 opto  3  -- mosfet (gate)
(100k across gate to gnd mosfet to stop runaway in unknown state)

this uses 13mA and can drive from 1.3v.  mosfet gets 12v gate so full capacity drive with no saturation issues :)

My problem:

I have no issues with  bed or H1 working but H2 does nothing.
I have tried mapping changes to see if it would help but suspect I have identified a different issue too.

M43 gives:

PIN: 07        <unused/unknown>                       Input  = 1
PIN: 08        HEATER_BED_PIN                         protected
.              RAMPS_D8_PIN                           protected
PIN: 09        HEATER_1_PIN                           protected
.              RAMPS_D9_PIN                           protected
PIN: 10        HEATER_0_PIN                           protected
.              RAMPS_D10_PIN                          protected

but my mapping (tried in pins_ramps and pins_ramps_duo) makes no difference.

I tried the obvious (works in 1.1) 
set as motherboard 
  #define MOTHERBOARD BOARD_RAMPS_DUO_EEB
in configuration.h

in pins_ramps.h

#ifndef MOSFET_D_PIN
  #define MOSFET_D_PIN     20
#endif
#ifndef RAMPS_D8_PIN
  #define RAMPS_D8_PIN      9
#endif
#ifndef RAMPS_D9_PIN
  #define RAMPS_D9_PIN      10
#endif
#ifndef RAMPS_D10_PIN
  #define RAMPS_D10_PIN    8
#endif

and then tried in pins_ramps_duo.h (V2):

#undef RAMPS_D8_PIN
#define RAMPS_D8_PIN      9

#undef RAMPS_D9_PIN
#define RAMPS_D9_PIN      10

#undef RAMPS_D10_PIN
#define RAMPS_D10_PIN     8

Not sure if i have missed something obvious but.....

Any advice welcome.

boelle, 2019-12-11:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

m00anj00, 2019-12-11:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3949881/Configuration.zip)

Hi please find attached.

The pin remap issue can be replicated by just cloning a copy of V2 and trying to alter 8,9 or 10 with pin debugging and look at the new assignments (which dont change).

I have also tested for clarity to whether its a misreported change by using an LED and 100ohm resistor then forcing heater on / off and looking which pin goes high,  pin9 never does.

Thanks

Arduino version: 1.8.10
Windows 10

shitcreek, 2019-12-13:
` #define MOTHERBOARD BOARD_RAMPS_EEB` - this should be `BOARD_RAMPS_DUO_EEB` instead.

I'm not too familiar with this board but looking at the `pins_RAMPS_DUO.h`
At the top, you'll see:
 ```
*  Differences between
 *  RAMPS_14 | RAMPS_DUO
 *    A9/D63 | A12/D66
 *   A10/D64 | A13/D67
 *   A11/D65 | A14/D68
 *   A12/D66 | A15/D69
 *       A13 | A9
 *       A14 | A10
 *       A15 | A11
 */
```
then if you scrolled down:
```
#undef TEMP_0_PIN
#define TEMP_0_PIN          9   // Analog Input

#undef TEMP_1_PIN
#define TEMP_1_PIN         11   // Analog Input

#undef TEMP_BED_PIN
#define TEMP_BED_PIN       10   // Analog Input
```

So it looks like pins 9 and 10 are assigned doubly assigned.

m00anj00, 2019-12-13:
Sorry you are right about the motherboard assignment.  I have used the duo
and ramps as I have tested it on both due and mega.  Just got mixed up.
As per the heater ramps, that works surprisingly.  Its d8,d9 and d10 that
wont reap

boelle, 2019-12-13:
so problem solved or did i miss something?

m00anj00, 2019-12-13:
No far from it.  its a bug.  repeatable from any stock V2 branch.  just try to reassign 8,9 or 10 is 1 bug.  also the second heater (pin9) never triggers but on V1 is fine so no hardware issues.  reproducible on mega or due with ramps 1.4 (due is 3v3 so a few hardware mods)

m00anj00, 2019-12-13:
I just ran up a virgin V2 bugfix and tested.  definitely the 2 are bugs.  attached v1 and v2 configs same setup.  V1 works fine, V2 no joy.
[v2config.zip](https://github.com/MarlinFirmware/Marlin/files/3961966/v2config.zip)
[v1config.zip](https://github.com/MarlinFirmware/Marlin/files/3961967/v1config.zip)

boelle, 2019-12-24:
is the issue still the same with all the updates in the last 11 days?

m00anj00, 2019-12-24:
Yep.  Tbh I've since moved to a diamond hotend, but it's easy to
replicate.  Just use a ramps and try to move pins 8,9 or 10 and look at
pins debugging.  I can still easily test the heater issue too if nobody
else is willing to.

Last build I tried was 2 days ago and still the same.  On 1.1 it's fine but
2.0 doesn't allow changes to pins.

boelle, 2019-12-24:
why move pins? there are several ramps board and one of them is for 2 heaters

m00anj00, 2019-12-24:
At the time I needed to test the heating issues but it exposed a second bug

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

m00anj00, 2019-12-25:
Tbh you can test this with just a mega or due.  Just use a 1k resistor
between gnd and the pin you are using.  Also using pronterface you can use
a pins debugging output and it shows no movement of revamped pins

LordBBQ, 2019-12-26:
@m00anj00 What if you set your extruders to 1 only? This let me start printing again?

m00anj00, 2019-12-27:
I'll try.  Nice find

LordBBQ, 2019-12-27:
Still not satisfied, as I want dual extruders, but at least you can actually turn your 3d printer from a desk ornament to a somewhat functioning machine. May not affect you as your issue is with heaters, not steppers, but it is worth a try.

boelle, 2020-01-08:
@m00anj00 is this still an issue or ddid you figure the heater problem out?

m00anj00, 2020-01-08:
Sorry no still no good.  I have now abandoned and moved to diamond hotend
instead.

boelle, 2020-01-24:
will close this one

we can always reopen if someone have same hardware and able to test from time to time

LordBBQ, 2020-01-24:
Sure. Thanks for a all your help.

closed 2020-01-24 by boelle