MarlinFirmware/Marlin #16183: Begins to print not in the center.

RonanMtm, 2019-12-10:
All hi!

Help to understand.
1. Anet A8 printer
2. Area of the press 220*220*140
3. Payment Anet V1.7
4. Insertion of Marlin 1.1.9
5. Ultimaker Cura 4.4, Arduino-1.8.5

The problem consists in the following:

1. When sending knots to HOME the parking occurs as has to be in the left bottom corner of a table.

2. I establish the printed object on the center of a table in Cura, and when sending a task for the press, begins to print not in the center, and on the periphery i.e. if we imagine a table 220*220mm., it begins to print a detail to 1\4 right top areas.
And in any way not on the center.

3. Edited an insertion itself (I am not an expert), but at forums looked at what parameters what are responsible for and that can be changed for positioning change, changed on a miscellaneous of value, places of positioning change, and the press begins not from the center of a table again, and again in the right top area +-the 1-2nd. I would not specify shifts, what sizes in an insertion.

Help to understand please where and that should be replaced in a configuration of Marlin 1.1.9 that was printed as it is necessary in the center of a table.   

Всем привет!

Помогите разобраться.
1. Принтер Anet A8
2. Область печати 220*220*140
3. Плата Anet V1.7
4. Прошивка Marlin 1.1.9
5. Ultimaker Cura 4.4, Arduino-1.8.5

Проблема заключается в следующем:

1. При отправке узлов в HOME парковка происходит как должно быть в левый нижний угол стола.

2. Устанавливаю печатаемый объект по центру стола в Cura, и при отправке задания на печать, начинает печатать не в центре, а на периферии т.е. если мы представим себе стол 220*220мм., то печатать он начинает деталь в 1\4 правой верхней области.
И никак не по центру.

3. Прошивку редактировал сам (я не специалист), но на форумах посмотрел какие параметры за что отвечают и что можно поменять для изменения позиционирования, менял по разному значения, места позиционирования меняются, а печать опять начинается не из центра стола, а опять в правой верхней области +- 1-2см. смещения, какие бы размеры я не указывал в прошивке.

Помогите пожалуйста разобраться, где и что надо сменить в конфигурации Marlin 1.1.9 чтобы печаталось как положено в центре стола.   
[MY Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3946872/MY.Configuration.zip)

m00anj00, 2019-12-10:
Hi
If im reading this right, you have set the bed size and shape in machine settings and selected origin at center?
also, i assume you chose the correct Gcode flavour?
on the next tab (extruder) there are 2 settings also, x offset and y offset.  they shoud be 0.  your filament size is also needed here.

I have had a look at origin at center and get the same results.  useless for me. as it tries to print an object in the top right corner.

untick origin at center and instead use x and y offsets to avoid a part of the bed you dont want to (for some reason) use.  that allows objects to be right clicked and select center object instead.

hope that helps

Vertabreak, 2019-12-10:
origin at center is incorrect setting for a non delta.

boelle, 2019-12-10:
will close as the config is wrong and not a marlin bug

you should not use org at center in the firmware nor in the slicer, but just place what you want to print in the center of the bed in slicer,

closed 2019-12-10 by boelle

RonanMtm, 2019-12-11:
I ask for help in Marlin configuration.
You what you advise?
Understood nothing.
Show please where a mistake in my file configurations.