MarlinFirmware/Marlin #16184: [BUG] S-Curve Acceleration Not Working With Junction Deviation

KlausHans, 2019-12-10:
### Bug Description

I upgraded Marlin from some older 2.0.x-bugfix release to the 2.0 release yesterday. In the 2.0 release junction deviation is enabled ootb, so i left it enabled. In the older bugfix release i used it was not. 
But apparently with junction deviation enabled the s-curve acceleration feature doesn't seem to work anymore. I just configured classic jerk and now s-curve acceleration works again as expected.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3947072/Configuration.zip)

### Steps to Reproduce

1. Flash latest 2.0 release of Marlin with junction deviation and s-curve acceleration enabled
2. Do a testprint or so, you should notice s-curve acceleration isn't working
3. Flash latest 2.0 release of Marlin without junction deviation but with s-curve acceleration enabled
4. Do a testprint or so, you should notice s-curve acceleration is working

**Expected behavior:** 
S-curve acceleration is working with junction deviation enabled.

**Actual behavior:**
S-curve acceleration is not working with junction deviation enabled.

#### Additional Information
Linear advance is also enabled in all cases.

Vertabreak, 2019-12-10:
been using both these features together for over a year in marlin 2.0 with no issue and i have a clone of your board.

boelle, 2019-12-10:
remember to calculate junction deviation correct

http://blog.kyneticcnc.com/2018/10/computing-junction-deviation-for-marlin.html

KlausHans, 2019-12-11:
Just activated junction deviation again. The printer was again a lot noisier and vibrated more. I missed the beautiful elegance of s-curve acceleration.
My junction deviation factor is calculated (Jerk 10, Acc. 2000).

KlausHans, 2019-12-11:
An idea: maybe the implementation of s-curve was changed during the work on 2.0 so that the curve is steeper and my expectations are just wrong?

boelle, 2019-12-11:
not calling you stupid or anything here (i'm sometimes bad at picking words)

but maybe explain what you mean by not working and what you would expect? it could be that its working fine and you just expect something different

KlausHans, 2019-12-11:
With working s-curve acceleration i mean: The printhead moves smooth back and forth if printing solid layers. The printer doesn't vibrate much and is quiet. If i disable junction deviation this is the case.

With not working i mean: The printhead moves abruptly. The printer vibrates a lot and makes rattling noises (not catastrophic) when the printheadmoves are short. If i enable junction deviation (and leave all other parameters untouched) this is the case.

Maybe i can make a short video the next days.

KlausHans, 2019-12-11:
Here we go:
junction deviation enabled (`//#define CLASSIC_JERK`): https://www.dropbox.com/s/cu1lhio3e5c69i1/Video%201%20JD%20on%202nd%20Layer.mp4?dl=0
junction deviation disabled (`#define CLASSIC_JERK`): https://www.dropbox.com/s/f82utduiahcz5q0/Video%202%20JD%20off%202nd%20Layer.mp4?dl=0

Another layer:
junction deviation enabled: https://www.dropbox.com/s/cjunm4q0to16jvt/Video%203%20JD%20on.mp4?dl=0
junction deviation disabled: https://www.dropbox.com/s/pu0fft0skza4wab/Video%204%20JD%20off.mp4?dl=0

The sound is very important. In the videos with junction deviation disabled the moves are way smoother and thus the printer is quieter. 
I payed attention that the position of my phone was always the same.

As you can see it is not the exactly the same gcode, i only changed the minimum travel after retraction to tackle down another problem. This has nothing to do with the issue of this ticket.

I hope this helps.

WireBe, 2019-12-14:
@KlausHans Thank you for pointing this. I compiled without junction deviation, and i just saw my printer going sooo fast and print like butter without any noise!!  definively there is something going weird with this

Vertabreak, 2019-12-14:
watched all 4 videos and i honestly don't see any difference or even notice any major sound difference.

KlausHans, 2019-12-14:
Glad i am not the only one.
@Vertabreak the question is not how big the difference is. There shouldn't be any difference (your "major" implies that you actually noticed a difference). It's not easy to catch on camera/microphone. It's not visible in the videos, only audible. And you have to set your volume high.

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

qwewer0, 2020-01-03:
Confirmed for my Ender 3, SKR Mini E3 v1.2, Marlin 2.0.1, LA K 0.4 - 1.4, S-Curve, Junction deviation on/off.
With the classic jerk (JD off) the print acceleration and deceleration is so smooth, ergo S-curve works.

yeowjoon99, 2020-01-05:
I definitely don't have the same hardware as OP but I've got the same problem on my corexy printer.
From what I can tell, the majority of the vibrations come from the E stepper which uses an LV8729 driver.

With JD and LA enabled, there's an incredible vibration as if the bearings on the carriage are extremely worn. 
Doesn't make a difference whether S-Curve is on or not.

If I enable CLASSIC_JERK I'd have to adjust the jerk for XY and E to around 15 or more to replicate the same level of vibration.
Changing acceleration between 800 to 2500 doesn't make any difference to the vibrations either.

My guess is either that because I'm using a bowden setup the LA value is too high and paired with JD it causes a very jerky extruder.

PaschyS, 2020-01-13:
Hello Guys, I think I have the same or similar problem with my SKR Pro 1.1.  since Christmas Day to find out for my Printer good settings.
 
So, I use instead of classic_jerk, junction_deviation with the calculations, but I see no difference between the prints. Today I tested some configurations in the bugfix firmware(01-11-2020):
Before I use the bugfix firmware (12-21-2019) with the same issue.
(12-21-2019):
-stepper drivers = tmc2209 uart
-S_CURVE_ACCELERATION = off
-JUNCTION_DEVIATION= on calculation 0.053
-SQUARE_WAVE_STEPPING = on 
- external eeprom over spi
(01-11-2020):
-stepper drivers = tmc2209 uart
-S_CURVE_ACCELERATION = on
-JUNCTION_DEVIATION= on (values from 0.02,0.04,0.053,0.06,0.08 tested) my calculation 0.053
-SQUARE_WAVE_STEPPING = on 
- external eeprom over spi

[image]

boelle, 2020-01-30:
a lot of updates and fixing has happend in the last week, is the problem still there?

PaschyS, 2020-01-30:
So, I've tested today(30 January) the normal Bugfix version and normal jerk works fine. I will test it soon with JD, square wave and scurve. The version from lino makes now problems with corners for the skr pro(bulge/bleeding).

boelle, 2020-01-30:
others versions and forks does not matter as we need to focus on main marlin

PaschyS, 2020-01-30:
Yeah I know it. 
And 3D Print friend from me have the skr 1.3 with the same Bugfix version like me and he has no problems with linear adcance and scurve.

boelle, 2020-01-30:
also same extruder?

boelle, 2020-01-30:
everything has to be the same to compare :-) or almost everything

boelle, 2020-01-30:
but @KlausHans has not reported back and it was 1 month since i asked him, so issue it most likely solved

PaschyS, 2020-01-30:
> also same extruder?

Yes, same Extruder(Hemera). So, I think you can close this report. If anything goes wrong I will write it here.

boelle, 2020-01-31:
yes and we can always reopen it

closed 2020-01-31 by boelle