MarlinFirmware/Marlin #16185: [BUG] error: 'EN_ENABLE_READ' was not declared in this scope

lostcontrol, 2019-12-10:
### Bug Description

Since 776632c503, i cannot compile Marlin anymore. I get the following error:
```
In file included from /tmp/arduino_build_953141/sketch/src/inc/../core/boards.h:24:0,
                 from /tmp/arduino_build_953141/sketch/src/inc/MarlinConfigPre.h:35,
                 from /tmp/arduino_build_953141/sketch/src/inc/MarlinConfig.h:28,
                 from /tmp/arduino_build_953141/sketch/src/Marlin.h:24,
                 from /tmp/arduino_build_953141/sketch/src/Marlin.cpp:31:
/tmp/arduino_build_953141/sketch/src/Marlin.cpp: In function 'void manage_inactivity(bool)':
/tmp/arduino_build_953141/sketch/src/Marlin.cpp:556:51: error: 'EN_ENABLE_READ' was not declared in this scope
           #define _CASE_EN(N) case N: oldstatus = E##N_ENABLE_READ(); enable_E##N(); break;
                                                   ^
/tmp/arduino_build_953141/sketch/src/inc/../core/macros.h:402:24: note: in definition of macro 'EVAL1'
 #define EVAL1(V...)    V
                        ^
/tmp/arduino_build_953141/sketch/src/inc/../core/macros.h:401:30: note: in expansion of macro 'EVAL1'
 #define EVAL2(V...)    EVAL1(EVAL1(V))
                              ^~~~~
/tmp/arduino_build_953141/sketch/src/inc/../core/macros.h:400:24: note: in expansion of macro 'EVAL2'
 #define EVAL4(V...)    EVAL2(EVAL2(V))
                        ^~~~~
```

If I revert back to `776632c503^`, everything compiles properly. I'm using Arduino 1.8.10 (same problem with 1.8.9) on Linux 64bit. My configuration is self made for a CR-10 Mini (Sanguino board).

Looking at the error, it looks like `E##N_ENABLE_READ` does not get substituted properly. Maybe some bug in the version of GCC shipped with Arduino (avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++)?

### My Configurations

You can find my config (and the actual branch I try to compile) [here](https://github.com/lostcontrol/Marlin/tree/cr10-mini-2.0.x/Marlin).

### Steps to Reproduce

1. git clone https://github.com/lostcontrol/Marlin.git -b cr10-mini-2.0.x
2. Open the project in Arduino and try to compile

boelle, 2019-12-11:
tried to use vscode+platformio ?
tried arduino ide 1.9 beta?

ellensp, 2019-12-11:
I tried vscode+platformio  on the supplied code...  same error.    The issue seems to be that N isnt known till run time so E##N resolves to EN instead of E0 or E1 etc...  but my brain cant see how to fix this, need someone with more experience. But very keen to see the solution.

randellhodges, 2019-12-11:
I can confirm this is a code issue. Looking at power.cpp and controllerfan.cpp, they reference `E##N##_ENABLE_READ`.

I changed that line in Marlin.cpp to `E##N##_ENABLE_READ` and it compiled.

lostcontrol, 2019-12-11:
Thank you all for looking at this and fixing it.

shitcreek, 2019-12-15:
Looks like this issue has been resolved.

closed 2019-12-15 by shitcreek