MarlinFirmware/Marlin #16186: Oops! Select 'Sanguino' in "Tools > Boards' and 'ATmega644P' or 'ATmega1284P' in 'Tools > Processor.

LilJashy, 2019-12-11:
Bug Description

Repeatedly getting error Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../pins/sanguino/pins_SANGUINOLOLU_11.h:56:4: error: #error "Oops! Select 'Sanguino' in 'Tools > Boards' and 'ATmega644P' or 'ATmega1284P' in 'Tools > Processor.'" when trying to build. Going to be using board SKR v1.3, so the only thing I've altered at all is the platformio.ini file. I'm also posting this to the platformio github as well.

### My Configurations

Config files are not changed at all. Issue is happening both in Bugfix 2.0 and Marlin 2.0

### Steps to Reproduce

1. Change default_envs to LPC1768
2. Click build
3. Fail

**Expected behavior:** [What you expect to happen]

It to build successfully

**Actual behavior:** [What actually happens]

Failure

ellensp, 2019-12-11:
You NEED to set motherboard BOARD_BIGTREE_SKR_V1_3 in Configuration.h

ellensp, 2019-12-11:
Also the default motherboard is a BOARD_RAMPS_14_EFB, from the error you have provided you clearly have this set to something with sanguino processor.  so either you changed it and didn't tell us, or you didn't get your source from the marlin repository.

kaspars-ozols, 2019-12-11:
Could it be that you just grabbed example config files from Creality folder (`Marlin\config\examples\Creality\Ender-3`)?

Since you are using SKR board, you should use config files from BigTreeTech (https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3).

ellensp, 2019-12-11:
@kaspars-ozols 
I disagree strongly. If you want marlin you come to the MarlinFirmware branch, not a badly maintained branch that is often months behind.

kaspars-ozols, 2019-12-11:
@ellensp 
I did not say to use firmware, but just config files just to get things going. It can be quite frustruating for novice users to configure Marlin from scratch.

In any case, it turns out that BigTreeTech don't even publish config files for SKR 1.3 and one needs to do it from scratch anyway. I updated my comment to be more specific about that.

boelle, 2019-12-11:
@LilJashy you did not read the issue template, it clearly says to add the configs

how are we supposed to help without a config that tells much about your setup?

also the title is mess, it does not say if this is bug or feature request

until the config is provided this "issue" will be held closed

closed 2019-12-11 by boelle

Felipe-NS, 2020-04-18:
> @ellensp
> I did not say to use firmware, but just config files just to get things going. It can be quite frustruating for novice users to configure Marlin from scratch. Otherwise, I fully agree with you.
> 
> In any case, it turns out that BigTreeTech don't even publish config files for SKR 1.3 and one needs to do it from scratch anyway. I updated my comment to be more specific about that.

I'm having the same error using when compiling for original ender 3 board.
I'm using the config files like you said