MarlinFirmware/Marlin #16189: [BUG]Steps per mm changing between firmwares

hussainsail2002, 2019-12-11:
### Bug Description

I was using an older version of marlin 1.1.9 and my extruder seemed to be working fine, I recently changed to a new firmware and my extruder is now extruding 3 times the amount of filament for the same E steps value

Everything with respect to the hardware is exactly the same and only when the firmware is changed I run into this problem

My other axis seem to be moving fine , the only issue is with the extruder
 
### My Configurations

I am attaching my config files here.

### Steps to Reproduce

1. Download the latest marlin firmware, 
2. Check the extruder steps per mm after making changes

**Expected behavior:** 

Expected the extruder to extrude the same amount of filament

**Actual behavior:** [What actually happens]

The extruder is extruding far more filament that it should be, I am guessing I have missed something in the firmware, my stepsticks and jumper pins are fine, not sure what else contributes to change in the steps per mm value
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3949443/Configuration.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

hussainsail2002, 2019-12-11:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3949446/Configuration.zip)

ellensp, 2019-12-11:
You had eeprom enabled. You probably had the correct setting in eeprom not in configuration.h. The new version can't read the old eeprom settings so used Configuration.h

hussainsail2002, 2019-12-11:
but the eeprom also had the same e steps which is 92 steps per mm so why would it change ?
Also both config files have the same e step values ! so why would the extruder extrude 3 times the amount ?

hussainsail2002, 2019-12-11:
is it possible that the micro stepping has changed some where and I might not aware of it ?

Cause the jumpers on the board are good and my old firmware prints fine , so I know that okay. maybe the firmware is overriding the settings of the micro stepping value ?

ellensp, 2019-12-11:
Cant change micro stepping in firmware on your controller BOARD_MKS_13
Only by changing the jumpers or replacing stepper drivers (different drivers can have different micro stepping settings)

tatusah, 2019-12-11:
The older config you uploaded seems to be Marlin version 1.1.0 and the newer for 1.1.9. Maybe just jump on the version 2.0.x & calibrate steps if needed.

hussainsail2002, 2019-12-11:
@ellensp I changed the stepper drivers as well and checked the jumpers they seem to be fine

@tatusah I was hoping both the configs can be similar

ellensp, 2019-12-11:
If you changed from A4988 to DRV8825 and kept the same jumpers for example this will double the micro stepping.

boelle, 2019-12-11:
this is most likely a user error so closing

closed 2019-12-11 by boelle