MarlinFirmware/Marlin #16192: [BUG] Build error with VS Code on the LPC1768 platform (SKR V1.3) with CR10S config file

Kaliin, 2019-12-11:
### Bug Description

I tried to build for the LPC1768 platform with CR10s configuration files(I just got an SKR V1.3) but it fails.

### My Configurations

Clean Windows 10 install (like 5 days ago), I also installed WinRAR.
I made all the necessary step to install VS Code and the platformio extension.
I tried Marlin2.0 downloaded from github, from the website and the bug-fix.
I tried to also compile from a not modified Marlin 2.0 (like just unzip it, and try to compile for the atmega platform but same problem appear)

My configuration files (Configuration.h, Configuration_adv.h and platformio.ini): 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3950397/Configuration.zip)

### Steps to Reproduce

1. Install VS Code and the platformio extension
2. Download Marlin 2.0.x from github and unzip it
3. Remove Configuration.h and Configuration_adv.h and copy and paste Marlin-2.0.x\config\examples\Creality\CR-10S content to Marlin-2.0.x\Marlin
4. Modify default_envs in platformio.ini to default_envs = LPC1768
5. Modify Configuration.h for the SKR v1.3 platform (change Serial_port, Serial_port_2 and motherboard)
6. Build
7. Fail :(

#### Additional Information

My terminal output: 
[Terminal Output.txt](https://github.com/MarlinFirmware/Marlin/files/3950430/Terminal.Output.txt)

Or with a simple copy and paste:

> Executing task in folder Marlin-2.0.x: C:\Users\sanfi\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1768; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
LibraryManager: Installing U8glib-HAL
Using cache: C:\Users\sanfi\.platformio\.cache\a7\b7a9b473d1c239d9843ca943887cd0a7
UnsupportedArchiveType: Can not unpack file 'c:\Users\sanfi\Downloads\Marlin-2.0.x\.pio\libdeps\LPC1768\_tmp_installing-jhks8mso-package\None':
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\sanfi\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\sanfi\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\sanfi\.platformio\platforms\nxplpc-arduino-lpc176x\builder\main.py", line 148:
    target_elf = env.BuildProgram()
  File "C:\Users\sanfi\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\sanfi\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 480:
    self.unpack(dlpath, tmp_dir)
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 218:
    with FileUnpacker(source_path) as fu:
  File "C:\Users\sanfi\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 133:
    raise exception.UnsupportedArchiveType(self.archpath)
======================================================================= [FAILED] Took 1.02 seconds =======================================================================

Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
megaatmega1280                IGNORED
rambo                         IGNORED
FYSETC_F6_13                  IGNORED
sanguino_atmega644p           IGNORED
sanguino_atmega1284p          IGNORED
melzi                         IGNORED
melzi_optiboot                IGNORED
at90usb1286_cdc               IGNORED
at90usb1286_dfu               IGNORED
DUE                           IGNORED
DUE_USB                       IGNORED
DUE_debug                     IGNORED
LPC1768                       FAILED    00:00:01.021
LPC1769                       IGNORED
STM32F103RC                   IGNORED
STM32F103RC_fysetc            IGNORED
STM32F103RC_bigtree           IGNORED
STM32F103RC_bigtree_USB       IGNORED
STM32F103RC_bigtree_512K      IGNORED
STM32F103RC_bigtree_512K_USB  IGNORED
STM32F103RE                   IGNORED
STM32F103RE_bigtree           IGNORED
STM32F103RE_bigtree_USB       IGNORED
STM32F4                       IGNORED
STM32F7                       IGNORED
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_lite                IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
include_tree                  IGNORED
================================================================= 1 failed, 0 succeeded in 00:00:01.021 =================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

boelle, 2019-12-11:
please use pastebin.com for error logs

ellensp, 2019-12-11:
You say all the right stuff, but the config file provided isn't modified for a skr1.3

boelle, 2019-12-11:
have you tried the cleanup button?

[image]

and as @ellensp says using the correct config might help too :-)

Kaliin, 2019-12-11:
Sorry, I sent the wrong files, but the one I compile where modified!

ellensp, 2019-12-11:
The error listed is definitely something to do with your os...  unsupportedArchiveType: Can not unpack file 'c:\Users\sanfi\Downloads\Marlin-2.0.x\.pio\libdeps\LPC1768\_tmp_installing-gx773gst-package\None':    platformio has to be allowed to download files..  do you have  something blocking files?  overly protective antivirus?    and remove winrar in case it is causing issues.  Windows can unzip zip files without it.

boelle, 2019-12-11:
attach the correct one so we have the same "picture" as you

Kaliin, 2019-12-11:
It updated the files, I believe that I closed the Windows anti-virus or protection system, I will check that and desinstall winRar, I keep you updated.
Thank you!

boelle, 2019-12-11:
if you use the build in windows antivirus it will not block it, it does not do on win10 home 64 bit

boelle, 2019-12-11:
but yes winrar for zip files are not needed, and you can use free online extractors for most other packed files

http://online.b1.org/online

supports: b1, zip, jar, xpi, rar, 7z, arj, bz2, bzip2, tbz2, tbz, txz, cab, chm, chi, chq, chw, hxs, hxi, hxr, hxq, hxw, lit, cpio, deb, gz, gzip, tgz, tpz, hfs, iso, lzh, lha, lzma, rpm, tar, xar, z, taz, xz, dmg, cb7, cbr,
cbt, cbz

Kaliin, 2019-12-11:
So, winRAR uninstalled, Windows security totally disabled, same error !
What format is it? Can I circumvent what it try to download? (like downloading and unzipping and pasting it where it should go)
Or is there something else that I should download? Like python or pip?

boelle, 2019-12-11:
vs code should download everything you need

did you try to click the trash can?

https://user-images.githubusercontent.com/3832239/70624602-e8e3dd00-1c20-11ea-8666-821515834337.png

boelle, 2019-12-11:
or delete these 2 folders manual

[image]

Kaliin, 2019-12-11:
Yes I tried it and it succeded, this is the terminal output: https://pastebin.com/YL4Q27ab

I delete both of these folders, and fails again, and in Marlin-2.0.x\.pio\libdeps\LPC1768 there is nothing same with Marlin-2.0.x\.pio\build\LPC1768
(but there is a project.checksum in Marlin-2.0.x\.pio\build )

ellensp, 2019-12-11:
also try running in verbose mode . To do this click on the console icon (last one on the right of boelle's image and type "platformio run -v"  <enter> It may tell you more about what is going on

Kaliin, 2019-12-11:
Hmm interesting it showed 3 other errors saying "Platform incompatible library"

https://pastebin.com/V5pyk6cF

ellensp, 2019-12-11:
himm try deleting the directory C:\Users\sanfi\.platformio and try again

Kaliin, 2019-12-11:
This seems to do it, it's building !
So apparently just deleting the .platformio made it reinstall.
I read that it can take some time to build, it has downloaded the U8glib-HAL (up to 100%) and now it just says Building. 
I will wait and if it works than I close the issue!
Thank you both!

Kaliin, 2019-12-11:
It got stuck but after closing and reopening VS code, the building went smoothly!
Thank you!

closed 2019-12-11 by Kaliin