MarlinFirmware/Marlin #16195: [BUG] M503 output uses old X/Y format for Mesh Bed Leveling

dseven, 2019-12-11:
### Bug Description

M503 output (2.0.x-bugfix) shows X/Y values in the `G29 S3` commands for Mesh Bed Leveling, but 2.x requires the I/J format. Attempting to paste the commands from the M503 output after a M502 reset results in error `J not entered.`

### My Configurations

https://github.com/dseven/Marlin/commit/b01bfe3ba378bb74ee656cf919987b7285d2d9b4

### Steps to Reproduce

1. Setup mesh bed leveling as usual
2. M500
3. M503

**Expected behavior:** 

```
...
Mesh Bed Leveling:
  M420 S0 Z0.00
  G29 S3 I0 J0 Z0.07500
  G29 S3 I1 J0 Z0.07500
  G29 S3 I2 J0 Z0.05000
  G29 S3 I0 J1 Z-0.02500
  G29 S3 I1 J1 Z0.00000
  G29 S3 I2 J1 Z0.02500
  G29 S3 I0 J2 Z0.02500
  G29 S3 I1 J2 Z0.10000
  G29 S3 I2 J2 Z0.10000
...
```

**Actual behavior:**

```
...
Mesh Bed Leveling:
  M420 S0 Z0.00
  G29 S3 X1 Y1 Z0.07500
  G29 S3 X2 Y1 Z0.07500
  G29 S3 X3 Y1 Z0.05000
  G29 S3 X1 Y2 Z-0.02500
  G29 S3 X2 Y2 Z0.00000
  G29 S3 X3 Y2 Z0.02500
  G29 S3 X1 Y3 Z0.02500
  G29 S3 X2 Y3 Z0.10000
  G29 S3 X3 Y3 Z0.10000
...
```

#### Additional Information

Offending code appears to be: https://github.com/MarlinFirmware/Marlin/blob/cb7817093fbe0ea7a3a2bac06f38d8bd66ba99d2/Marlin/src/module/configuration_store.cpp#L2872-L2878

I may attempt a PR with a fix for this. While there, I think that `G29 S4 Zx` (global offset) should be added to the output too.

randellhodges, 2019-12-12:
Check out `G29 S-1`

dseven, 2019-12-12:
> Check out `G29 S-1`

Not sure what I'm looking for here? That gives me the current status, but not in a format that I can paste in as commands to put that status back after a reset....

randellhodges, 2019-12-12:
Sorry, I read right past the "Mesh Bed Leveling" part and went right to UBL. Disregard.

boelle, 2019-12-13:
since PR was merged i assume this is fixed and closing it

PR #16199 (2019-12-12): M503 Mesh Bed Leveling 2.x format

### Description

The Mesh Bed Leveling output in M503 was still using the 1.x format (X/Y
coordinates). This updates it to use the I/J coordinates, as needed to
paste the configuration, e.g. after a M502 reset.

Also adds G29 S4 (Z offset) to the output.

### Benefits

Output from M503 may be captured and stored, then used later to restore MBL after any loss (intentional or not) of EEPROM data.

### Related Issues

Fixes: #16195
closed 2019-12-13 by boelle