MarlinFirmware/Marlin #16196: [BUG] Serial Communication with Octoprint stalls for several seconds

T69, 2019-12-11:
### Bug Description

Since several weeks I get more and more timeouts in the communication with Octoprint. From Octoprints serial log:

Send: N41606 G1 X126.156 Y113.389 E0.0144*80
Recv: ok
Send: N41607 G1 X125.801 Y113.565 E0.0102*95
Recv: ok
Send: N41608 G1 X125.530 Y113.657 E0.0074*93
Recv: ok
Send: N41609 G1 X125.530 Y116.011 E0.0607*95
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N41610 M105*37
Recv: okok T:200.24 /200.00 B:65.01 /65.00 @:73 B@:22
Send: N41611 M105*36
Recv: ok T:200.35 /200.00 B:65.00 /65.00 @:71 B@:23

As you can see, Octoprint waits for a confirmation, but it doesn't get the "ok" for some reason, so the print stalls. After 5 seconds (Octoprint just continues), we see the missing "ok" in the next received message, which is "okok" ...

### My Configurations

Arduino Due with a modified RAMPS 1.5

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Not really reproducable, but it came with the latest 2.0 and the 2.0-bugfix branch ...

**Expected behavior:**
Stable communication

**Actual behavior:** 
Communication errors, which stall the print ...

#### Additional Information

I've change every single bit of hardware ( Due, Raspberry, Cable, Power Supply) and  upgraded Octoprint, but the problem doesn't go away unfortunately ...

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

m00anj00, 2019-12-11:
Not sure if it helps, i found that raspberry pi 3 is minimum for printing or the gcode stalls (I have a 2, and a few 1 ediitions).  you are running a higher microstep and faster printer than me.  could be your pi not your printer.  try with a faster PC first.  I found that an old laptop I have (dual core) is too slow and causes the odd stall.

T69, 2019-12-12:
It was already working without issues ... I've replaced the TMC2208 with TMC2100 on xyz and LV8729 on e ... This was my last change before the stalls appeared ... Testing at the moment...

T69, 2019-12-12:
Changing the drivers didn't solve the problem unfortunately ... That would have been too easy ...

italocjs, 2019-12-12:
I know you probably did it already, but try changing to a new, high quality usb cable (preferably with those ferrite things).  this usually is the first thing to cause issues for me, (i had the exact same problem in the past)

T69, 2019-12-12:
I did already, but I´m going to try another again...
I played around with Marlins serial buffer settings, but that didn‘t change anything ...
Next test will be to revert to an archived Marlin 2.0 bugfix version from June and see what happens...

shitcreek, 2019-12-12:
I found this issue over at Octoprint: https://github.com/foosel/OctoPrint/issues/1578
It might be what you are experiencing - basically make sure there is no electrical interference.

karabas2011, 2019-12-13:
Why don't you use hardware serial? I use it on my sbase and now I can reboot sbase without connection lost. In case of USB - serial disappears during MCU reboot.

T69, 2019-12-13:
Ok, seems to be interference ... I've rerouted the cables to separate Stepper / Heatbead cables far away from the USB cable. Added a ferrit ring to the USB cable and cut the power cable. Finally I switched from Due's programming port to the native one ...
Printing stable without timeout for 2 hours now ... finger's crossed.
Thanks for all the suggestions, guys!

closed 2019-12-13 by T69