MarlinFirmware/Marlin #16200: [FR] [help-wanted] (Watchdog for hot nozzle without extrusion)

italocjs, 2019-12-12:
### Description
The problem is simple, if the hotend stay hot for too much time without extruding, the plastic (abs, pla, any plastic) will degrade and possible carbonize, clogging the nozzle tip.  to prevent it, we could create a watchdog that check if there was any movement on the extruder on the last x seconds, if not, it would cool down the nozzle to a safe temperature or even turn it off. 

I'm a beginner at programming, i really have the will, but no idea on where to start writing this code, i need some help to do it.  at least to point me in the right direction

### Feature Workflow
1. Watchdog monitors the amount of time passed and the extrusion lenght on this time
2. If there was any movement dont do anything, if not turn the nozzle temperature to a value set in configurations.h (something like #anticlog_timeout_time  and #anticlog_timeout_temperature) 

#### Additional Information
I know this feature exists on some makerbot machines (Z18 and Method), but never seen it anywhere else.  This problem (forgetting the nozzle hot) when preheating is very common,

thisiskeithb, 2019-12-17:
This would be a good feature to have built into Marlin. If you run OctoPrint, the [HeaterTimeout](https://plugins.octoprint.org/plugins/heatertimeout/) plugin does this well.

italocjs, 2019-12-17:
I'll try learning how the marlin code works these holidays and build this code.   I have used octoprint before, but it isnt gonna be possible to do on all printers.

rado79, 2019-12-28:
EXTRUDER_RUNOUT_PREVENT does this job for me

rado79, 2019-12-28:
EXTRUDER_RUNOUT_PREVENT does that job for me.

thisiskeithb, 2020-06-24:
Closing since it was implemented in https://github.com/MarlinFirmware/Marlin/pull/16362.

closed 2020-06-24 by thisiskeithb