MarlinFirmware/Marlin #16208: [FR] Ability to configure 1 to 7 preheat settings

AJMartel, 2019-12-13:
### Description
I have added the ability to store up to 7 preheat settings.
In my case PLA, ABS, HIPS, PP, PET, FLEX and NYLON

### Feature Workflow

1. You can set 1 to 7 different preheat settings in the Configuration.h file
2. Under the menu you can access the preheat functions under 3 sub menus
     - Temperature
     - Configure
     - UBL

#### Additional Information

I  have implemented the changes and wanted to see if anyone else finds this useful.
A link [bugfix-2.0.x-addtional-preheat-settings](https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...AJMartel:bugfix-2.0.x-addtional-preheat-settings)

m00anj00, 2019-12-14:
Thanks buddy.  sounds useful

AJMartel, 2019-12-14:
I find it useful. I still have a bug in the menu_configuration.cpp file.
The last menu item steams to have a mind if its own. (hard to describe, kind of a must see to understand).
Without changes to "menu_configuration.cpp", I think it is almost ready to be merged, I use it.

AJMartel, 2019-12-23:
Tried a slightly different approach based on [Rainer-M's pull request v1.1](https://github.com/Rainer-M/Marlin/commit/1460baba4709ddace84abab31db461cbf4e10cf0). It is the second of my 2 links.

closed 2020-06-30 by github-actions[bot]

thinkyhead, 2020-07-03:
Marlin now (2.0.6) supports several preheat options. But soon the plan is to make the material name one of the configurable items in each material preset and slightly extend the edit menu.