MarlinFirmware/Marlin #16210: [BUG] TMC2130 Doesn't work when using PS_ON pin

safaenet, 2019-12-13:
### Bug Description
I switched from DRV8825 to TMC2130 (IdeaFormer) on all my axis (X, Y, Z, E0). But i couldn't make them move at all. Finally found out that Power Supply must be ON before the USB is connected. I'm using an ATX power supply and really need to have this feature on my Printer. Is there a way to solve this problem ? I have another DIY printer and it uses TMC2130 on X and Y (FYSTEC) but it doesn't have this problem. Is there a problem with the manufacturer ?
Marlin 1.1.9
Ramps 1.4

safaenet, 2019-12-13:
Just wanted to add that the other printer which has no problem with ATX has an ACTIVE LOW relay on the power sup[ply to turn it ON/OFF. But this one has an ACTIVE HIGH one. Can it be the issue ?

safaenet, 2019-12-13:
Just change relay config to ACTIVE LOW and made no difference!

thisiskeithb, 2019-12-13:
This is normal. TMCs need full 12/24 power initially on boot or you'll see TMC CONNECTION errors.

safaenet, 2019-12-13:
Isnt there a way to fix it ? to make them work along with ATX PSU ?

> This is normal. TMCs need full 12/24 power initially on boot or you'll see TMC CONNECTION errors.

thisiskeithb, 2019-12-13:
> Isnt there a way to fix it ? to make them work along with ATX PSU ?

I believe `PSU_POWERUP_DELAY` is the feature you're looking for. See PR #16050:

> Some power supplies require an extended time to come online. In my case, I have a +24V PSU connected via a solid state relay which takes about one second to provide power. In conjunction with TMC2130 drivers, the delay of 100ms is too short and initializing them fail.

houseofbugs, 2019-12-13:
This is not a bug but a config issue. The issue queue is for firmware bugs only, not technical support. This should be closed.

closed 2019-12-13 by boelle