MarlinFirmware/Marlin #16212: [BUG] G28 : TouchMi probe stow

Kebims, 2019-12-13:
### Bug Description

When I make a G28 command, after the home of my X axis the probe of my TouchMi goes down and the head moves to the center of the board to make the home of the Z axis with the probe.

After the home of the Z axis, the function STOW_PROBE() is called while the probe must be lowered in order to execute the G29 command.

### My Configurations

Marlin 2.0
Bigtreetech SKR V1.3
TouchMi

#### Additional Information

I think this is located in the _motion.cpp_ file at the comment line `// Put away the Z probe `

boelle, 2019-12-13:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

ellensp, 2019-12-13:
use M401 deploy probe before G29?

Kebims, 2019-12-13:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3960714/Marlin.zip)

Kebims, 2019-12-13:
> 
> 
> use M401 deploy probe before G29?

I think it is possible to use this command but I prefer not to retract my probe and launch the G29 command directly.

boelle, 2019-12-13:
is it unfair to say the you just expect G28 to work different than it is?

Kebims, 2019-12-13:
In previous version of Marlin ( 1.1.9 ), only G28 and G29 commands are necessary for using TouchMi probe.
It is logic to not raise the probe at the end of the G28 command and to raise the Z axis at the beginning of the G29 command.

Kebims, 2019-12-13:
> 
> 
> use M401 deploy probe before G29?

M401 command doesn't work for me.

At end of G28 command, STOW_PROBE() working fine ( like M402 command ) but at beggining of G29 command, DEPLOY_PROBE() doen't work ( like M401 command ).

In _run_deploy_moves_script()_ , such as the TOUCH_MI_DEPLOY_XPOS is not defined in _Configuration.h_ so nothing is done.

shitcreek, 2019-12-14:
For whatever reason, it looks like this bit of code was removed:
```
#ifndef TOUCH_MI_DEPLOY_XPOS
     #define TOUCH_MI_DEPLOY_XPOS X_MIN_POS
```
in `planner.cpp`

https://github.com/MarlinFirmware/Marlin/pull/14736

@thinkyhead any reason why it was stripped?

Kebims, 2019-12-14:
This modification solve my problem.

shitcreek, 2019-12-15:
I've created a PR to put it back in.

Kebims, 2019-12-16:
Ok. Thank you all.

closed 2019-12-16 by Kebims

yahyah206, 2020-02-11:
It was so simple on 1.1.9 to put the code
G1 Z0.5 in the start Gcode
I got around the problem of automatic retraction after a G28
indicating 15 on this line
#define TOUCH_MI_RETRACT_Z 15
Like this I let my start Gcode manage the retraction of the Pin from the TouchMi when I want.
Why make it simple, when you can make it complicated ^^