MarlinFirmware/Marlin #16215: [FR] (Screen Saver Function For OLED Displays)

borland1, 2019-12-14:
### Description
Suggestion: Add ability to enable a 'Screen Saver Function' and countdown time-out 'Duration' in Configuration.h file.
Reason: Screen Saver Function could be useful to minimize screen burn-in for OLED displays.

### Feature Workflow
Setup:
1. Enable Screen Saver Function and Timeout Setting in Configuration.h file.

Firmware Operation:
1. When a countdown timer times-out with no rotary encoder input being detected, the display mode status is set to SLEEP (blank display).
2. Any detection of rotation of the rotary encoder, resets countdown timer and changes display mode to AWAKE (active display).

ellensp, 2020-01-03:
Isn't this issue #4728

borland1, 2020-01-03:
Similar, but power saving on OLED displays would be negligible.  For FR #4728, LCDs do use significant more power than OLED displays.  This FR is to minimize OLED burn-in over hundreds of hours of use.

borland1, 2020-01-30:
Closing this. Reconsidering this FR as not necessary.

closed 2020-01-30 by borland1