MarlinFirmware/Marlin #16216: TFT35 onboard SD card printing problem

lijinlin19970813, 2019-12-14:
I have a TFT35 serial port screen, and print the gcode file through the onboard SD card, but the final position is incorrect after printing is completed, and the serial port continues to send M105 ca n’t receive the corresponding return information, I use the latest marlin, this kind of problem Occasionally, and every time the final position is incorrect.I think bigtreetech's screen firmware should be fine, because it just sends commands to display some information on the screen, it will not affect the final printing position.

lijinlin19970813, 2019-12-14:
[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3963685/marlin.zip)

lijinlin19970813, 2019-12-14:
Because the screen firmware sends M105 cyclically, it is sent cyclically after printing is completed, and then the system is busy, please wait. I used the serial tool to find that no corresponding response message of the M105 command appeared after printing was completed.

ellensp, 2019-12-14:
What do you have in your gcode.end file or equivalent? wondering if it has something wrong like set temp to 0 and wait...  which will wait till your room is 0 degrees.   A long of the serial might be useful Slice up a 10x10x0.4 cube (or what ever your single layer height is)  and log the serial for analysis.

lijinlin19970813, 2019-12-16:
[GCODE.zip](https://github.com/MarlinFirmware/Marlin/files/3965866/GCODE.zip)
This is my GCODE file. When the printing is completed, is the M105 sent is not parsed in marlin, because the screen firmware sends the M105 cyclically and then receives the correct information, the screen will not be busy, I want to know if I can modify it marlin to keep my screen from showing busy after printing, and can exit. I want to know how to modify marlin to solve this problem temporarily.

lijinlin19970813, 2019-12-16:
[image]
Is it possible to make changes here to deal with this problem temporarily.

lijinlin19970813, 2019-12-17:
> What do you have in your gcode.end file or equivalent? wondering if it has something wrong like set temp to 0 and wait... which will wait till your room is 0 degrees. A log of the serial might be useful Slice up a 10x10x0.4 cube (or what ever your single layer height is) and log the serial for analysis.

I have tested many times, and I found that the temperature of the NOZZLE and the  BED will not drop when the file is printed, and the M27 sent when the file is printed is not returned（ Not SD printing）

lijinlin19970813, 2019-12-18:
I tried to print with ch340, and sent M105 and M27 every three seconds during printing. When the printer stopped, the nozzle position was not correct because I added the G28 command to the gcode file, and the temperature of the bed and nozzle was kept. The RX does not receive the data returned by the corresponding command. I think the file has not been completed, but I do n’t know why. Or I get stuck sending M105 every three seconds, so why?

closed 2019-12-18 by lijinlin19970813