MarlinFirmware/Marlin #16218: Waves when printing

dark184, 2019-12-14:
### Bug Description
I apologize in advance for my English, it is not my native language.

Configured Marlin firmware 2.0.x for LPC1768 processor, compiled and flashed. Boards MKS sBase (DRV8825 with microstep 1/32), BigTreeTech SKR v1.3 (DRV8825 with microstep 1/32) and many others. On both boards, printing occurs in waves, as in the photo. On all photos on the left-the seal with firmware Marlin 2.0.x, on the right-Smoothieware. Cost to flash again Smoothieware, the printer once again prints perfectly.
To build the firmware, Platformio was used, with a choice in the file "Platformio.ini " LPC1768.
Both of these boards work correctly with Marlin 2.0.x firmware, with the exception of print defects.

### My Configurations

[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/3963726/MarlinConfig.zip)

### Steps to Reproduce

1. Configure the Marlin firmware 2.0.x for LPC1768 processor
2. Select in the file Configuration.h "MKS sBase" motherboard or customize for "BigTreeTech SKR V1.3" motherboard.
3. Compile firmware and flash the printer motherboard.
4. Run "Autotune" for Hotend pid controllers and heating bed. Enter the received values into the firmware, compile and flash the motherboard again.
5. Make a trial print.

**Expected behavior:** After the test printing was expecting a straight vertical wall on the model.
[image]

**Actual behavior:** The outer walls of the model are completely in horizontal waves. This defect is manifested exclusively with the firmware Marlin. On smoothieware firmware, the walls of the model are perfectly straight.
[image]

additional photo
[image]
[image]
[image]
[image]
[image]
[image]
[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2019-12-14:
Can you attach your smoothieware config, so we can also test this.

dark184, 2019-12-14:
One moment
My smoothieware config
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3963901/config.zip)
Only on my printer specific settings of kinematics, it should be borne in mind.
Both firmware are configured exactly for my kinematics.

ellensp, 2019-12-14:
was more thinking of running two virtual prints of the same gcode (minus heating instructions) one using marlin, the other smoothieware. No real stepper drivers or other hardware other than a lpc1768 based controller, so kinematics don't matter.  Just log the step and direction signals.  Sadly I don't own such a device, but someone here must who is keen to get to the bottom of this....

dark184, 2019-12-14:
I can not 100% say that the problem is in the firmware. But in my opinion the problem lies precisely in the instability (or phase) of the "step" signal.

thisiskeithb, 2019-12-14:
Can you attach your Smoothieware config for the SKR 1.3 setup as well?

sjasonsmith, 2019-12-14:
This is almost certainly the issue with inaccurate step pulse times. I don’t have the existing issue number or my open PR number handy.

If you find my PR that is open you could try that, or you can set MINIMUM_STEPPER_PULSE to 3, in Configuration_adv.h.

dark184, 2019-12-15:
**sjasonsmith** thank you very much, you hit exactly the target. Increased MINIMUM_STEPPER_PULSE at once to 4 that for certain, the printing more or less normalized. It remains to calibrate the temperature of the hotend. There were printing defects, including barely noticeable shifts of layers, but everything else is 100% not related to the firmware.
Can you tell me where I can follow the progress of fixing this bug? For me, this bug is very critical, since I only use DRV8825. To date, we have to use only Smoothieware. And this firmware has many disadvantages.

sjasonsmith, 2019-12-17:
This is a duplicate of issue #11047.  @dark184, you should continue to track the issue from there.

axl12ap, 2019-12-17:
I also have the same problem reported in the initial message, yesterday I printed the superman torso with layer height 0.1mm and stopped when I saw the horizontal bands.

My current configuration is as follows:
- CoreXY kinematics with SKR 1.3 board
- TMC2208 connected in UART, stealthchop mode (also extruders)
- Marlin 2.0.X ver. 06/28/2019
- Fysetc Mini12864 display

Attacched my configurations header files, anyone can help me please? Shall I try to change the MINIMUM_STEPPER_PULSE manually?? Or something else??

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3972343/Config.zip)

[image]
[image]
[image]

dark184, 2019-12-17:
In my opinion this is an unstable Hotend temperature. Calibrate the hotend pid controller.

axl12ap, 2019-12-17:
> In my opinion this is an unstable Hotend temperature. Calibrate the hotend pid controller.

In the next few hours (during my lunch) I'll try to calibrate the hotend and extruder with PID autotune, then I'll retry to print the base for checking 

BTW, any suggested change to the stepper defines (MINIMUM_STEPPER_DIR_DELAY, MINIMUM_STEPPER_PULSE, MAXIMUM_STEPPER_RATE) related to the TMC2208?

Thanks

ellensp, 2019-12-17:
@axl12ap your issue is not related...  your corners are not wavy!

sjasonsmith, 2019-12-20:
@boelle, could you close this as a duplicate of #11047?

PR #16128 has been merged into bugfix-2.0.x, which should fix the issue anyway.

boelle, 2019-12-20:
dublicate of  #11047 and fixed by pr #16128

boelle, 2019-12-20:
should #11047 also be closed?

closed 2019-12-20 by boelle