MarlinFirmware/Marlin #16222: [Z_SAFE_HOMING BUG] (Z_SAFE_HOMING_X_POINT not moving on G28)

Zatarines, 2019-12-14:
### Z_SAFE_HOMING_X_POINT not moving on G28, and causing G29 not to move properly

### My Configurations

**Required:**[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3964222/Marlin.zip)

### Steps to Reproduce

1. [G28]
2. [X axis not moving to bed center]
3. [G29 not moving to the probing points]

**Expected behavior:** [G28 center Nozzle to the center of bed ]

**Actual behavior:** [X axis not moving]

#### Additional Information

https://youtu.be/Z6ilpecv_W8 

https://youtu.be/1VUfplDMGT4

shitcreek, 2019-12-15:
It looks to me that it's not moving in the Y axis instead of X. In the 1st video, did you do a G28 then followed by a G29?

This probably has nothing to do with your issue, but I noticed that your probe is on the left side and yet you have the X `NOZZLE_TO_PROBE_OFFSET` set to a positive value.

tatusah, 2019-12-15:
Not a bug. You have a mistake in the configuration_adv.h file

```
#if PROBE_SELECTED && !IS_KINEMATIC
  #define MIN_PROBE_EDGE_LEFT 20
  #define MIN_PROBE_EDGE_RIGHT 190
  #define MIN_PROBE_EDGE_FRONT 190
  #define MIN_PROBE_EDGE_BACK 30
#endif
```

Those values are supposed to mean a margin from the specified edge and not from the home position. So with settings you have now the "safe" probing area starts 190 mm from the right side and 190 from the front of the bed. And because your left margin is 20 and bed width set 230, this would mean that the area you can probe is just 20 mm. More likely you would like to set the values to something around 10-30 mm. Or alternatively return them to the default state which is disabled by commenting them. Check the example Marlin configs.

Vertabreak, 2019-12-15:
properly defining things for your machine helps alot the example configs are just examples. these company's make changes to there printers every batch sometimes. so it is literally impossible to have it perfect for everyone.

tatusah, 2019-12-15:
Yeah, can't beat proper configuration, especially when dealing with this kind DIY monster as pictured in the video. @Zatarines close down this issue when ready.

Vertabreak, 2019-12-15:
the 3dprinting hobby is inherently a tinkering project. if even one thing is changed on the hardware the variables need to change in the firmware. sometimes we can do this with gcode or with the lcd but some things are just hard coded.

Zatarines, 2019-12-15:
@tatusah 
Thank you much for pointing out that my config. was improper, for I am coming for the Marlin 1.1 version and the MIN_PROBE_EDGE logic of things were somewhat like that. I made the following changes to my Monster's FW based on your suggestion:

#if PROBE_SELECTED && !IS_KINEMATIC
  #define MIN_PROBE_EDGE_LEFT MIN_PROBE_EDGE
  #define MIN_PROBE_EDGE_RIGHT MIN_PROBE_EDGE
  #define MIN_PROBE_EDGE_FRONT MIN_PROBE_EDGE
  #define MIN_PROBE_EDGE_BACK 30 (Because my BLtouch is behind nozzle)
#endif

Conf.h
#define MIN_PROBE_EDGE 20

#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif

It did help with the G29 trying to probe points in the bed, but I did not help solving the Main issue with G28 X axis not moving to the center of the bed. therefore, the probing points will never align because the X axis is not centered.

Zatarines, 2019-12-15:
@shitcreek 
My apologies for not being clear with the angle of the media I posted here. The angle that you see the printer is front the back right side. If you were looking at it from the front, the Bltouch is behind the nozzle.

closed 2019-12-15 by thinkyhead