MarlinFirmware/Marlin #16223: [BUG]  ( LCD SD-card 'Insert' Not Detected For MKS_SGEN_L  Board )

borland1, 2019-12-14:
### Bug Description

When inserting SD-card in SD-card slot of LCD display module, SD-card is not detected, so display's menu does not update with file list/directory.

### Steps to Reproduce

With Controller Board MKS_SGEN_L selected,
when enabling LCD's SD-card in Configuration_adv.h file,
line 1153...
#if HAS_SDCARD_CONNECTION
with..
#define SDCARD_CONNECTION LCD

Above configuration properly disables the ONBOARD SD-card and enables the LCD SD-card because both slots use same SD-DETECT_PIN ( P0.27 ).

Suggested Fix. 
In file:  pins_MKS_SGEN_L.h, line 269... 
#if SD_CONNECTION_IS(LCD),
add...
#define SD_DETECT_PIN    P0_27
 
#### Additional Information

Pin mapping of SD_DETECT_PIN is only defined for SD_CONNECTION_IS( ONBOARD ).

[ConfigurationFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3964419/ConfigurationFiles.zip)

thisiskeithb, 2019-12-14:
That's odd that you have to define it again, because `SD_DETECT_PIN` should be set to `P0_27` as long as you're not using a `CR10_STOCKDISPLAY` (which you aren't):

https://github.com/MarlinFirmware/Marlin/blob/6a514a1e8e96bcd26985615377523d1e45ca7b10/Marlin/src/pins/lpc1768/pins_MKS_SGEN_L.h#L192-L225

thisiskeithb, 2019-12-14:
I moved the `SD_DETECT_PIN` define under the `SD_CONNECTION_IS(LCD)` section in PR #16224.

cjsoong, 2019-12-17:
The issue was not resolved after the update.

thisiskeithb, 2019-12-17:
> The issue was not resolved after the update.

Does [this](https://github.com/thisiskeithb/Marlin/blob/172987534008c487237a3907f90bad52320d1243/Marlin/src/pins/lpc1768/pins_MKS_SGEN_L.h) pins file work? This is what I submitted originally.

borland1, 2019-12-17:
How is that different than what was pushed to bugfix-2.0.x ?
This is what is there now.... and looks correct ....

#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION ONBOARD
#endif

#define ONBOARD_SD_CS_PIN  P0_06   // Chip select for "System" SD card

#if SD_CONNECTION_IS(LCD) || SD_CONNECTION_IS(ONBOARD)
  #define SD_DETECT_PIN    P0_27
  #define SCK_PIN          P0_07
  #define MISO_PIN         P0_08
  #define MOSI_PIN         P0_09
  #if SD_CONNECTION_IS(ONBOARD)
    #define SS_PIN         ONBOARD_SD_CS_PIN
  #else
    #define SS_PIN         P0_28
  #endif
#elif SD_CONNECTION_IS(CUSTOM_CABLE)
  #error "No custom SD drive cable defined for this board."
#endif

thisiskeithb, 2019-12-17:
Do the merged changes work for you @borland1?

borland1, 2019-12-17:
Keith,   I'm not sure what you mean by 'merged changes'.  The pull request you sent to bugfix-2.0.x compile with 'Success'.   I have not verified those changes with SD-card and firmware.bin file on my MKS_SGEN_L board.  I did verify the display SD-card slot working with my own edited version, but that was before opening this 'new issue'.  Do you want me 'Close' this issue or verify again with firmware.bin file on the board with display attached?

thisiskeithb, 2019-12-17:
> Keith, I'm not sure what you mean by 'merged changes'.

I am referring to the the changes I submitted in the now merged PR #16224.

Please pull the latest `bugfix-2.0.x`, configure it for your printer, flash the updated firmware, & report back if your issue (LCD-based SD card not working correctly on SGEN-L) is resolved.

thisiskeithb, 2019-12-17:
@cjsoong: I didn't realize you weren't the author of this bug report, so unless you have the exact same LCD and motherboard/problem as @borland1, please open a new issue.

borland1, 2019-12-17:
Keith,
I just confirmed the LCD SD-card slot is working with the changes you submitted.

Confirmed the following: Display properly shows 'Media Inserted' and 'Media Removed' with card inserted and removed, while menu states Print from Media,  When inserting or removing card from Onboard SD slot, display properly shows 'Media Inserted' and 'Media Removed' and display menu always states: 'No media'.

Please close this issue report.  Thanks for your effort.

thisiskeithb, 2019-12-17:
> Please close this issue report. Thanks for your effort.

I don't have that power, but you do 🙂
[image]

closed 2019-12-17 by borland1

leytanelectronics2, 2020-05-12:
is there a solution for the ramps 1.4 board? that's the one I'm using for my setup and I am having the same issue, when I plug the sd in it shows that it has been inserted, but when I go to print from it I get "no media" text. if anyone knows how to fix this for the ramps please let me know.

ellensp, 2020-05-12:
@leytanelectronics2 that is an incorrectly formatted or just incompatible sd card. And not at all related to this error.

PR #16224 (2019-12-15): Update SD detect pin on MKS SGEN-L
### Description

This PR fixes #16223/allows LCD-based SD cards to be detected on an MKS SGEN-L. 

### Benefits

Allows LCD-based SD cards to be detected on an MKS SGEN-L.

### Related Issues

Fixes #16223