MarlinFirmware/Marlin #16225: Unable to compile SKR_Pro V1.1 with current Marlin 2.0.x

ebraiman, 2019-12-15:
Steps to Reproduce
Download and unzip Marlin bugfix -2.0.x
Open in PlatformIO
Change in configuration.h
#define SERIAL_PORT -1
#define MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1
Change in platforio.ini
\default_envs = BIGTREE_SKR_PRO
Save and compile.
Expected behavior: [What you expect to happen]
Builds without issue

Actual behavior: [What actually happens]
Builds and fails in multiple places.

shitcreek, 2019-12-15:
Please provide more information such as any warnings and errors, but before we go further. Please make certain that this is a bug and not a configuration error. On BTT's github, there is mention of compilation errors sometimes: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

ebraiman, 2019-12-15:
The software is downloaded from www.marlinfw.org
The IDE is VScode
the errors are as follows:

> Executing task in folder Marlin-2.0.x: C:\Users\ebraiman\.platformio\penv\Scripts\platformio.exe run <

Processing BIGTREE_SKR_PRO (platform: ststm32; board: BigTree_SKR_Pro; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 5.7.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32 3.10700.191028 (1.7.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.1
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.3.4
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\HAL.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\HAL_SPI.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\Servo.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\SoftwareSerial.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\fastio.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\persistent_store_impl.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\persistent_store_sdcard.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\timers.cpp.o
In file included from Marlin\src\HAL\HAL_STM32\HAL.cpp:26:0:
Marlin\src\HAL\HAL_STM32\HAL.h:47:4: error: #error "Serial port 0 does not exist"
   #error "Serial port 0 does not exist"
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32\Servo.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/HAL.h:47:4: error: #error "Serial port 0 does not exist"
   #error "Serial port 0 does not exist"
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32\timers.h:25,
                 from Marlin\src\HAL\HAL_STM32\SoftwareSerial.cpp:40:
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/HAL.h:47:4: error: #error "Serial port 0 does not exist"
   #error "Serial port 0 does not exist"
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32\persistent_store_impl.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/HAL.h:47:4: error: #error "Serial port 0 does not exist"
   #error "Serial port 0 does not exist"
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/../HAL/HAL.h:26:0,
[... 57 lines cut ...]
STM32F4                       IGNORED
STM32F7                       IGNORED
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_lite                IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               FAILED    00:00:09.1000
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
include_tree                  IGNORED
======================================================== 1 failed, 0 succeeded in 00:00:09.1000 ========================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

ebraiman, 2019-12-15:
[Marlin-2.0.x (1).zip](https://github.com/MarlinFirmware/Marlin/files/3964516/Marlin-2.0.x.1.zip)
This is from today's download on Marlinfw.org

ebraiman, 2019-12-15:
It has the modifications from the above description.

ebraiman, 2019-12-15:
I might have found the issue, it needs to be changed for
#define SERIAL_PORT -1

closed 2019-12-15 by ebraiman