MarlinFirmware/Marlin #16232: Ender 3 SKR E3 Bug 2.0.X Marlin

jschliebe, 2019-12-15:
Hi all,

I seem to experience a bug I am not quite sure how to describe. The issue is as follows:

After slicing Gcode with S3D and sending to print on my octoprint interface serveral things happen. First I get a warning that the item to be printed is too big for my build platform. If I attempt to print the item I notice the stepper mottors moving a lot faster then they should. Once the printer starts the print it is skewed, like everything is zoomed in and everything is larger than it should be. This happens on all 3 of my printers. This causes the X and Y axis to run into their hard limits skipping the motors. The only way I have found to circumvent this bug is by restoring fail safes via the menu on the printer. Without changing Gcode or any upload I reprint and everything is back to normal. Not sure where to go from there.

Thanks, 

Jamie S

shitcreek, 2019-12-16:
Could be you have something in your start Gcode in S3D. Did you configure your printer settings in S3D to match your printers' parameters?

You need to follow the template and provide more information such as your configuration files and machine make/model. But before we go further, please be certain this is a bug and not some configuration issue as we do not provide tech support here.

 For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, let us know and we'll gladly reopen this issue.

closed 2019-12-16 by shitcreek

jschliebe, 2019-12-16:
It definitely is not the Gcode, as I have used different slicers and get the same result. Also if you had read the post you would have seen that I can restore the fail safe of the printer with the same exact Gcode and it will print just fine. Will update to correct format.