MarlinFirmware/Marlin #16234: [BUG] (Continuation) Auto-Home causes glitchy X-Axis Homing + Halt

T-SquaredProductions, 2019-12-16:
### Bug Description

(Continuation of Issues #16132 and #16179)

After attempting to get my printer working by using the software filter, I still have problems getting the system to auto home correctly. Before, it did not even move. Now what happens is the system attempts to auto home, but with the wrong kind of microstepping. It does attempt to return after the endstop has been hit, but it instead jitters in place, and again throws up a "Homing Failed! PRINTER HALTED - Please Reset" message. Sometimes it even goes the wrong direction.

### My Configurations

[CONFIG3.zip](https://github.com/MarlinFirmware/Marlin/files/3965608/CONFIG3.zip)

### Steps to Reproduce

1. Uncomment "#define ENDSTOP_NOISE_THRESHOLD" and set to "2"
2. Go to Auto Home

**Expected behavior:** Smooth Auto Home sequence

**Actual behavior:** Behavior is detailed in the video contained within this ZIP file.

[20191215_175752small.mp4.zip](https://github.com/MarlinFirmware/Marlin/files/3965632/20191215_175752small.mp4.zip)

#### Additional Information

I forgot to add this to my previous 2 issues, but I am using a RAMPS 1.6 board.

T-SquaredProductions, 2019-12-16:
Ok, after doing another test, I'm beginning to think that something is wrong with the X-axis driver, because I'm not getting any homing problems with the rest of the axes.

T-SquaredProductions, 2019-12-16:
Nope. Now the X-Axis won't even move.

T-SquaredProductions, 2019-12-16:
Ok, rolling back to 1.1.9. That seemed to work. (Except now I have to figure out how to get my heated bed to be seen.)

@boelle, I seem to have found a genuine bug, as I cannot get the x-axis to auto home in v2.0.x.

Vertabreak, 2019-12-16:
have you checked the board pins for X maybe something changed?

tatusah, 2019-12-16:
How have you @T-SquaredProductions  come up with these settings? Are you sure about these settings you have done:

```
#define X_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
```

Those settings differ from the default Marlin 2.0 configuration for you board. You of course can have endstop switches that are wired like this but it's a thing you should confirm if not entirely sure. Compare those settings to what you have in your old 1.1.9 configs. At least having these settings set incorrectly would cause the behaviour seen in the video. Also you could move the carriage by hand away from the endstops. Then turn on the printer and check the endstop status with M119, if the x/y are triggered while the carriage is actually away from the enstops that would confirm that you have these settings incorrectly.

tatusah, 2019-12-16:
Oh, sorry, I just checked those other closed issues you had previously and there you had already played around with the endstop inversion. But still, if the hardware hasn't changed when you moved to the Marlin 2.0.x then closely check whether you have copied all the relevant settings that allowed the system to work with the old Marlin. And if you changed also some hardware then double check connections. After all the homing works as it should be with other people.

Vertabreak, 2019-12-16:
homing works on all 6 of my printers it has to be hardware failure or miss configured.

boelle, 2019-12-24:
will close this one as its most likely a hardware fail or config issue

closed 2019-12-24 by boelle