MarlinFirmware/Marlin #16237: Z axis loses zero occasionally

Routout, 2019-12-16:
I have OctoPi, Marlin 2.0, BLTouch and SKR1.3 with TMC2209s.
I have problem that when I do auto home, printer goes to x,y endstops , then center correct but not down. BLTouch do normal 2 tries, doesn't find bed and error.
I compiled new Marlin today because I thought is a bug, because this setup worked few weeks before this same problem.
Today I made 2 quick prints just fine, then raised z from the z-rod manually to clean the nozzle and again the problem I first described.
Any ideas wheres the problem?

GMagician, 2019-12-16:
First important thing when posting a bug:

> **Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

reversed Z endstop level? Please use M119 and check z_probe status

Routout, 2019-12-16:
> First important thing when posting a bug:
> 
> > **Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
> 
> reversed Z endstop level? Please use M119 and check z_probe status

Thanks, M119 revealed BLTouch broken wire.

closed 2019-12-16 by Routout