MarlinFirmware/Marlin #16239: [BUG] (short description)

vMeph, 2019-12-16:
### Bug Description

marlin EEPROM version mismatch (EEPROM=? Marlin=V74)

#### Additional Information

using latest bug fix 2.x and  BOARD_MKS_GEN_L
i tries use M502 M500 M501 but nothing still same problem

ellensp, 2019-12-17:
Where are your config files as requested by the template?  Without that how do we know what eeprom emulation your using, part of flash or sd card or something else?

boelle, 2019-12-17:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

vMeph, 2019-12-17:
@boelle  i just was able to male it work using the arduino serial console with the M502 and M500

closed 2019-12-17 by vMeph

Patag, 2019-12-17:
Bo the robot ;-) @boelle

shitcreek, 2019-12-17:
Please, everyone, be courteous to all contributors and maintainers, and keep this space free of personal drama. We've got a Discord channel for that.