MarlinFirmware/Marlin #16240: Bilinear bed leveling badly broken on 2.0.0

token47, 2019-12-17:
### Bug Description

Upgraded from 1.1.9 to 2.2.0 and everything seems to be working except for bilinear bed leveling which seems completely broken.

When I issue a G29, nozzle goes to the corner as usual and then make all 9 readings (3x3 grid) there, in the same place, without moving to the next spot between each reading. Sometimes (very sporadically) it will move once between some readings.

### My Configurations

I put my configs in my google drive at https://drive.google.com/file/d/1vFU2gaLGN5rtOFSVrCzUtaWaJucdCww3/view?usp=sharing

but essentially all that I have enabled (regarding bed leveling) is:

#define AUTO_BED_LEVELING_BILINEAR
#define ENABLE_LEVELING_FADE_HEIGHT
#define GRID_MAX_POINTS_X 3
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
#define EXTRAPOLATE_BEYOND_GRID

Already tried to disable both FADE_HEIGHT and EXTRAPOLATE_BEYOND_GRID and they do not seem to affect the problem (again, I was using them in 1.1.9).

### Steps to Reproduce

1. Home (G28)
2. Level bed (G29)

**Expected behavior:** [What you expect to happen]

I expect nozzle to go around the bed taking measures in a 3x3 grid.

**Actual behavior:** [What actually happens]

It stays in the corner taking all measures in the same place.

#### Additional Information

It is like if X/Y movements were crippled during the procedure. What makes me think of that is that even if I set the Z_PROBE_END_SCRIPT to move in X/Y/Z space it will only move in Z and ignore X/Y. I even tried a G27 P2 (which would move the head 20 mm up and to the corner) and it still only moves up.

Just after G29 is complete, normal movement is restored again, and I can move Z/Y/Z fine, even a G27 will correctly park in the desired location.

Current for motors are ok (all other movements seem fine). Drivers configs (TMC2208) seem ok too. I can't see any other configs that would influence. The configs were basically ported from 1.1.9 where they were working fine.

I switched to a 2x2 matrix and it tries to start on the right/front corner, as expected, but the problem persists, it wil do all probes there. Switching back to a 3x3 matrix makes it start at the left/front corner again.

3POINT and LINEAR have the same results, did not test UBL. I went back to 1.1.9 and it works fine, so it is something with version 2.0.0 or my configs. I find it very strange because it seems so broken that it would be very strange for nobody else to have noticed it yet, so it probably is my config, but I did not set anything that I didn't already have in 1.1.9, so maybe it is some special combination of factors.

Of special note: Z is using A4988 driver while X+Y are using TMC2208 driver.

token47, 2019-12-17:
Hummm, I also had a different problem ( bug #14183 ) that made me disable MONITOR_DRIVER_STATUS to stop the knocking from drivers.

I was not making a big deal of it because I do not want to use hybrid mode with the TMC drivers, and I also do not mind losing the extra details on M122, so I just let if off. In my configs, I set X+Y to be STEALTHCHOPP only and E to be SPREADCYCLE only. It's fine because my printer do not lose steps with stealthchopv2.

But now I remembered that marlin forces spreadcycle for homing and I'm wondering if these two problems have any relation.

token47, 2019-12-17:
I tried forcing spreadcycle in X+Y but it didn't fix.

But I still think the problem has some relation with the drivers, will change them back to A4988 as soon as possible to test.

token47, 2019-12-17:
Another note, I have the TMC2208 drivers configured with UART on the default pins for MKS GEN L, when MONITOR_DRIVER_STATUS is enabled, M122 correctly shows all info about the drivers and connection tests pass.

token47, 2019-12-17:
Ok, confirmed. Just changing driver type from TMC2208 to TMC2208_STANDALONE is enough to make the problem go away, did not even had to switch them back for the allegro ones.

This is not optimal because now I have to rely on microstep setting on jumpers (which I had removed), current on the potentiometer and will loose all other goodness of the driver. It will also operate on stealthchop on my extruder, where I prefer spreadcycle.

Marlin is doing something odd when entering bed leveling routine that makes the driver not move the motors (sometimes they move very little, like a few mm between random probes).

Vertabreak, 2019-12-17:
are you sure UART is wired up right from the drivers to the board.

token47, 2019-12-17:
Yes, they are all wired with TX and RX, M122 shows "connection ok" on all three drivers, microstepping is set correctly even without jumpers and I can hear the difference of stealthchop and spreadcycle when I change them. Don't think communication is the problem.

sjasonsmith, 2019-12-17:
What branch are you using? Can you specify the exact commit you are using?

token47, 2019-12-17:
I'm using 2.2.0-release but also tested with latest bugfix as of today and the result was the same. I had even reset the EEPROM just to be safe, and it had not helped.

But.... since I could successfully probe the bed once with G29 after changing drivers to standalone model, I changed back to UART mode and cannot reproduce it again (G29 keeps working).

But the drivers are sizzling very loud, even when stopped/enabled, only stop when disabled. I'll do some more tests, but the G29 problem seems to be gone. Very odd.

token47, 2019-12-17:
Yep, it is gone. Very weird. Thanks for the interest. Closing now.

closed 2019-12-17 by token47