MarlinFirmware/Marlin #16244: [BUG] BL Touch on SKR PRO 1.1 without Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN doesn't work on G28

DarkShortie, 2019-12-17:
### Bug Description

BL Touch does not deploy on G28 if Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is disabled on SKR Pro 1.1.
It works without any issues on Configuration -> BLTouch and M43 Probe test.
It also works when enabling Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and set Z_MIN_PIN to PA2 which is the same pin like Z_MIN_PROBE_PIN.

`Send: M43 S
Recv: Servo probe test
Recv: . using index:  0, deploy angle: 10, stow angle:   90
Recv: . Probe Z_MIN_PROBE_PIN: 37
Recv: . Z_MIN_PROBE_ENDSTOP_INVERTING: false
Recv: . Check for BLTOUCH
Recv: = BLTouch Classic 1.2, 1.3, Smart 1.0, 2.0, 2.2, 3.0, 3.1 detected.
Recv: ** Please trigger probe within 30 sec **
Recv: . Pulse width (+/- 4ms): 12
Recv: = BLTouch pre V3.1 (or compatible) detected.`

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3974587/config.zip)

### Steps to Reproduce

1. Connect BL Touch to the Probe header on SKR Pro 1.1
2. Ensure Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is disabled in Configuration.h
3. Issue G28

**Expected behavior:** [What you expect to happen]
BL Touch deploys and works

**Actual behavior:** [What actually happens]
BL Touch doesn't deploy.

shitcreek, 2019-12-18:
This is the same issue with SKR 1.3. Even though `Z_MIN_PROBE_PIN` is defined in the pin files, but if you look at these boards, there is no separate `Z_MIN_PROBE_PIN` - so you are using / have to use `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`.
If you are using **VSCode** with **intelliSense**, when you disable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`, a warning is generated for `SET_INPUT_PULLUP` on line 248 in `endstops.cpp` stating that:
**Identifier "DIOP1_24_DDR" is undefined**

Although this warning doesn't show up when compiling, it seems to be important enough to cause the probe not to operate as it should.

I will investigate further.

DaVincIan, 2019-12-19:
I'm trying to get my BLTouch to work on the SKR Pro V1.1 as well.

Some pointers I discovered. 
The polarity of the power to the heated bed is reversed.
The filament sensors ports E0, E1, and E2 are actually Xmax, Ymax, and Zmax, used for this function.

**Back to BLTouch**
I am able to test and confirm the correct trigger of the Z_MIN_PROBE_ENDSTOP, by using GCode. 
The symptoms present as if the Deploy and Stow commands are revered when doing Z home.
Also with the probe Stowed, the probe Deploys when I send the command M119. 
My knowledge of Marlin FW code is minimal. As far as I could determine, the Z_MIN_PROBE (connected to pin PA2) sensing part is working correct. to me it seems as if the probe stow/deploy sequence is problematic.

I have mine connected to the BLTouch "port" with:

Config.h
//============================== Endstop Settings 
//================================================================
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
**All endstops as I use them, execpt ZMin. If I disable ZMin, Marlin does not compile**

// Enable pullup for all endstops to prevent a floating state
//#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX  //I need this disabled for my filament sensor
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
  #define ENDSTOPPULLUP_ZMIN_PROBE
#endif

**Below, with Z_MIN_ENDSTOP_INVERTING false, sanity check fails**
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.

**Z Min pin <> Z Min probe pin and Z_MIN_PROBE_PIN is correctly defined in relevant PINS.h**
/**
 * Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *
 * Enable this option for a probe connected to the Z Min endstop pin.
 */
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

There are a few Marlin issues very similar to this topic. From issue "bltouch z not going down rc8
#6371", Bob-the-Kuhn posted this test sequence:

_An alternative is to use the following sequence. The blue LED is present only on BLTouch units that have a serial number.

Connect BLTouch, power up controller and upload the firmware
Blue LED will be ON if a servo signal is present.
M119 ; should return OPEN. If not check that the BLTouch is enabled in the configuration. If enabled then check wiring and check that the correct pin is being used.
M280 P0 S10 ; should deploy the probe and the orange LED will be OFF.
M119 ; should return OPEN
M280 P0 S90 ; should stow the probe and the orange LED will be ON
M280 P0 S60 ; puts it into the M119 test mode.
The probe should remain stowed and the blue LED should be OFF. If it's ON then the unit needs to be adjusted.
M119 ; should return TRIGGERED
M280 P0 S160 ; returns it to normal operation
M119 ; should return OPEN._

Issue "BLTouch not working, bug? #8970" is another source with a lengthy chat and a lengthy work around. I have not tried it yet.

Bob-the-Kuhn seems to be a very knowledgeable contributor on the subject BLTouch.

DaVincIan, 2019-12-19:
I would recommend looking at issue #5705 as well, scroll down to Bob-the-Kuhn commented on Jan 16, 2017.

DaVincIan, 2019-12-19:
I believe I found a simple solution (work around/hack)

Configure the setup to use Zmin and not the Zmin probe.

Hardware change:
Plug the BLTouch sensing wire into the Zmin plug. White in Z- (PG8), Black in GND. As per SKR-PRO-V1.1-Pin.pdf

Marlin configuration (standard BLTouch config plus a change in SKR_PRO_V1_1 pins)

Marlin\src\pins\stm32\pins_BTT_SKR_PRO_V1_1.h
// Z Probe must be this pins
#ifndef Z_MIN_PROBE_PIN
**///  #define Z_MIN_PROBE_PIN  PA2**
#endif

**Configuration.h**
// Endstop Settings 
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
**#define USE_ZMIN_PLUG**
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
**#define Z_MIN_ENDSTOP_INVERTING false** // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
**#define Z_MIN_PROBE_ENDSTOP_INVERTING false** // Set to true to invert the logic of the probe.

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

//#define Z_MIN_PROBE_PIN PA2 // Defined in pins

#define BLTOUCH

**Configuration_adv.h**
 #define BLTOUCH_DELAY 500

//Take note of the 5V warning preceding this line in the Marlin FW
#define BLTOUCH_SET_5V_MODE  

And that is all.

shitcreek, 2019-12-19:
I think the recommendation is outdated. There is no need to undefine ** Z_MIN_PROBE_PIN** and with BLTouch, the logic of the endstop is always automatically set to **false**:

in **Conditionals_LCD.h**
```
// Always disable probe pin inverting for BLTouch
  #undef Z_MIN_PROBE_ENDSTOP_INVERTING
  #define Z_MIN_PROBE_ENDSTOP_INVERTING false
  #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
    #undef Z_MIN_ENDSTOP_INVERTING
    #define Z_MIN_ENDSTOP_INVERTING false
  #endif
```

The only things needed are:
**BLTOUCH
Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
USE_ZMIN_PLUG**

carl1961, 2019-12-20:
BLTouch working fine for me.  G28, G29 , Z auto align

Using BLTouch Header

configuration.h 

line: 659   #define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.
line: 829     #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
line: 879     #define BLTOUCH

pins_BTT_SKR_PRO_V1_1.h

line: 49  #define Z_MIN_PIN          PA2  // PG8 BLTouch Header Pin

shitcreek, 2019-12-20:
> line: 659 #define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.

this part is not necessarily since it is always reset to false in **Conditionals_LCD.h** for the BLTouch - see my comment above.

carl1961, 2019-12-20:
Ok thanks

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

DarkShortie, 2019-12-25:
I just tried it with the latest code, it's commit 1c9ccce5209cd1727bf80e632f4f781c651e0c35 and the result slightly changed.

At the first test when opening the bug report it didn't deploy but Z started to move down, now it doesn't deploy, but Z doesn't move down and leveling is finished. It looks like the endstop is directly hit.

M43 S results are still the same.
One thing I've just recognized is that M43 S reports Probe Z_MIN_PROBE_PIN: 37 but it should be PA2.

DaVincIan, 2019-12-26:
With Marlin-2.0.x

I use the default Marlin settings, but with the BLTouch limit sensor connected to Z Min and it works fine. No changes to pins file.

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.

shitcreek, 2019-12-26:
This is not a bug. These boards (SKR PRO 1.1 and SKR 1.3) don't physically have a `Z_MIN_PROBE_PIN`. User must use `Z_MIN_PIN` and thus must enable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`. `Z_MIN_PROBE_ENDSTOP_INVERTING` does not need configuring as it is always automatically set to `false` for **BLTouch**.

The confusion arises with the declaration of `Z_MIN_PROBE_PIN` pin in the pin files. 
```
//
// Z Probe (when not Z_MIN_PIN)
//
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN  P1_24
#endif
```
Note the comment. For whatever reason @Msq001 has changed it to **// Z Probe must be this pins** for the **skr pro 1.1** pin file which doesn't make it as clear as the original comment.

closed 2019-12-26 by shitcreek

sjasonsmith, 2019-12-26:
> These boards (SKR PRO 1.1 and SKR 1.3) don't physically have a `Z_MIN_PROBE_PIN`.

Actually, the SKR Pro does have a probe pin on a BLTouch header, although I don't think Marlin lets you use it if you are also using the BLTouch for Zmin homing, it expects that Z_MIN_PIN will be used in that case, even when using a probe.

[image]

That said, I think the easiest solution for everybody is to just plug the BLTouch into the Zmin header anyway. The only difference is that the probe pin would allow using a single 4-pin connector, but BLTouch doesn't come wired that way anyway.

carl1961, 2019-12-27:
Zmin on dedicated header works fine, you just change PG8 to PA2 in
pins_BTT_SKR_PRO_V1_1.h file,

#define Z_MIN_PIN          PA2  // PG8 BLTouch Header Pin             You
can use  PG8  for anything else ( Like Kill Pin)

In configuration.h you need to uncomment  ( around line 829  ) #define
Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

DarkShortie, 2019-12-27:
Why is this closed now? The issue is still there.

SKR Pro 1.1 has a Probe pin, which is also defined correctly in the pins file (PA2), so I expect that it works when I configure Marlin to use the Z_MIN_PROBE_PIN in the Configuration.h.

Yes, it works if you change Z_MIN_PIN to PA2 and activate Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN but this does render the whole option for Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN useless as you can always argue like this and Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN can be completely removed from the code base.

AnHardt, 2019-12-28:
Originally Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN had additional information. Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN implicit says "There is no other z-min-endstop - we have only a probe - so if you want to home to z-min you have to deploy the probe".
While an active Z_MIN_PROBE_PIN told: " The probe is not connected to z-min-endstop but here. So we have the z-min-endstop pin occupied by a read endstop - we don't need to deploy the probe when we home to z-min,"

LinoBarreca, 2020-01-07:
> Originally Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN had additional information. Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN implicit says "There is no other z-min-endstop - we have only a probe - so if you want to [...]

[image]
not anymore, it seems