MarlinFirmware/Marlin #16245: [BUG] Sensorless homing fails for Z-axis using COREXZ

colemarkham, 2019-12-17:
### Bug Description

I am testing a setup that uses COREXZ and sensorless homing using TMC2130 drivers. I switched the motor connections and changed the config to use COREXY and it works as expected for the X and Y axes.

### My Configurations

[Marlin-configs.zip](https://github.com/MarlinFirmware/Marlin/files/3969725/Marlin-configs.zip)

### Steps to Reproduce

1. Home X Axis
2. G28 Z

**Expected behavior:** Z-axis moves to the end and stops.

**Actual behavior:** Z-axis moves very slowly about 5-10mm at most and stops. X works as expected.

#### Additional Information

* Switching to COREXY with the exact same mechanical setup works as expected

shitcreek, 2019-12-17:
Looking at the **G28.cpp** I suspect the issue is on line 289:
 ```
 #if BOTH(STEALTHCHOP_XY, HOME_USING_SPREADCYCLE)
    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Set XY to spreadCycle...");
    process_subcommands_now_P(PSTR("M569S0XY"));
  #endif
```
It looks like it is not switching Z axis to spreadCycle mode.

You could try enabling `HYBRID_THRESHOLD` **or** if you are willing to experiment, try changing the aforement code snippet to something like this:
 ```
 #if BOTH(STEALTHCHOP_XY, HOME_USING_SPREADCYCLE)
    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Set XY to spreadCycle...");
    process_subcommands_now_P(PSTR("M569S0XY"));
  #elif ALL(STEALTHCHOP_XY, STEALTHCHOP_Z, HOME_USING_SPREADCYCLE)
    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Set XZ to spreadCycle...");
    process_subcommands_now_P(PSTR("M569S0XZ")); 
  #endif
```
Sensorless homing of Z is not recommended in general - maybe home Z with a probe instead?

colemarkham, 2019-12-17:
Thanks for the quick reply. I will try those things out and get back to you.

This is not actually a 3D printer, just a cartesian robot. I could switch to using XY instead of XZ if necessary. We're not actually using the Y axis at the moment. I'd like to get this working either way though.

shitcreek, 2019-12-19:
Recent commit makes **HOME_USING_SPREADCYCLE** obsolete. Would you mind testing if this affects your setup in anyway. Thanks.
https://github.com/MarlinFirmware/Marlin/commit/b49a0e39d5617ba3752b4b406cc663a7c6a88908

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle