MarlinFirmware/Marlin #16246: [BUG] SKR 1.3 SD DOESNT START TO HEAT

Erokawa, 2019-12-17:
### Bug Description
With SKR 1.3 i have been using a 2.0 version and everything were going perfectly, the problem is that when i updated the final version of marlin the sd was connecting correctly but when i try to print it doesnt start to heat nothing

### My Configurations
https://drive.google.com/open?id=1U1PNlBJI6Re7Jsqg1XdGLNiTkBMjtan2
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. try to print something with this config in a skr 1.3
2. if you start a print won't heat nothing will just run the time and it appears the name scrolling in the LCD

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-12-19:
Could you post your gcode file?
Next time, please only include the config files and not the entire Marlin build, thank you.

Erokawa, 2019-12-20:
Simply3D for some reason was generating gcodes of 0KB xD, okey thanks anyway i'll close this

closed 2019-12-20 by Erokawa