MarlinFirmware/Marlin #16248: [BUG] x-y motor clicking during diagonal movement with bed leveling

emaayan, 2019-12-17:
### Bug Description

when both motors operate at the same time a faint clicking sound is heard from both motors. 

### My Configurations

[16-12-2019.ZIP](https://github.com/MarlinFirmware/Marlin/files/3975905/16-12-2019.ZIP)
[TopCover.txt](https://github.com/MarlinFirmware/Marlin/files/3975912/TopCover.txt)

### Steps to Reproduce

1. run the attached TopCover.gcode it will reset the eeprom for default and generate a debug mesh 

**Expected behavior:** 
printing with no clicking 

**Actual behavior:** 
faint random clicks 

#### Additional Information
with no bed leveling is enabled the clicks won't be heard 
i've replaced firmware, various mks gen 1.4 boards, drivers (2100,2208) , motor cables, motors, tried with or without SD cards. it still goes on .

Powerprobot, 2019-12-18:
Your Z-Axis have not enough speed and acceleration to compensate these stuttering. to test it, set the Z-Axis on 150 movement and 300 acceleration and you wil see that is no more stuttering.

emaayan, 2019-12-18:
> Your Z-Axis have not enough speed and acceleration to compensate these stuttering. to test it, set the Z-Axis on 150 movement and 300 acceleration and you will see that is no more stuttering.

not sure i follow, the clicks are coming from the x and y motors not z one. they also come when there's no z motor actually attached.

Powerprobot, 2019-12-18:
With Bed-Level On, marlin will try to compensate the Z-differences. But if the Z-Axis to slow, marlin will wait to finish the move on Z-Axis and stop or deaccelerate  XY-Axis during this process.

emaayan, 2019-12-18:
this is my current acceleration settings, so you're saying i should run them higher? 

#define DEFAULT_MAX_ACCELERATION      { 1500, 1000, 100, 10000 }
#define DEFAULT_ACCELERATION          1000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1500    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   1500    //

Powerprobot, 2019-12-18:
If your steps in XY are 80 and Z 400 then divide them = 5 
So divide 1000 / 5 = 200 . Set Z acceleration to 200 and what are the mm/s ? 
If your mm/s in XY = 300 then 300 / 5 = 60. Also 60mm/s for the Z-Axis.
Test it. But it is not for Prduction use. No one use higher rates than 80mm/s in the first ~3 layers.
Use Z-Fading by Bed leveling max 3mm.

emaayan, 2019-12-18:
my #define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,1600,812.56}
are you talking about max z acceleration or default acceleration ?

emaayan, 2019-12-18:
> If your steps in XY are 80 and Z 400 then divide them = 5
> So divide 1000 / 5 = 200 . Set Z acceleration to 200 and what are the mm/s ?
> If your mm/s in XY = 300 then 300 / 5 = 60. Also 60mm/s for the [...]

my #define DEFAULT_AXIS_STEPS_PER_UNIT {80,80,1600,812.56}
are you talking about max z acceleration or default acceleration ?

emaayan, 2019-12-18:
> my #define DEFAULT_AXIS_STEPS_PER_UNIT {80,80,1600,812.56}
> are you talking about max z acceleration or default acceleration ?

btw should i return to classic jerk? 
https://www.reddit.com/r/3Dprinting/comments/dx8bdb/here_is_why_you_should_disable_junction_deviation/

emaayan, 2019-12-18:
this is an update there seem to be a difference in sounds in 1200 feed rate 

G1 X63.740 Y35.648 Z0.5 F4800
G1 X63.852 Y35.760 **F1200** 
G1 X63.852 Y36.439    
G1 X63.061 Y35.648 
G1 X62.382 Y35.648 
..
https://www.youtube.com/watch?v=hQeY35qkp7w

and 600 feedrate
G1 X63.740 Y35.648 Z0.5 F4800
G1 X63.852 Y35.760 **F600** (feed rate changed )
G1 X63.852 Y36.439    
G1 X63.061 Y35.648 
G1 X62.382 Y35.648 
...

https://www.youtube.com/watch?v=27qj_1KE3KQ

boelle, 2019-12-24:
@emaayan is the issue still the same with all the updates in the last 6 days?

emaayan, 2019-12-24:
yes, although i think if i decrease the feedrate from 1200 to 600 it might
be weaker.. i'm  trying skr mini and skr 1.3 to see if this is an 8bit vs
32 bit issues  (or maybe i'ts just a bug in how marlin handles mks gen 1.4
, or maybe a bug how it handles mega cpu in general i dunno)

boelle, 2019-12-24:
1200 for feedrate seems crazy high try with { 200, 200, 12, 120 }

Vertabreak, 2019-12-24:
for sure that is way to high i run (250,250,5,30)

emaayan, 2019-12-24:
if i recall correctly this was the original value from the g-code file i've
striped down that was for 9 rectangle for bed leveling model, it was
created s3d, andi think my speed back then was 80mm/s (i don't recall what
percentage for other speed types in it)
what values are those?

boelle, 2019-12-24:
those are values we use

yours is just to high no matter how you got them

emaayan, 2019-12-24:
wait are you talking about marlin firmware values or the actual feedrate,
cause i'm was talking about the actuall feed rate i meant to say that in
here intead of using 1200:

G1 X63.740 Y35.648 Z0.5 F4800
G92 E0.0000
G1 X63.852 Y35.760 *F1200*
G1 X63.852 Y36.439
G1 X63.061 Y35.648

i use 600

G1 X63.740 Y35.648 Z0.5 F4800
G92 E0.0000
G1 X63.852 Y35.760 *F600*
G1 X63.852 Y36.439
G1 X63.061 Y35.648

boelle, 2019-12-24:
> wait are you talking about marlin

what ever is highest

emaayan, 2019-12-24:
if you look at the configuration.h in my attachment you'll see i'm using
#define DEFAULT_MAX_FEEDRATE          { 200, 150, 10, 50 }

boelle, 2019-12-24:
and your slicer? some slicers override firmware

boelle, 2019-12-24:
and eeprom values?

emaayan, 2019-12-24:
oh i got burned with eeprom values before, the script i'm using always
resets using m502

the feedrate on the g1 codes are expressed in mm/min while the ones in the
firmware are expressed in mm/s

so 200 mm/s should be 12000 mm/min so i'm well below the limit

;READ:       I0      I1
;READ:  J0 +0.925 +0.191
;READ:  J1 +0.652 +0.565
;READ:

M502
G28 ; home all axes

G29 C1

boelle, 2020-01-08:
@emaayan  still no good news with the latest bugfix 2.0.x?

emaayan, 2020-01-08:
it's like a never ending swamp of goo, i'm attempting to see if i can use
an SKR 1.4.. but the only usable drivers i have right now are TMC2100, but
there doesn't seem to be any configuration options on how to use them

emaayan, 2020-01-11:
> @emaayan still no good news with the latest bugfix 2.0.x?

well after many many many hurtles, i've managed to get marlin 2.0.1 working with skr mini e3 version 
1.2 
i still get clicks.. this is a recording of the x-motor for better hearing 
https://www.youtube.com/watch?v=5dpWGD_DE2s

boelle, 2020-01-18:
completely noiseless, i would love to have it so good

emaayan, 2020-01-18:
then check this out, i think i've managed to get rid of them, this is SKR
1.4 with TMC2208 16 steps and 256 interpolation

https://www.youtube.com/watch?v=AXEL_f_rkj4
but i discovered another problem un-related, i can't use baby-stepping, if
try to turn the knob 2 twice, the z-axis just responding, and from then on
, the z-axis doesnt' move at all. (even after the print)

boelle, 2020-01-23:
no difference to me

please create a new issue if there is an unrelated issue, just so we dont mix oranges and potatos :-)

emaayan, 2020-01-23:
> no difference to me
> 
> please create a new issue if there is an unrelated issue, just so we dont mix oranges and potatos :-)

already did, issue 16617 was confirmed by scott. but at least i have  workground for that one.

and i was so hoping it would solve my original bed leveling issue because i thought those were mis-steps

boelle, 2020-01-30:
a lot of updates and fixing has happend in the last week, is the problem still there?

boelle, 2020-02-08:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-02-08 by boelle