MarlinFirmware/Marlin #16250: Printing communication problems

lijinlin19970813, 2019-12-18:
I need help. I use CH340 to send commands to print GCODE files. During the printing process, I send M105 and M27 every three seconds to get the temperature and printing status. I set the G28 command at the end of the gcode, but when the printing stops running, G28 was not executed and the temperature of the bed and nozzle did not drop, and my RX did not receive the corresponding command to return the data. It seems that the file was not finished. When I sent the M105 or M27 in a separate cycle, sometimes I could receive the print completion, sometimes the card Live, so why?
[image]

lijinlin19970813, 2019-12-18:
[my_gcode.zip](https://github.com/MarlinFirmware/Marlin/files/3977472/my_gcode.zip)
This is my gcode file,I added G28 X0 Y0 at the end, but in the end the motor stopped and G28 was not executed, and I found that the temperature did not drop and all the motors were disabled.I want to know if there is something wrong with my method.

boelle, 2019-12-18:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

lijinlin19970813, 2019-12-18:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3977519/marlin.zip)
Here, you can use this and try using my GCODE printing. Every time I get stuck at the end of the file, this situation only occurs when I send M105 and M27 through the serial port.

lijinlin19970813, 2019-12-18:
Forgot to add, the problem is caused by onboard SD printing,using marlin 2.0.

ellensp, 2019-12-19:
in your gcode G28 X0 Y0 ;

but this is not valid should be G28 X Y

tpruvot, 2019-12-19:
or G1 X0 Y0

lijinlin19970813, 2019-12-19:
> in your gcode G28 X0 Y0 ;
> 
> but this is not valid should be G28 X Y
> 
> Edit: but on trying it marlin doesn't seem to care if the 0's are there or not.

I don't think this is the problem. After I use the CH340 to let the printer start printing, I stop sending M105 and M27, and the result is that I can go home every time. But it doesn't work when I send it in a loop, it seems to be stuck, I still don't know where the problem is.

shitcreek, 2019-12-19:
I think it's because it interferes with the planner block. Instead of sending **M27** and **M105** in a separate cycle, you should include them in your start gcode like so:
**M27 S3**   ; report SD print status every 3 seconds
**M155 S3** ; report temp every 3 seconds - use this instead of **M105**

http://marlinfw.org/docs/gcode/M027.html
http://marlinfw.org/docs/gcode/M105.html
http://marlinfw.org/docs/gcode/M155.html

lijinlin19970813, 2019-12-24:
> ~我认为这是因为它干扰了计划程序块。~与其在单独的周期中发送**M27**和**M105**，不如将它们包括在您的起始gcode中，如下所示：
> **M27 S3** ; 每3秒钟报告一次SD打印状态
> **M155 S3**；每3秒报告一次温度-使用此温度代替**M105**
> 
> [http://marlinfw.org/docs/gcode/M027.html ](http://marlinfw.org/docs/gcode/M027.html)
> [http://marlinfw.org/docs/gcode/M105.html ](http://marlinfw.org/docs/gcode/M105.html)
> http://marlinfw.org/docs/gcode/M155.html

I checked the firmware of the TFT35 screen, and it was printed by sending M27 S3 once and continuously sending M105 before printing, but similar problems also occurred.

lijinlin19970813, 2019-12-24:
I found that this is not a problem with GCODE, this is a communication problem with marlin firmware. No matter what GCODE, as long as I continuously send M105 and M27 during the printing process, it will be stuck, because I will not return information when I send M119 or M122. But the motherboard still maintains the printing temperature, and when I restart the motherboard, the communication is normal again.

boelle, 2020-01-07:
@lijinlin19970813 is this still a problem?

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle